turtlebot/turtlebot4 #181: No map received with ros2 launch turtlebot4_navigation slam.launch.py

huayil, 2023-05-27:
**Using**
PC OS: Ubuntu 22.04
PC ROS Distro: Humble
Raspberry Pi OS: Ubuntu 22.04
Raspberry Pi ROS Distro: Humble
Create® 3 firmware version: H.1.2

 **Expected behaviour**
Follow the [tutorial](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html) and generate a map.

 **Actual behaviour**
No map is received.

**To Reproduce**

- Power on Turtlebot4.
- Confirm the PC is connected to the Raspberry Pi properly.
- On PC, run `ros2 launch turtlebot4_navigation slam.launch.py` and `ros2 launch turtlebot4_viz view_robot.launch.py` to generate a map, but no map is received.

[image]

**Other notes**
After Turtlebot4 is powered on, the lidar is spinning but no `/scan` topic is published. If call `ros2 launch turtlebot4_bringup rplidar.launch.py` after ssh into the Raspberry Pi, `/scan` appears in the topic list, but the lidar stop spinning. 

- Is lidar supposed to stop?

- Now, call `ros2 launch turtlebot4_navigation slam.launch.py` on the PC. The lidar spins, and `ros2 topic echo /scan` shows the `/scan` topic, but the no map received issue persists.

- The `/scan` topic is in the `rplidar_link` frame, but there is not transformation of it to `odom` or `base_link` by checking `ros2 run tf2_tools view_frames`. Is this normal?

- The timestamp of `/scan` is different compared with other topics such as `odom`. Is this normal?

- The followings are returned from running SLAM:

> [sync_slam_toolbox_node-1] [INFO] [1685221627.911840133] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1685221627.781 for reason 'discarding message because the queue is full'

> [sync_slam_toolbox_node-1] [INFO] [1685221628.112638819] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1685221627.903 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

roni-kreinin, 2023-05-29:
@huayil The RPLIDAR not spinning is normal. The turtlebot uses the [auto standby](https://github.com/turtlebot/turtlebot4_robot/blob/3d92a3f2b7ea889e198352322e231a7fab540dcf/turtlebot4_bringup/launch/rplidar.launch.py#LL34C18-L34C30) feature to save some power consumption. The RPLIDAR will start spinning when a node is subscribed to the `/scan` topic. You shouldn't need to manually call `ros2 launch turtlebot4_bringup rplidar.launch.py` as it should be launched by the turtlebot4 service. Can you share the logs from `sudo journalctl -u turtlebot4 -r` when the `/scan` topic doesn't show up?

Regarding the `odom` to `base_link` transform, those should be supplied by the Create 3. Can you check the `/tf` and `/odom` topics to make sure you are receiving data?

huayil, 2023-05-29:
I called `sudo journalctl -u turtlebot4 -r` after power on and `ssh` into the turtlebot4. I got the following logs of the last session. Does it mean the turtlebot4 service fail to run? 
[image]

And to clarify, the `odom` to `base_link` transformation exists, but there is no `odom` to `rplidar_link` transformation or `base_link` to `rplidar_link` transformation.

roni-kreinin, 2023-05-30:
@huayil Try running `sudo apt install ros-humble-joint-state-publisher` to install the missing package, then `sudo systemctl restart turtlebot4` to restart the service.

@jenssjogren Does `sudo systemctl restart turtlebot4` fix the issue?

Hendercraft, 2023-06-01:
> @huayil Try running `sudo apt install ros-humble-joint-state-publisher` to install the missing package, then `sudo systemctl restart turtlebot4` to restart the service.
> 
> @jenssjogren Does `sudo systemctl restart turtlebot4` fix the issue?

Also had issue during bringup, this seems to have fixed it for me

roni-kreinin, 2023-06-01:
The missing `joint_state_publisher` package has been added to `turtlebot4_description` in version 1.0.3. If you are seeing the error `[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'joint_state_publisher' not found, searching: ['/opt/ros/humble']"`, try updating all of your packages with `sudo apt update && sudo apt upgrade`.

huayil, 2023-06-06:
@roni-kreinin Thank you for your suggestions.

I ran `sudo apt update && sudo apt upgrade` and then power off and power on. The bringup issue seems to be resolved, but I still have the no map received problem. 

Here is a screenshot of rviz after I run `ros2 launch turtlebot4_navigation slam.launch.py` and `ros2 launch turtlebot4_viz view_robot.launch.py`.
[image]

Here is a screenshot of the last session when I run `sudo journalctl -u turtlebot4 -r`.
[image]

Could you please provide further debugging steps?

I have other observations that seem difficult to make sense. Could you please explain?
- The lidar always spins. The stop_motor service is ignored because it is in auto standby mode as shown in the log, but the lidar keeps spinning after power on.
- As I mentioned in the first post, the `/scan` and `/odom` topics have different timestamp. Is this normal?
[image]

jenssjogren, 2023-06-07:
After running `sudo apt update && sudo apt upgrade` my issue with the service crashing seems to be solved now. Still same problem as @huayil with the rviz map. My lidar always spin aswell, stops for a sec only when I restart the robot service. I noticed my computer, the Raspberry PI and the Create 3 all ran at different timezones. Maybe that could be affecting your timestamp @huayil?

jenssjogren, 2023-06-07:
Finally got the map after running `turtlebot4-ntpd-sync` to sync the time between PI and Create 3 as mentioned in #141.

huayil, 2023-06-07:
@jenssjogren Thanks for your comments.

`turtlebot4-ntpd-sync` did not fix the time synchronization issue for me. I also tried going to the web server of Create3, and under the tab of "Beta Features", I tried "Restart ntpd", which did not fix the issue either.

However, after manually changing the time in "Set Date and Time" (under the "Beta Features" tab) to a time that is close enough to the Raspberry Pi's time, I can now see the map in rviz. I am not sure if this is the correct way to fix the issue because
1. When the turtlebot is physically moving, the turtlebot in rviz is not moving on the map; Also, the map drifts and rotates randomly while it is still being generated.
2. The Create3 does not seem to remember the time so I need to manually set the time from the web server after reboot.

closed 2023-09-21 by hilary-luo

sardanian, 2023-10-11:
I am running into these same issues. My map is also not updating. How was this resolved?

huayil, 2023-10-11:
@hilary-luo My questions about fixing the time synchronization issue were never answered. It's been a few months since my last post. When I update to the latest TurtleBot4 ROS packages and Create3 firmware, I still have the issue. 
The time of the Create3 topics and the time of the RPi topics, such as /odom v.s. /scan, are different. As a result, mapping and navigation cannot run properly. `turtlebot4-ntpd-sync` does not fix the issue for me.
My temporary "fix" is to turn on the wifi of Create3 so that its time can be automatically set to the actual time, but I use Discovery Server for the connection, and the documentation specifically mentions the Create3 wifi has to be disabled. 
Could you provide the solution if you believe this issue has been resolved?

hilary-luo, 2023-10-12:
@huayil This issue was closed based on a lack of activity and a quick scan of the conversation seemed like people had gotten their systems working. If you still have a persistent issue then feel free to reopen this ticket and we can continue the dialog. Have you tried implementing the fix listed here: https://github.com/turtlebot/turtlebot4/issues/216 ?

@sardanian the specific fix depends on the specifics of your system and the specifics of your problem. There are a number of problems and solutions described in this ticket so please go through and try the things listed in response to the type of problem that you have. If you are looking for more customized assistance then please open your own ticket and share all of the information about your system and what problems you are encountering.

coco-fr, 2024-02-29:
I do have the same Problem. My Lidar is Turning the whole Time, but when I open Rviz2, I can`t see the map. The /map topic doesnt recieve a map.

coco-fr, 2024-02-29:
The map only appears in this case 

[INFO] [launch]: All log files can be found below /home/piekoszadm/.ros/log/2024-02-29-13-19-33-155847-innb03005-19933
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [19934]
[sync_slam_toolbox_node-1] [INFO] [1709209173.310027927] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1709209173.552073722] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1709209173.552366439] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] [INFO] [1709209173.832557045] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1709209173.587 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1709209173.967274249] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1709209173.710 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1709209174.088735961] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1709209173.833 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]

Only when the Sensor is Registered I get a map, if it isnt then i get nothing. (This only happens after i source the underlay and the first time i run the slam and Rviz2)

hilary-luo, 2024-03-07:
@coco-fr Please open your own troubleshooting ticket with full details about your system. Provide information about what commands that you are launching and the exact terminal output that you get for each. Please also include intermediate information such as in the fault situation, are you still able to echo the scan topic. Include if you have made any modifications to the system and when you are seeing the issue that you are wanting help with.