turtlebot/turtlebot4 #182: Battery issue and communication not working

schenkch, 2023-05-29:
**Please provide the following information:**
 - OS: 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Installed
 
 **Expected behaviour**
When connected to dock all LED lights green. 
 **Actual behaviour**
 Battery and communication light are off, cannot control with controller and cannot access via RPi IP:8080 in the browser but can ssh into RPi.

**To Reproduce**
Provide the steps to reproduce:
1. Flashed SD card to downgrade to 0.1.2 Galactic image 
2. Basic setup and connection to WIFI using simple discovery as described here https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html

**Other notes**
As I had quite a few issues with Galactic especially battery issues, I decided to upgrade to Humble but with Humble I could not manage to get communication working at all. So, I decided to downgrade to Galactic again but starting with a fresh image instead of using the previous one. I tried to install the 0.1.3 image but that did not work at all: display black, LEDs off and turtlebot4 wifi did not show up. I tried it twice reflashing the card. Then I decided to go for the 0.1.2 image, such that I could upgrade manually as I had already done with my previously installed Galactic image but I have not got to that yet because of the above issues. 
I would really appreciate your help as at this point, we are stuck and cannot integrate the robot into the lab at all.

roni-kreinin, 2023-05-30:
@schenkch Can you detail how you are flashing the SD card? Are you following the instructions [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image)?

schenkch, 2023-05-30:
@roni-kreinin Yes, I followed these instructions.

schenkch, 2023-06-01:
I just wanted to let you know that the next day all the lights were on again but I could still not access the robot through the browser. So, I flashed the card again with the latest Galactic image. Now, with all lights on and the display as well. I managed to get it working with Simple Discovery again. I have no idea what happened.

roni-kreinin, 2023-06-01:
@schenkch Glad to hear that its working now. Something may have just been misconfigured or the SD card didn't get flashed properly. It's hard to say.