turtlebot/turtlebot4 #183: Can't connect Create 3 to Wifi

xaxam2001, 2023-05-30:
- OS: Ubuntu 20.04
 - ROS Distro: Galactic

Hi,

I’m using a Turtlebot4 Lite (but I had the same issue with a normal one) and I can’t connect the Create 3 platform to the Wi-Fi network. I’m using a Netgear N900 router as an access point and it has two separate Wi-Fi networks, one for 5GHz and one for 2.4GHz. My computer and the RPI4 are connected to the 5GHz network but when I try to connect the Create 3, I get the error below. I should mention that the router isn’t connected to the internet (I don’t know if it should be) and I have already managed to connect the Create 3 to the Wi-Fi once, but after restarting the robot, I can’t connect it anymore. It seems to be random.

Here is the log of the Create 3 after I send the connection information:
```
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    PRODUCT         : create3
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    SERIAL          : e17628
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    CLEAN MODE      : normal
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    CONNS SVC MODE  : normal
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    AP RADIO        : wlan1/g/6/[SHORT-GI-20][GF]/Create-0558/US
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    AP IP ADDR      : 192.168.10.1
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    CL IP           : dhcp/#/#/#
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    CL DNS          : #/#/#
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    CL DHCPCD T/O   : 3
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    CL REG DOMAIN   : ETSI
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    ETH0 IP         : static/192.168.186.2
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    USB0 IP         : 192.168.186.2/192.168.187.2/192.168.186.3
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    SYSTEM_ACCESS   : locked
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    DEVELOPMENT MODE: disabled
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    RADIO_LIST      : wlan0 wlan1
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    HOSTNAME        : iRobot-05587128A5F545C1BB37D1D9F053EEC3
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    ROBOT ID        : 05587128A5F545C1BB37D1D9F053EEC3
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    BOOT COUNT      : 39
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    LANGUAGE PACK   : unknown
Apr 21 14:41:13 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:    EXTRA LANGUAGE  : not_installed
Apr 21 14:41:14 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: Provisioning of iRobot-05587128A5F545C1BB37D1D9F053EEC3 Complete
Apr 21 14:41:14 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.notice network[1466]: [INFO] (src/provisioning_sm.c:542) provisioning_state_machine_run: rx event PROVISIONING_WACTIVATED in state WAITING_FOR_WACTIVATE
Apr 21 14:41:18 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.warn network[1466]: [WARN] (src/firr.c:535) Timed out waiting for a response
Apr 21 14:41:18 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.err network[1466]: [ERROR] (src/mqtt_client.c:695) firr_transfer failed
Apr 21 14:41:18 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.err network[1466]: [ERROR] (src/provisioning_sm.c:637) Error publishing {"event":"disconnect_local"} firr request.
Apr 21 14:41:18 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: ifdown: interface wlan1 not configured
Apr 21 14:41:18 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.notice network[1466]: [INFO] (src/wifi_control.c:399) SoftAP is down
Apr 21 14:41:18 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.notice network[1466]: [INFO] (src/wifi_control.c:245) iface control: sudo network_device_interfaces_control.sh --interface=softap --action=del
Apr 21 14:41:19 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: We trust you have received the usual lecture from the local System
Apr 21 14:41:19 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: Administrator. It usually boils down to these three things:
Apr 21 14:41:19 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:     #1) Respect the privacy of others.
Apr 21 14:41:19 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:     #2) Think before you type.
Apr 21 14:41:19 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager:     #3) With great power comes great responsibility.
Apr 21 14:41:19 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper
Apr 21 14:41:19 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.err network[1466]: [ERROR] (src/wifi_control.c:207 errno:No child process) sudo network_device_interfaces_control.sh --interface=softap --action=del returned status 1
Apr 21 14:41:20 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: ifdown: interface wlan0 not configured
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: Successfully initialized wpa_supplicant
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.030388] AndesSendCmdMsg: Could not send in band command due to diable fRTMP_ADAPTER_MCU_SEND_IN_BAND_CMD
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.035610] RtmpOSFileOpen(): Error 2 opening /etc/wireless/mediatek/SingleSKU.datET
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.035619] --> Error opening /etc/wireless/mediatek/SingleSKU.datET - try /etc/wireless/mediatek/SingleSKU.dat
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.035648] RtmpOSFileOpen(): Error 2 opening /etc/wireless/mediatek/SingleSKU.dat
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.035653] --> Error opening /etc/wireless/mediatek/SingleSKU.dat
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.333598] RtmpOSFileOpen(): Error 2 opening /etc/wireless/mediatek/SingleSKU.datET
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.333612] --> Error opening /etc/wireless/mediatek/SingleSKU.datET - try /etc/wireless/mediatek/SingleSKU.dat
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.333626] RtmpOSFileOpen(): Error 2 opening /etc/wireless/mediatek/SingleSKU.dat
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 kern.warn kernel: [ 6412.333631] --> Error opening /etc/wireless/mediatek/SingleSKU.dat
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.warn : Libgcrypt warning: missing initialization - please fix the application
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: udhcpc: started, v1.30.1
Apr 21 14:41:24 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.err udhcpc[19581]: started, v1.30.1
Apr 21 14:41:25 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.err udhcpc[19581]: sending discover
Apr 21 14:41:25 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: udhcpc: sending discover
Apr 21 14:41:26 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.err udhcpc[19581]: sending discover
Apr 21 14:41:26 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: udhcpc: sending discover
Apr 21 14:41:26 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.notice network[1466]: [INFO] (src/gateway_pinger.c:113) wifi is disconnected
Apr 21 14:41:27 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.err udhcpc[19581]: sending discover
Apr 21 14:41:27 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice network_manager: udhcpc: sending discover
[... cut ...]
Apr 21 14:42:26 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.notice network[1466]: [INFO] (src/gateway_pinger.c:113) wifi is disconnected
Apr 21 14:42:28 iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice create-platform: [INFO] [1650552148.610123976] [system_health]: CPU usage: max 99 [%] mean 80 [%] RAM usage: 36/59 [MB]
Apr 21 14:42:46 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.notice network[1466]: [INFO] (src/gateway_pinger.c:113) wifi is disconnected
Apr 21 14:43:06 iRobot-05587128A5F545C1BB37D1D9F053EEC3 daemon.notice network[1466]: [INFO] (src/gateway_pinger.c:113) wifi is disconnected
```

xaxam2001, 2023-05-31:
UPDATE:
I'm using CycloneDDS on my computer and the RPI4, and I was using FastDDS on Create 3. So I changed that and tried to use CycloneDDS on the Create 3 platform. But when I made this change; I kept getting the same error in a loop in the logs: ```iRobot-05587128A5F545C1BB37D1D9F053EEC3 user.notice create-platform: 1650546342.469873 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.3:50534 failed with retcode -1```.

Despite this error, I tried to connect Create 3 to my phone's access point (which has access to internet) and it worked! But when I try to connect it to the router, it doesn't work again (Same error as describe above).

I'm not sure but it seems to be an issue with the fact that the router doesn't have access to the internet. I can't connect the router to the internet because of my work environment, so it would be great if someone could find a solution.

roni-kreinin, 2023-06-20:
@xaxam2001 Sorry for the delay. Is it possible that your router is blocking multicast packets? If so, I would suggest trying to use the FastDDS discovery server as instructed [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html).

xaxam2001, 2023-06-21:
I am going try this but I am not sure about it because the create 3 platform can connect to my phone's access point but all other access point that I tried did not work (of course they all had 2.4GHz signal). The only explanation I would have is that almost all other access points that I tried had both 2.4GHz and 5GHz signals (sometimes on different Network sometimes on the same network)  and my phone is the only one that have only a 2.4GHz network. It is just a theory but the issue could came from this.

roni-kreinin, 2023-06-21:
Are you able to separate the network into the 2.4 and 5GHz bands? That may help with this issue.

Also, if you configure the robot for the Discovery Server, the Create 3 does not need to be connected to Wi-Fi anymore. The robot can function off of just the RPi4 Wi-Fi.

xaxam2001, 2023-06-21:
Yes, I have a network with separate 2.4 and 5GHz bands and in this kind of configuration the Create 3 does not connect to the 2.4GHz band.

I will try to configure the robot for the Discovery server if using fastDDS does not work.

hilary-luo, 2023-09-21:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-09-21 by hilary-luo

shamlian, 2023-09-21:
In the case you are still having a problem: the Create 3 team has released G.5.3 and H.2.3 to correct a bug in the connection workflow which may improve your experience. After updating your firmware, re-enter the network SSID, password, and region to refresh the robot's configuration, and please do let us know in https://github.com/iRobotEducation/create3_docs/issues/395 if it does not.