turtlebot/turtlebot4 #184: Led screen and light not turning on

Tanjim13, 2023-05-31:
- OS: (e.g. Ubuntu 20.04)
 - ROS Distro: (e.g. Galactic)

 **Expected behaviour**
 Should see the WiFi, COMM or any other LED light on and screen shows something.
 **Actual behaviour**
 Black screen. I can access to raspberry pi though.

**Want**
What to do to solve this issue.
[image]
[image]
[image]
[image]

Tanjim13, 2023-06-01:
As, I am trying to connect to static ip. I got this ip address. But right now it is not even changing even if i change the wifi.
[image]

Tanjim13, 2023-06-01:
@roni-kreinin Yes, this is another turtlebot4. I don't know why the led screen and lights are not turning on.

Hendercraft, 2023-06-06:
We have the same issue with our turtlebot4s pro, we switch screen between the two turtlebot and the issue remains.
It only seems to affect humble tho, as one of the turtlebot was running galactic and working fine until we flashed it to humble

roni-kreinin, 2023-06-06:
Can you check `sudo systemctl status turtlebot4` when SSH'd into the turtlebot for any errors?

Hendercraft, 2023-06-13:
Doesn't seems like there's any
```
● turtlebot4.service - "bringup turtlebot4"
     Loaded: loaded (/lib/systemd/system/turtlebot4.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-03-20 15:32:42 CET; 2 months 24 days ago
   Main PID: 767 (turtlebot4-star)
      Tasks: 124 (limit: 4416)
     Memory: 391.6M
        CPU: 8min 41.632s
     CGroup: /system.slice/turtlebot4.service
             ├─ 767 /bin/bash /usr/sbin/turtlebot4-start
             ├─ 993 /usr/bin/python3 /opt/ros/humble/bin/ros2 launch /tmp/turtlebot4.launch.py
             ├─1169 /opt/ros/humble/lib/turtlebot4_node/turtlebot4_node --ros-args -r __ns:=/pro_2 --params-file /tmp/tmpty21grko --params-file />
             ├─1171 /opt/ros/humble/lib/joy_linux/joy_linux_node --ros-args -r __node:=joy_linux_node -r __ns:=/pro_2 -r /diagnostics:=diagnostics
             ├─1173 /opt/ros/humble/lib/teleop_twist_joy/teleop_node --ros-args -r __node:=teleop_twist_joy_node -r __ns:=/pro_2 --params-file /t>
             ├─1175 /opt/ros/humble/lib/rplidar_ros/rplidar_composition --ros-args -r __node:=rplidar_composition -r __ns:=/pro_2 --params-file />
             ├─1177 /opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=oakd_container -r __ns:=/pro_2
             ├─1179 /opt/ros/humble/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=robot_state_publisher -r __ns:=/pro_2 ->
             ├─1181 /usr/bin/python3 /opt/ros/humble/lib/joint_state_publisher/joint_state_publisher --ros-args -r __node:=joint_state_publisher >
             ├─1183 /opt/ros/humble/lib/diagnostic_aggregator/aggregator_node --ros-args -r __ns:=/pro_2 --params-file /tmp/tmph_b7yrzv -r /diagn>
             └─1185 /usr/bin/python3 /opt/ros/humble/lib/turtlebot4_diagnostics/diagnostics_updater --ros-args -r __ns:=/pro_2 -r /diagnostics:=d>

Jun 13 16:30:00 localhost.localdomain turtlebot4-start[993]: [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/pro_2/oakd' in cont>
Jun 13 16:32:27 localhost.localdomain turtlebot4-start[993]: [turtlebot4_node-1] [INFO] [1686666747.892747920] [pro_2.turtlebot4_node]: OAKD stop>
Jun 13 16:32:27 localhost.localdomain turtlebot4-start[993]: [turtlebot4_node-1] [INFO] [1686666747.893876258] [pro_2.turtlebot4_node]: RPLIDAR s>
Jun 13 16:32:28 localhost.localdomain turtlebot4-start[993]: [turtlebot4_node-1] [INFO] [1686666748.893608010] [pro_2.turtlebot4_node]: oakd/stop>
Jun 13 16:32:28 localhost.localdomain turtlebot4-start[993]: [component_container-5] [INFO] [1686666748.894565499] [pro_2.oakd]: Stopping camera.
Jun 13 16:32:28 localhost.localdomain turtlebot4-start[993]: [turtlebot4_node-1] [INFO] [1686666748.894748108] [pro_2.turtlebot4_node]: stop_moto>
Jun 13 16:32:29 localhost.localdomain turtlebot4-start[993]: [rplidar_composition-4] [INFO] [1686666749.176037659] [pro_2.rplidar_composition]: I>
Jun 13 16:32:29 localhost.localdomain turtlebot4-start[993]: [turtlebot4_node-1] [INFO] [1686666749.177920617] [pro_2.turtlebot4_node]: stop_moto>
Jun 13 16:32:30 localhost.localdomain turtlebot4-start[993]: [turtlebot4_node-1] [INFO] [1686666750.061697191] [pro_2.turtlebot4_node]: oakd/stop>
Jun 13 16:32:30 localhost.localdomain turtlebot4-start[993]: [turtlebot4_node-1] [INFO] [1686666750.061825986] [pro_2.turtlebot4_node]:
~
~
```

roni-kreinin, 2023-06-20:
@Hendercraft Can you share the output of `ros2 topic list` on each robot? 

@Tanjim13 Can you share the logs from `sudo journalctl -u turtlebot4 -r`.

MinsooK, 2023-07-27:
I am using Turtlebot4 pro and ROS Galactic version. 
The leds and screen are not turned on usually when I power on. So I just turn off and on several times and those are working finally.

MinsooK, 2023-08-04:
I had same issue.
In my case, "turtalebot4_test_results" folder size was over 22G and it had no space for RPi booting.
I erased all files in the folder and everything fine now.

Murad275, 2023-08-08:
I'm having the same issue on Turtlebot4 pro and ROS Galactic version. Any updates on possible reasons?

hilary-luo, 2023-09-11:
@Murad275 please open your own ticket in regards to the screen not working. There are many possible reasons that can cause that to happen. Also, feel free to peruse all of the closed tickets in regards to screens not working due to improper flashing of the sd card or disconnected components.

closed 2023-11-16 by hilary-luo