turtlebot/turtlebot4 #185: Create3 unable to connect

Tanjim13, 2023-05-31:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04)
 - ROS Distro: (e.g. Galactic)
 - Built from source or installed:

 **Expected behaviour**
 Should take me to Create3 page, after assigning 192.168.0.104:8080

 **Actual behaviour**
 Unable to connect. Why??

[image]
[image]

Tanjim13, 2023-06-01:
I think this robot has previous galactic image. Cause most of these robot came earlier, which don't have this feature. So, what should i do now?

ipa-rar, 2023-06-02:
@Tanjim13 You can update your RPi to the latest image from [here](http://download.ros.org/downloads/turtlebot4/). 

More details in this issue (https://github.com/turtlebot/turtlebot4_setup/issues/5)

roni-kreinin, 2023-06-05:
@Tanjim13 Pressing and holding Create 3 buttons 1 and 2 together for a few seconds will make the lightring turn blue and place the Create 3 into AP mode. You should then see an open Wi-Fi network called something like `CreateXXXX`. Connect to it then navigate to `192.168.10.1` in a browser to access the webserver.

Tanjim13, 2023-06-15:
@roni-kreinin, Thank you for the guidance. It solved the issue. 
@ipa-rar, Yes you are right. Currently, I believe we're using turtlebot4_standard_galactic_0.1.2. I will consider updating this in the near future.

To both @ipa-rar @roni-kreinin, I amexploring the possibility of establishing a multi-robot setup under our institutional network. The challenge I encounter is the unavailability of separate SSIDs for 2.4 GHz and 5 GHz. We only have a single SSID, "X_wifi". Is there a potential workaround to successfully set up multiple robots under this singular SSID? Your expert advice would be greatly appreciated.

roni-kreinin, 2023-06-20:
@Tanjim13 By multi-robot setup do you mean you are trying to run a fleet of turtlebots that interact with each other, or just have multiple individual robots on the same network?

Tanjim13, 2023-06-22:
@roni-kreinin , I am trying to set up multiple individual robots on the same network under the institutional internet. Basically, I have already set up robot with router using $ROS_DOMAIN ID already. But I do not want to use the wifi router that restricts us in a room. I want to use institutional wifi which is only come with wifi ssid 'Redrover'. There is no 2.4 or 5GHz. My robots are all ROS_GALACTIC with turtlebot4_standard_galactic_0.1.2 image.

How can I do that?

closed 2023-09-21 by hilary-luo