turtlebot/turtlebot4 #186: Multiple Issues with ROS2 Humble communication with Turtlebot4

Kazimbalti, 2023-06-01:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 
The image of turtlebot4_standard_humble.zip file is stored in the SD card and then inserted on the RPI, All the processes went very smoothly, and I have updated the firmware of create3. 
Following are some of the issues:
1. When I launch "ros2 run teleop_twist_keyboard teleop_twist_keyboard" or any other file, the turtlebot4 doesn't move, same with other commands the turtlebot4 doesn't respond.
2. When I connect the turtlebot4 with 2.4GHz it shows "UNKNOWN" IP, while when I connect it with 5.0GHz it shows IP address.
3. On create3  to connect robot to Wi-Fi, it shows only 2.4GHz and 5.0 GHz doesn't appear.
5. on create3 it even doesn't shows any IP Address: as shown in figure
[image]

Following are the topic list on both PC and RPI 
ROS2 toic list 

/battery_state
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/parameter_events
/robot_description
/rosout
/scan
/stereo/depth
/tf
/tf_static
/wheel_status

ipa-rar, 2023-06-02:
It is very difficult to use TB4 for multirobot navigation. I have been trying this since December 2022 and still not there yet, you can check the issues opened by me to get some insights. 
However If you figure out a way to do this please submit some documentation on it.

Kazimbalti, 2023-06-05:
thanks @ipa-rar for your response.
Yes, I am still working on it and didn't get succeeded.
Please can you share the link to the issues you opened it?
thank you

Kazimbalti, 2023-06-05:
Secondly, @ipa-rar @roni-kreinin 
I am thinking to switch back to Ubuntu20.04, Galactic, and try to work on multi-robot navigation using NAV2.
What do you suggest?
Thank you

closed 2023-09-21 by hilary-luo