turtlebot/turtlebot4 #187: Raspberry Pi IP "Unknown"

EMWetzel, 2023-06-02:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed
 - Package version: Don't recall

 **Expected behaviour**
 ssh into Raspberry Pi once connected to network

 **Actual behaviour**
Our university wifi was problematic in getting the Turtlebot4 to connect. I assigned the bot to a wifi network, got an IP for the Raspberry Pi, but the university network wasn't allowing the robot to connect. Needed to update wifi connection, but any attempt to get back into the Pi failed. Hard wired into raspberry pi to ssh and change wifi, did it successfully, but it still wouldn't connect. After many attempts and a restart, Raspberry Pi now reads "Unknown" IP and I can't ssh. I suspect I need to reset the raspberry pi to put on a new network I have setup that doesn't have nearly the restrictions that the university wifi has. 

**To Reproduce**
Provide the steps to reproduce:
Tough to do because of how many attempts I did to try to get the bot to connect to the various networks. 

**Other notes**
If I can't ssh I am dead in the water. My knowledge of networks is not robust.

Kazimbalti, 2023-06-05:
Hello @EMWetzel 
You need to check your Internet(WiFi), speed.
This means you need to connect the TB4 with 5.0GHz, then this issue will be resolved, 
 I think you have a 2.4GHz Wifi Connection that's why it shows an "Unkwon" IP address.
 Hope it will resolve your problem.
 thank you

roni-kreinin, 2023-06-05:
@EMWetzel If you are unable to SSH in over ethernet, you can either try to connect a micro HDMI to the RPi and use a keyboard and monitor to log in and change the networking, or re-flash the SD card with the latest image. Instructions for flashing are available [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image).

EMWetzel, 2023-06-13:
I have reimaged the sd card, but I am still getting the "Unknown" error when I reinsert and restart the bot. I assumed reimaging would take me all the way back to the start - is that not the case? Do i need to clear the sd card before reimaging?

roni-kreinin, 2023-06-20:
@EMWetzel Are you able to SSH into the RPi4? If so can you check the turtlebot4 service logs with `sudo journalctl -u turtlebot4 -r`?

HyydenJakcz, 2023-09-20:
I have the same problem as OP and am unable to ssh into the Pi. Did you find a solution to this problem?

hilary-luo, 2023-09-21:
@HyydenJakcz It is not clear to me which part of this issue that you are running into. These sorts of issue are generally user error and can have many many different sources based on the specifics of what a particular user has done.  Can you please open your own issue and give us the full details on what you have done and where you started to encounter issues? I can help you troubleshoot it there. If your issue is that the robot won't turn on properly after flashing the sd card then most commonly it means that the sd card was not flashed properly. If your issue is being unable to ssh in then there are several different possible causes including misconfigured networking, improper sd card flashing, or the raspberry pi having its power disconnected.

judyakl, 2023-10-12:
> Got this issue fixed. That said, I ran updates to the Raspberry Pi (as is stated in the documents), and now the robot wont start up correctly. Only the light ring, no screen, no lights, and the raspberry pi has red lights inside the bot. [image]

how did you manage to solve the Unkown IP@ for the RPi, i put wrong credentials for my wifi, so i connected the RPi to the laptop via ethernet, and i tried to ssh into the rpi but they are requesting a password that i'm confused about. what password should i put? the default rpi password doesn't work

HyydenJakcz, 2023-10-12:
@hilary-luo , you were right. I am new to using raspberry Pis and Linux so I did it incorrectly. It works fine now.

judyakl, 2023-10-12:
> @hilary-luo , you were right. I am new to using raspberry Pis and Linux so I did not flash the SD card correctly. It works fine now after reflashing it correctly.

@HyydenJakcz i am also a beginner, and i see that you ran into my problem. I put wrong wifi credentials (i think) and i got unkown ip@, i connected my rpi to my laptop via ethernet, but i cannot ssh because i dont know what password they want. do you have any suggestions?

hilary-luo, 2023-10-12:
@judyakl I believe your problem is different. When you ssh in across the wired connection you need to use the turtlebot user credentials. These are easily confused with the access point credentials but are not the same. The credentials are listed here: https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#ssh-into-the-raspberry-pi , the only difference being that if you are connecting over the ethernet connection then you will use the ethernet port IP address as listed here https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi. 

In summary, you should be able to ssh using the password `turtlebot4` and then use the `turtlebot4-setup` tool again to fix the wifi credentials.

judyakl, 2023-10-13:
@hilary-luo thank you for your reply! i've been trying to follow the links for several days now but i'm getting "no route to host" error, according to the setup webpage i might need to work on the rpi's sd card. i hope i figure it out before i do that though

EMWetzel, 2023-10-13:
@judyakl I don't know where you are located, but I am at a university with wifi that has a number of secondary password requirements. Two factor authentication and Turtlebot don't mix well. I ended up just getting a hotspot for the bot, reflashing the sd (mmcblk0, not SDA), and then entering the single UN and PW. A couple of troubleshooting things I went through. Initially the bot should be really close to the computer and controller it is connecting to and the wifi signal should be strong between the two. Startup takes longer than you would expect for all 5 LEDs to shine green. I had trouble with the controller they sent and had to go buy a separate PS4 controller. I started with the youtube video setup because I figured that would be easier, but it never worked for me. I moved over to Github and got it after the university wifi debacle. Because I stated with the youtube video, I removed the Cyclone DDS line from the .bashrc before doing the Github setup because that is not in there. 

All of this to say, if you are still having trouble with setup, the easiest thing to do is to start fresh. Pull the SD, reflash, and start over.

judyakl, 2023-10-15:
@EMWetzel that's what i'm gonna do. i will sadly start fresh as well. i will pull the SD, reflash and do everything all over again. maybe i should've thought of the university wifi security before connecting but it never crossed my mind. thank you for replying!

closed 2023-11-16 by hilary-luo