turtlebot/turtlebot4 #188: Attempts to send the robot destinations to navigate to were unsuccessful. (associated with: ntp time, transform, odom and base_link)

tschinfra, 2023-06-05:
- OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed
 - fastrtps enabled

 **Expected behaviour**
Being able to send the robot to goal poses on a map.

 **Actual behaviour**
The robot wont't start moving.

**To Reproduce**
I was following the [tutorial for navigation on the documentation](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html). When launching the following launchfiles i get some infos/warnings. The mapping beforehand had worked. Only the saving of the map with a specified name not.
1. running: `ros2 launch turtlebot4_navigation nav2.launch.py`.
output:
```
...
[controller_server-1] [INFO] [1685961758.902200663] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
...
```
I am seeing the /odom topic and i can echo it.

2. running: `ros2 launch turtlebot4_viz view_robot.launch.py`.
output:
```
...
[rviz2-1] [INFO] [1685961757.008412115] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1685961755.735 for reason 'discarding message because the queue is full'
[rviz2-1] Warning: Invalid frame ID "left_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "right_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "left_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "right_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "left_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "right_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "left_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "right_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
...
```

3. Running: `ros2 launch turtlebot4_navigation localization.launch.py map:=scs-map.yaml`.
output:
```
...
[amcl-2] [INFO] [1685961708.094046398] [amcl]: initialPoseReceived
[amcl-2] [INFO] [1685961708.094282927] [amcl]: Setting pose (1685961708.094282): -0.258 -0.314 3.045
[amcl-2] [INFO] [1685961708.111703829] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1685961707.110 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [INFO] [1685961708.234964788] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1685961707.232 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
...
```

**Other notes**
I am sorry if there were already some of the issues mentioned. However, I did not quite understand how to solve them.

tschinfra, 2023-06-06:
I am seeing the following topics when the robot is docked:
```
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/parameter_events
/robot_description
/rosout
/scan
/stereo/depth
/tf
/tf_static
/wheel_status
```

And the following when it is off the dock:
```
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels
```

roni-kreinin, 2023-06-20:
@tschinfra Are you running simple discovery or discovery server? Also can you check the Create 3 webserver to see if the Create 3 clock is sync'd with the RPi4?

hilary-luo, 2023-09-21:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-09-21 by hilary-luo

swapchal, 2024-08-29:
Hi, did you find a fix for this?