turtlebot/turtlebot4 #190: TB4 cant avoid static and dynamic obstacles

sashag1991, 2023-06-07:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed:

 **Expected behaviour**
Avoid static and dynamic obstacles without collision 

 **Actual behaviour**
When see a static or dynamic obstacle TB4 doesn't responds and keep moving towards the obstacle

**To Reproduce**

1. run: ros2 launch turtlebot4_navigation localization.launch.py map:=map.yaml
2. run: ros2 launch turtlebot4_navigation nav2.launch.py
3. run: ros2 launch turtlebot4_viz view_robot.launch.py
4. set point to navigate to using Nav2go button in RVIZ

**Other notes**
Add anything else you thing is important.

sashag1991, 2023-06-07:
I  tried turtlebot4-ntpd-sync before the run

sashag1991, 2023-06-07:
I already try turtlebot4-ntpd-sync from RPi4 and from Create 3 webserver.
Will there be some fix for the CPU issue ?

sashag1991, 2023-06-07:
I use Discovery server with  FastRTPS.
Is moving to Discovery server with CycloneDDS can help ?

roni-kreinin, 2023-06-09:
@sashag1991 Discovery server is only available in FastRTPS. The CPU issue is being tracked [here](https://github.com/iRobotEducation/create3_docs/issues/381).

sashag1991, 2023-06-11:
Correct me if I'm wrong but in https://github.com/iRobotEducation/create3_docs/issues/381. said:  "The individual robots can run fine but as soon as two are on the same network they run into issues".
I that c ase running only 1 robot should work, right ?

roni-kreinin, 2023-06-12:
@sashag1991 That will depend on what software you are running. In my experience, running the navigation stack can bring the Create 3 CPU to 100% which will cause issues such as the TF frames having old timestamps.

sashag1991, 2023-07-09:
[roni-kreinin](https://github.com/roni-kreinin) do you have any suggestions how do I can procced ? I need complete some tasks with the robot and I cant right now.

sashag1991, 2023-07-19:
[roni-kreinin](https://github.com/roni-kreinin), Any update?

roni-kreinin, 2023-07-19:
@sashag1991 I would recommend updating your NTP config as mentioned in #216. Some users have reported that this improves their experience with mapping. There isn't yet a fix for the CPU usage issue.

sashag1991, 2023-07-27:
The NTP config and latest H2.2 DIDNT helped to solve my issue with avoid obstacles. Maybe I should change some parameter in nav2 config yaml ? I have no idea how to solve it and this ability is necessary for my activity.

sashag1991, 2023-08-20:
Any update/help ?

hilary-luo, 2023-08-28:
@sashag1991 Can you please provide the create3 log after recreating this issue? and please clarify for me your exact setup. Are you still using Discovery Server (with Fast DDS) and operating a single robot? What firmware are you running on the create3?

After doing that, can you please download and make copies of the [nav2 configs](https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_navigation/config). Then in your local copies, modify every occurence of `topic: scan` to indicate the global topic name, for example [here](https://github.com/turtlebot/turtlebot4/blob/d4412899e428196b0905e4edd2896b8c98ce6580/turtlebot4_navigation/config/nav2.yaml#L169). If you don't have a namespace then that would be `topic: /scan` or if you do have a namespace then `topic: /<namespace>/scan`. Then, launch navigation using these modified configs. 

```
ros2 launch turtlebot4_navigation localization.launch.py map:=map.yaml params:=path_to_config/localization.yaml
ros2 launch turtlebot4_navigation nav2.launch.py params:=path_to_config/nav2.yaml
```

If this still doesn't work then please once again provide the create3 log from the attempt.

sashag1991, 2023-08-29:
Hi @hilary-luo, I changed the scan topic to /scan in localization.yaml and nav2.yaml, TB4 still not avoids obstacles.
Here the log from create3 : [create3_log.txt](https://github.com/turtlebot/turtlebot4/files/12461414/create3_log.txt)
I use latest FW humble H.2.2

closed 2023-09-21 by hilary-luo