turtlebot/turtlebot4 #192: Turtlebot4 Raspberry Pi unreachable over Wi-Fi on first start

smatar22, 2023-06-09:
**Please provide the following information:**
 - OS: Unknown at this point will update soon
 - ROS Distro: Unknown at this point will update soon
 - Built from source or installed:
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**
Raspberry Pi should be in AP mode with a Wi-Fi network named Turtlebot4 visible to host computer

 **Actual behaviour**
Raspberry Pi is unreachable, no display on the small screen, all LEDs are dark, no Turtlebot4 Wi-Fi network

**To Reproduce**
TB4 is docked, and powered on, center ring is white.

**Other notes**
Clip of TB4 docked: https://drive.google.com/file/d/102YbEdG1AAKVyjYKJZ4iiDcjREsruCI-/view
Is there a way to access the Raspberry Pi?

yorambenhaim, 2023-06-14:
Hi.  I noticed that the RPI red LED was ON. But the small screen near the buttons was OFF (i.e No display).
Another question. What do I've to do with the SanDisk micro-SD attached to the package.

llyf-s, 2023-06-15:
the SanDisk is an adapter for your turtlebot's RPi ssd card, for you to install latest Pi image if necessary (1. Basic Setup)

yorambenhaim, 2023-06-15:
Great. Thank You.
Do you think I should update/upgrade the PI image?  Is there any important added value or some other reason to do so ?
BTW. The micro-SD is on the far side of the PI board (on the other side of the USB connectors) and it is not accessible , unless I’ll uninstall the PI.

B.R.
Yoram Ben-Haim

Intel Israel (74) Limited

This e-mail and any attachments may contain confidential material for
the sole use of the intended recipient(s). Any review or distribution
by others is strictly prohibited. If you are not the intended
recipient, please contact the sender and delete all copies.