turtlebot/turtlebot4 #194: Turtlebot4 bringup failed after update to latest [cannot access *.xacro]

ipa-rar, 2023-06-14:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Installed:

 **Expected behaviour**
Robot should bringup all the nodes and all the topics should be visible

 **Actual behaviour**
Robot nodes are not started up and the Error message looks like this.

```
× turtlebot4.service - "bringup turtlebot4"
     Loaded: loaded (/lib/systemd/system/turtlebot4.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2023-06-14 11:09:37 CEST; 15s ago
    Process: 2439 ExecStart=/usr/sbin/turtlebot4-start (code=exited, status=1/FAILURE)
   Main PID: 2439 (code=exited, status=1/FAILURE)
        CPU: 5.705s

Jun 14 11:09:32 ubuntu systemd[1]: Started "bringup turtlebot4".
Jun 14 11:09:34 ubuntu turtlebot4-start[2506]: ls: cannot access '/etc/ros/humble/turtlebot4.d/*.xacro': No such file or directo>
Jun 14 11:09:35 ubuntu turtlebot4-start[2517]: [INFO] [launch]: All log files can be found below /tmp/2023-06-14-11-09-35-347789>
Jun 14 11:09:35 ubuntu turtlebot4-start[2517]: [INFO] [launch]: Default logging verbosity is set to INFO
Jun 14 11:09:37 ubuntu turtlebot4-start[2517]: [ERROR] [launch]: Caught exception in launch (see debug for traceback): executabl>
Jun 14 11:09:37 ubuntu systemd[1]: turtlebot4.service: Main process exited, code=exited, status=1/FAILURE
Jun 14 11:09:37 ubuntu systemd[1]: turtlebot4.service: Failed with result 'exit-code'.
Jun 14 11:09:37 ubuntu systemd[1]: turtlebot4.service: Consumed 5.705s CPU time.
```

**To Reproduce**
Provide the steps to reproduce:
1. turtlebot4-setup
2. Robot Upstart --> Status
3. see the error

**Other notes**
Robot was updated to the latest version yesterday and before the updates the nodes were running fine.
`sudo apt update && sudo apt upgrade --y`
Also the date and time of the robot seems to be off the actual values.

ipa-rar, 2023-06-14:
Digging further into the issue, I came across this error while trying to launch from the robot directly.
**To repoduce**
`ros2 launch turtlebot4_bringup standard.launch.py 
`

```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-06-14-11-36-55-762782-ubuntu-3449
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'rplidar_composition' not found on the libexec directory '/opt/ros/humble/lib/rplidar_ros' 
```
**Cause**
`rplidar_composition` is  `rplidar_node`. I am not sure how this is left unchecked during the release.

agrueneberg, 2023-06-15:
Thanks @ipa-rar, I can confirm that replacing `rplidar_composition` with `rplidar_node` in both `name` and `executable` in `/opt/ros/humble/share/turtlebot4_bringup/launch/rplidar.launch.py` fixes the bringup issue.

roni-kreinin, 2023-06-15:
Looks like the most recent release for rplidar_ros changed the node name. Looking into a solution.

ipa-rar, 2023-06-16:
> Looking into a solution

As @agrueneberg suggested modifying the ``/opt/ros/humble/share/turtlebot4_bringup/launch/rplidar.launch.py`` is a hotfix right now on the newly updated robots. I have tried this and works now as expected. 
But on a longer term there should be a better solution for the users.

roni-kreinin, 2023-06-16:
The 2.1.2 release of `rplidar_ros` has been reverted (https://github.com/ros/rosdistro/issues/37679), so you should be able to reinstall the `ros-humble-rplidar-ros` package to go back to 2.1.0 now. Any users running `sudo apt update && sudo apt upgrade` should not run into this issue now. We will continue to monitor this.

closed 2023-06-16 by roni-kreinin

agrueneberg, 2023-07-03:
Thanks, I managed to downgrade to version 2.1.0:

    sudo apt-get install ros-humble-rplidar-ros=2.1.0-1jammy.20230624.121119