turtlebot/turtlebot4 #200: Rviz not generating a map, but Turtlebot4 is visable

TRVLabs-Axel, 2023-06-19:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Humble 1.2
 - Built from source or installed:
 - Package version: 
`sudo dpkg -s ros-humble-turtlebot4-navigation`
Package: ros-humble-turtlebot4-navigation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 2470
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 1.0.2-1jammy.20230516.173216
Depends: ros-humble-nav2-bringup, ros-humble-nav2-simple-commander, ros-humble-slam-toolbox, ros-humble-ros-workspace
Description: Turtlebot4 Navigation

 **Expected behaviour**
I want to generate a map of the room I am in using Rviz as per the [guide](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html)

 **Actual behaviour**
 When launching Rviz I can see the Turtlebot4, but the room is not being mapped with the lidar. 

[image]

These are the errors that are being presented in Rviz:

[image]

[image]

**To Reproduce**
Provide the steps to reproduce:
1. run `ros2 launch turtlebot4_navigation slam.launch.py`
2. run `ros2 launch turtlebot4_viz view_robot.launch.py`
3. see the error

**Other notes**
The terminal that I run Rviz in generates this warning, but I do not know if it is relevant:
`Warning: Invalid frame ID "" passed to canTransform argument source_frame - in tf2 frame_ids cannot be empty`

And the terminal which runs SLAM prints this:
`LaserRangeScan contains 1014 range readings, expected 1006`

I have tried to run asynchronous SLAM, but that does not change anything. I have also tried to change the `Fixed Frame`, in Global Options, to `odom` (which does not change anything) and `rplidar_link` which removes the error with the `LaserScan`, but does not generate a map. 

Here is the output from `ros2 topic list`:
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/dock_status
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

And here is from `ros2 node list`:
/_internal/composite_hazard
/_internal/kinematics_engine
/_internal/mobility
/_internal/stasis
/analyzers
/joy_linux_node
/mobility_monitor
/motion_control
/rgb_stereo_node
/robot_state
/robot_state_publisher
/rplidar_composition
/static_transform
/system_health
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node
/ui_mgr

VicenteMoraes, 2023-06-21:
Hi,

I'm having the exact same issue and haven't been able to run SLAM in neither the real nor simulated robot. Just wondering if there's something I'm doing wrong as well.

TRVLabs-Axel, 2023-06-30:
I have now flushed the Turtlebot 4 so that it is a clean install with humble. All parts of the system are now running ROS 2 Humble, but the issue is still the same. My Ubuntu version is 22.04 and not 20.04 as I stated above so sorry for that mistake. Do you have any other tips that I could try?

TRVLabs-Axel, 2023-07-04:
Luckily we had access to a Turtlebot 4 lite, so we took the RPLIDAR from the lite version and put it on the standard version. This solved our problem and we can now generate a map. I do not know if we could have solved it another way, but maybe it can be of help to others that the RPLIDAR seems to be the problem in this case.

closed 2023-07-04 by TRVLabs-Axel