turtlebot/turtlebot4 #206: Fail to install ros-humble-turtlebot4-desktop

yorambenhaim, 2023-06-24:
Ubuntu 22.04.2 LTS 
ROS Distro HUMBLE installed thru 'sudo apt update && sudo apt install ros-humble-turtlebot4-desktop' 
ROS2 Installed snapshot:
administrator@vtglnx117:~/.ros$ dpkg -l | grep ros2
ii  ros-humble-demo-nodes-cpp                         0.20.3-1jammy.20230118.004446           amd64        C++ nodes which were previously in the ros2/examples repository but are now just used for demo purposes.
ii  ros-humble-demo-nodes-py                          0.20.3-1jammy.20230117.232639           amd64        Python nodes which were previously in the ros2/examples repository but are now just used for demo purposes.
ii  ros-humble-ros2action                             0.18.5-1jammy.20230117.234634           amd64        The action command for ROS 2 command line tools.
ii  ros-humble-ros2bag                                0.15.4-2jammy.20230118.030752           amd64        Entry point for rosbag in ROS 2
ii  ros-humble-ros2cli                                0.18.5-1jammy.20230117.232935           amd64        Framework for ROS 2 command line tools.
ii  ros-humble-ros2cli-common-extensions              0.1.1-4jammy.20230118.012959            amd64        Meta package for ros2cli common extensions
ii  ros-humble-ros2component                          0.18.5-1jammy.20230118.004321           amd64        The component command for ROS 2 command line tools.
ii  ros-humble-ros2doctor                             0.18.5-1jammy.20230117.233330           amd64        A command line tool to check potential issues in a ROS 2 system
ii  ros-humble-ros2interface                          0.18.5-1jammy.20230117.233637           amd64        The interface command for ROS 2 command line tools
ii  ros-humble-ros2launch                             0.19.4-1jammy.20230117.234435           amd64        The launch command for ROS 2 command line tools.
ii  ros-humble-ros2lifecycle                          0.18.5-1jammy.20230118.012705           amd64        The lifecycle command for ROS 2 command line tools.
ii  ros-humble-ros2multicast                          0.18.5-1jammy.20230117.233732           amd64        The multicast command for ROS 2 command line tools.
ii  ros-humble-ros2node                               0.18.5-1jammy.20230117.233331           amd64        The node command for ROS 2 command line tools.
ii  ros-humble-ros2param                              0.18.5-1jammy.20230117.233714           amd64        The param command for ROS 2 command line tools.
ii  ros-humble-ros2pkg                                0.18.5-1jammy.20230117.233641           amd64        The pkg command for ROS 2 command line tools.
ii  ros-humble-ros2run                                0.18.5-1jammy.20230117.234434           amd64        The run command for ROS 2 command line tools.
ii  ros-humble-ros2service                            0.18.5-1jammy.20230117.233350           amd64        The service command for ROS 2 command line tools.
ii  ros-humble-ros2topic                              0.18.5-1jammy.20230117.233355           amd64        The topic command for ROS 2 command line tools.
ii  ros-humble-sros2                                  0.10.4-2jammy.20230117.234843           amd64        Command line tools for managing SROS2 keys
ii  ros-humble-sros2-cmake                            0.10.4-2jammy.20230117.235240           amd64        CMake macros to configure security

**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04)
 - ROS Distro: (e.g. Galactic)
 - Built from source or installed:
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Actual behaviour**
The following failures printed after 'sudo apt update && sudo apt install ros-humble-turtlebot4-desktop' 
Ign:1 https://download.docker.com/linux/ubuntu jammy InRelease                                                                                                                                                    
Ign:2 http://il.archive.ubuntu.com/ubuntu jammy InRelease                                                                                                                                                         
Ign:3 http://security.ubuntu.com/ubuntu jammy-security InRelease                                                                                                                      
Ign:4 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                                         
Ign:5 http://packages.ros.org/ros2/ubuntu jammy InRelease                                                                                                                             
Ign:6 https://packages.microsoft.com/repos/vscode stable InRelease                                                                                                                    
Ign:7 http://il.archive.ubuntu.com/ubuntu jammy-updates InRelease                                                                                                                     
Ign:8 http://il.archive.ubuntu.com/ubuntu jammy-backports InRelease
Ign:4 https://dl.google.com/linux/chrome/deb stable InRelease
Ign:6 https://packages.microsoft.com/repos/vscode stable InRelease
Ign:1 https://download.docker.com/linux/ubuntu jammy InRelease
Ign:5 http://packages.ros.org/ros2/ubuntu jammy InRelease
Ign:2 http://il.archive.ubuntu.com/ubuntu jammy InRelease
Ign:3 http://security.ubuntu.com/ubuntu jammy-security InRelease
Ign:7 http://il.archive.ubuntu.com/ubuntu jammy-updates InRelease
Ign:8 http://il.archive.ubuntu.com/ubuntu jammy-backports InRelease
Ign:6 https://packages.microsoft.com/repos/vscode stable InRelease
Ign:1 https://download.docker.com/linux/ubuntu jammy InRelease            
Ign:4 https://dl.google.com/linux/chrome/deb stable InRelease             
Ign:2 http://il.archive.ubuntu.com/ubuntu jammy InRelease                 
Ign:3 http://security.ubuntu.com/ubuntu jammy-security InRelease
Ign:5 http://packages.ros.org/ros2/ubuntu jammy InRelease
Ign:7 http://il.archive.ubuntu.com/ubuntu jammy-updates InRelease
Ign:8 http://il.archive.ubuntu.com/ubuntu jammy-backports InRelease
Err:6 https://packages.microsoft.com/repos/vscode stable InRelease
  Could not resolve 'proxy-iil.intel.com'
Err:4 https://dl.google.com/linux/chrome/deb stable InRelease
  Could not resolve 'proxy-iil.intel.com'
Err:1 https://download.docker.com/linux/ubuntu jammy InRelease
  Could not resolve 'proxy-iil.intel.com'
Err:2 http://il.archive.ubuntu.com/ubuntu jammy InRelease
  Could not resolve 'proxy-iil.intel.com'
Err:5 http://packages.ros.org/ros2/ubuntu jammy InRelease
  Could not resolve 'proxy-iil.intel.com'
Err:3 http://security.ubuntu.com/ubuntu jammy-security InRelease
  Could not resolve 'proxy-iil.intel.com'
Err:7 http://il.archive.ubuntu.com/ubuntu jammy-updates InRelease
  Could not resolve 'proxy-iil.intel.com'
Err:8 http://il.archive.ubuntu.com/ubuntu jammy-backports InRelease
  Could not resolve 'proxy-iil.intel.com'
Reading package lists... Done
W: Failed to fetch http://il.archive.ubuntu.com/ubuntu/dists/jammy/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Failed to fetch http://il.archive.ubuntu.com/ubuntu/dists/jammy-updates/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Failed to fetch http://il.archive.ubuntu.com/ubuntu/dists/jammy-backports/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/jammy-security/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Failed to fetch https://download.docker.com/linux/ubuntu/dists/jammy/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Failed to fetch https://dl.google.com/linux/chrome/deb/dists/stable/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Failed to fetch http://packages.ros.org/ros2/ubuntu/dists/jammy/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Failed to fetch https://packages.microsoft.com/repos/vscode/dists/stable/InRelease  Could not resolve 'proxy-iil.intel.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.

My question : 
Why do I have these failures and what is the impact on the next steps ?

deebot, 2023-06-28:
I am also on Ubuntu 22.04.2 LTS and doing sudo apt install ros-humble-turtlebot4-desktop throws error. "unable to locate package  ros-humble-....."

yorambenhaim, 2023-06-28:
> @deebot<https://github.com/deebot> please ensure that you have gone through the full ros humble install instructions<https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html> before attempting to install the turtlebot4 packages.
>
> ---------------------------------------------------------------------
> Intel Israel (74) Limited
>
> This e-mail and any attachments may contain confidential material for
> the sole use of the intended recipient(s). Any review or distribution
> by others is strictly prohibited. If you are not the intended
> recipient, please contact the sender and delete all copies.

Thank you 🙏