turtlebot/turtlebot4 #207: How to make the TB4 command move 1 meter corresponding to 1 meter in reality

LIANXIAOXIAN, 2023-06-25:
I have obtained the actual map through SLAMMapEasy, but I am not very clear about the coordinate relationship of TB4 in the map and the coordinate relationship in the actual space. I use the command to make TB4 go straight forward 1 meter,
ros2 action send_goal /drive_distance irobot_create_msgs /action/DriveDistance \
"distance: 1.0
max_translation_speed: 0.3"
However, the actual measurement found that there is an error of about 10 cm. When moving two meters forward, the error reached 20 centimeters. I'm wondering how to make the distance TB4 moves correspond to the distance in real space.
Looking forward to your reply！

Jayden-F, 2023-06-26:
Hello @LIANXIAOXIAN,

The default actions of the create3 platform uses dead reckoning combination of the data from (odom, imu and mouse) to approximate its position. https://iroboteducation.github.io/create3_docs/api/odometry/. Unfortunately this will drift overtime. To account for the drift you could try using the navigation2 and slam_toolbox package. The exact setup and config is mentioned here: https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html. This should reduce the error accumulation.

Jayden-F, 2023-06-26:
Additionally, If you like a more guided approach I found the following resources very helpful:
Articulated Robotics: https://www.youtube.com/watch?v=jkoGkAd0GYk
The Construct: https://app.theconstructsim.com/courses/109.

LIANXIAOXIAN, 2023-06-30:
@Jayden-F ，Thank you very much for your reply. I will try the method you provided later. I will discuss the latest developments with you again. Sincere thanks.