turtlebot/turtlebot4 #209: No map received

AnOrdinaryUsser, 2023-06-27:
**Using**
PC OS: Ubuntu 22.04
PC ROS Distro: Humble
Raspberry Pi OS: Ubuntu 22.04
Raspberry Pi ROS Distro: Humble
Create® 3 firmware version: H.1.2

**Expected behaviour**
Follow the [tutorial](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html) and generate a map.

**Actual behaviour**
No map is received.

**To Reproduce**

- Power on Turtlebot4.
- Confirm the PC is connected to the Raspberry Pi properly.
- On PC, run ros2 launch turtlebot4_navigation slam.launch.py and ros2 launch turtlebot4_viz view_robot.launch.py to generate a map, but no map is received.

I get this in RVIZ2:
[image]

I have connected my pc, the nuc and the roomba iCreate 3 in the same wifi. I can get from my pc the topics of the nuc connected via USB C to the roomba, like the following picture:

```
sergio@sergio:~$ ros2 topic list
/battery_state
/clicked_point
/cmd_audio
/cmd_lightring
/cmd_vel
/dock_status
/downsampled_costmap
/downsampled_costmap_updates
/global_costmap/costmap
/global_costmap/costmap_updates
/global_costmap/voxel_marked_cloud
/hazard_detection
/imu
/initialpose
/interface_buttons
/ir_intensity
/ir_opcode
/kidnap_status
/local_costmap/costmap
/local_costmap/costmap_updates
/local_costmap/published_footprint
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_metadata
/map_updates
/mobile_base/sensors/bumper_pointcloud
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/particle_cloud
/plan
/pose
/robot_description
/robot_state/transition_event
/rosout
/scan
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/waypoints
/wheel_status
/wheel_ticks
/wheel_vels
```

AnOrdinaryUsser, 2023-06-30:
> 

If I do a echo of /scan topic i don't recieve nothing. But I haven't been researching and I don't know if you have to indicate on which USB port the lidar is running, but I couldn't find anywhere to indicate it.

hilary-luo, 2023-07-26:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-07-26 by hilary-luo

sdp-team-12, 2023-11-15:
Hello. I am experiencing the same issue, but the only difference is that I am able to see the RPLidar data using rplidar.launch.py. I am running this on a simulator, however, but I have set the simulated robot in an empty world. I can see on RViz that the real lidar data is being read in as well as the simulated world lidar data, but when I try to map using slam toolbox, it does not generate the map I want of the real world, but only the simulated world. 

Here is what I am seeing in Gazebo with the empty world:
[image]

Here is what I see on RViz:
[image]

So this is after I launch everything with the robot_localization package and I can see the real data on there, but the map I generate does not take that into account. Don't know what is happening. Please help.

hilary-luo, 2023-11-16:
@sdp-team-12 This venue is for issues regarding the Turtlebot4 package. Are you using the Turtlebot4 package because it does not look like it. I would encourage you to enter your issue on the github repo for the packages that you are using and/or on https://robotics.stackexchange.com/.

What I will say though is make sure that slam is looking at the right scan topic and that only 1 robot OR simulated robot is publishing to any given topic (not both)