turtlebot/turtlebot4 #213: error to update and upgrade

nidia-bucarelli, 2023-07-05:
I installed the latest pi4 image. I can connect to the network and see the topics of the iCreate and pi4.

I tried to follow the instructions for setting up the robot. I get this error when using `sudo apt update && sudo apt upgrade`: 
`Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 2706 (unattended-upgr)        
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 2706 (unattended-upgr)        
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 2706 (unattended-upgr)        
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by ^Cocess 2706 (unattended-upgr)... 495s
`

`bluetoothctl` is not installed on the pi4., so I cannot pair my controller/joystick. I cannot install it either because I get a similar error.

schenkch, 2023-07-06:
@nidia-bucarelli  The error you are seeing is probably just because the rpi is installing automatic updates. To see whether it is just related to daily updates you can do what is described here https://itsfoss.com/could-not-get-lock-error/. And if that is the case, just wait some time. 

To the second issue: I believe that installing the bluetooth package on the rpi should fix this, run:
sudo apt install pi-bluetooth
If after installing this you get an error saying that there is no default controller available:
You have to do the following:
sudo systemctl start hciuart
And such that hciuart.service is started at boot do this:
sudo systemctl enable hciuart.service

nidia-bucarelli, 2023-07-07:
Hello. Thanks much. The Pi4 is now working. I just left the RPi ON for 1 day.

closed 2023-07-07 by hilary-luo