turtlebot/turtlebot4 #215: Turtlebot4 cannot hold power

languyen14507, 2023-07-07:
Something has gone wrong with my turtlebot4. I left it docked overnight to charge it and found it flashing red light today when I came in to use it. I powered it down and docked it again to reboot. Attached is a short video which shows the red light flashing after reboot is done. I managed to use the controller to undock and drive it forward for a few seconds then the robot just shutdown by itself.

https://github.com/turtlebot/turtlebot4/assets/12131715/11cd5aba-1737-4683-a02e-53514494131d

languyen14507, 2023-07-07:
Battery is bad! Sorry for the inconvenience.

closed 2023-07-07 by languyen14507

hilary-luo, 2023-07-13:
I am glad to see that you have been able to address your issue. In the future please use one of the default templates to ensure that the issue is assigned to someone and then that person will get notified to respond.