turtlebot/turtlebot4 #217: turtlebot4_tests humble RTPS_READER_HISTORY error

1ind0r, 2023-07-12:
This is the first time I write a bug report. Besides solving the problem, I am grateful for any comments on how to make such reports better. 
- OS: Ubuntu Server 22.04.2 LTS
- ROS: Humble
- Create® 3 Firmware: H.2.1
 - Built from source or installed: installed with http://download.ros.org/downloads/turtlebot4/ latest image and Raspberry Oi Imager v1.7.2

 **Expected behaviour**
With the ros_tests node I expect to test some basic functions of the robot. I didn't expected errors. 

 **Actual behaviour**
[image]
Of course I pressed enter to start the process but nothing happened. 
I waited for 30min. Then I pressed CLT + C. I restarted the node and used Test Option 3 ('Drive Test'). There I got an Error message. 
[image]
I am a beginner and I am not sure how to fix this. I tried an other 'RMW_IMPLEMENTATION=rmw_connext_cpp' but with this I couldn't start the ros_tests node.

**To Reproduce**
Provide the steps to reproduce:
1. Flash SD card with latest humble image
2. update Create 3 to H2.1
3. update and upgrade Pi
4. 
[image]
select Enter
5. shutdown the hole system including the hole turtlebot4 ('sudo shutdown now' and turn off the Power)
6. restart the system
7. reconnect to the TurtleBot4 via your PC 
8. start the node: 'ros2 run turtlebot4_tests ros_tests' 
9. select 8 of the Test Options ('TurtleBot 4 Lite Tests)

hilary-luo, 2023-07-12:
Hello, thanks for reaching out. Overall this issue is well laid out with good details so thank you for that. 

Your initial presenting symptom where the drive section of the test hangs is likely caused by this issue: https://github.com/iRobotEducation/create3_docs/issues/381 . It is being worked on and hopefully soon we will have a work around for that. This issue is primarily seen with FastDDS. 

It is important to note that the only DDS options for the Turtlebot4 are FastDDS and CycloneDDS. You can try using [CycloneDDS](https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html#additional-cyclonedds-configuration) to see if this fixes your issues. Note: If you are using CycloneDDS then you need to be using simple discovery networking. 

The second error I have not seen before, was that a repeatable error when you tried to run the drive test?
Finally, I do want to ask, is there a problem with the blue lights? you responded no that they did not all light up blue.

1ind0r, 2023-07-14:
okay Thanks! =]

Could you show me how to change to 'simple discovery networking'? I tried using 'export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp', but it didn't affect the behavior.

### Regarding the second error:

Yes, I encountered the RTPS_READER_HISTORY error after using the ros_test option 3, but (as I now understand) it doesn't happen consistently. Sometimes, I receive no output from the command line (as shown in my first screenshot), while other times I see the text 'Waiting for odometry...', 'reserved odometry...', or even this error message. In all cases, I have rebooted the system or restarted the ros_test node. By "restarting," I mean that I stopped the ros_test node by pressing '^C', '10' (Exit), or the node crashed on its own. Then, I ran the command 'ros2 run turtlebot4_tests ros_tests' to start the node again.

### Blue lights -- no relay reproducibility:

When I took the screenshot, it didn't work. I attempted some other tries where all lights worked, but only one or two 'light' options were correct (in almost every combination).

Today, I started the system and conducted the light test. In the first round, the red light didn't work. In the second round, everything passed successfully. I tried option 4 ('Dock Test'), but it didn't work. I interrupted the process using '^C'. I then attempted the 'Light-test' again, but the node crashed. I restarted the node, but then none of the lights were functioning.

Here are the massages : 
[image]
[image]
[image]

trvlabs10, 2023-07-14:
I have the same issue when I try to do the driving test but all the other test I pass

hilary-luo, 2023-07-15:
@trvlabs10 this is expected intermittently until https://github.com/iRobotEducation/create3_docs/issues/381 is addressed (it is being actively worked on). 

@1ind0r Given that all of these issues are intermittent, I am inclined to believe the 100% CPU is the core issue, even for the lights. If you look at the [create3 webserver logs](https://iroboteducation.github.io/create3_docs/webserver/logs/) then you can confirm when the robot is near 100% cpu (it is logged periodically).

The instructions to set up simple discovery are [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html ), and heed the note near the bottom for cycloneDDS