turtlebot/turtlebot4 #218: Errors running SLAM

languyen14507, 2023-07-14:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04) : Ubuntu 20.04
 - ROS Distro: (e.g. Galactic): Galactic
 - Built from source or installed: installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash): not sure
 **Expected behaviour**
When running the command "ros2 launch turtlebot4_navigation slam_sync.launch.py" on one terminal, and the command "ros2 launch turtlebot4_viz view_robot.launch.py" on another terminal, I expect no error on the first terminal and the rviz display of the immediate environment of the Turtlebot 4

 **Actual behaviour**
 The following error messages were encounterd on the first terminal:
 [sync_slam_toolbox_node-1] [INFO] [1689279445.402710615] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650607932.261 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1689279445.528948282] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650607932.384 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

On the second terminal, the rviz display does not show a map of the TB4 environment, only an empty black map with the TB4 at the center. The following errors messages were seen on the therminal:
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
I had updated the Create 3 firmware to the latest G.4.5, tried to do a factory reset and Enable Fast DDS discovery server but the manual was not not clear so I went back to cyclone_DDS

Before these changes, I had successfully run SLAM and Navigation on this TB4 with Simple Discovery mode with the TB4 configuration as delivered from manufacturer

hilary-luo, 2023-07-15:
Please update the create3 firmware to G.5.1 and try this solution to see if it addresses your issue: https://github.com/turtlebot/turtlebot4/issues/216

languyen14507, 2023-07-15:
I did the update but it did not fix the issue. Please see the screenshots and the errors below.
[image]

[image]

[image]

[image]

[sync_slam_toolbox_node-1] [INFO] [1689453594.308049875] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650545806.762 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1689453594.445221930] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650545806.897 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1689453594.578914949] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650545807.032 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[sync_slam_toolbox_node-1] [INFO] [1689453594.601220923] [rclcpp]: signal_handler(signal_value=2)
[INFO] [sync_slam_toolbox_node-1]: process has finished cleanly [pid 5274]
languyen:turtlebot4_ws$ 

[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp

languyen14507, 2023-07-15:
I also download the Logs and include it here
[messages (3).txt](https://github.com/turtlebot/turtlebot4/files/12062798/messages.3.txt)

languyen14507, 2023-07-21:
@hilary-luo How do I reset the Turtlebot to the delivered configuration?

hilary-luo, 2023-07-21:
Hello again @languyen14507,

You can reset the turtlebot by [reimaging the raspberry pi](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image) and doing a [factory reset of the create3 base](https://iroboteducation.github.io/create3_docs/webserver/about/#:~:text=set%20to%20USB.-,Factory%20Reset,-A%20hyperlink%20to)

I haven't yet had a change to look into your earlier error messages but it is on my to do list. Let me know if it becomes no longer relevant.

languyen14507, 2023-07-21:
Hello @hilary-luo
I am reviewing the instructions to make sure I understand everything before proceeding. On the "reimaging the raspberry_pi" as shown here
it says humble but I have galactic running on my PC, will it make a difference or should I change humble to galactic on the wget command?
On the "factory reset of the create3 base" as shown here
it says firmware will not change. Currently, I have updated the Create3 firmware to G.5.1 and don't remember what the original firmware version is. Does it mean that the Create3 on the robot will stay at G.5.1? Will that be compatible or is there a way for me to find out what the original firmware version is and change back to that instead? I received my Turtlebot 4 in December 2022. Thank you so much for your help.
Best ***@***.***

hilary-luo, 2023-07-24:
That one particular script is the same for both humble and galactic so you can go ahead and follow that instruction set as is for either distribution. In general though you are correct that it is important to use the correct distribution. 

I cannot tell you what firmware that you were shipped but you can find the older versions on the [create 3 website](https://iroboteducation.github.io/create3_docs/releases/g_4_4/). That being said, I would encourage you to stick with the newer firmware and try the suggestion below:

I've now had a chance to look at the logs that you sent. The create3 is showing DDS errors that would have impeded the ROS 2 nodes from working properly. Can you switch to Fast DDS but continue using Simple Discovery. You can do this by running the ROS_config mentioned [here](https://iroboteducation.github.io/create3_docs/releases/g_4_4/). If you have imaged your pi to the latest (0.1.3) then the script should already be present in the `/usr/local/bin` folder.

Then try to run slam again. If you continue to have issues then share the terminal outputs and create3 logs again. I should be able to be more responsive this week to help you work through this.

languyen14507, 2023-07-24:
Hello @hilary-luo,
Just one quick question: I have not changed the RPi software at all, can I skip the reimaging of the RPi to the latest or do you still recommend that I do the reimaging of the RPi?
***@***.***

hilary-luo, 2023-07-24:
Based on the information so far, I do not have a reason to indicate that you should reimage the pi or revert firmware of the create3 at this point. If the DDS connection is corrected (using the aforementioned script), it may resolve the issues. It is up to you if you are wanting to revert the system instead of continuing to pursue this troubleshooting.

languyen14507, 2023-07-24:
I would like to continue this troubleshooting without reimaging the Pi. I have followed the link you provided to download and update the Create3 firmware to version G.4.4 which has    
   - irobot_create_msgs - 1.2.4
   - cyclonedds - 0.8.1
   - Fast-DDS - 2.3.3
Then tried running SLAM again. I still get the empty rviz view and the following error messagesOn terminal 1:[sync_slam_toolbox_node-1] [INFO] [1690227141.958102976] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650548404.735 for reason 'the timestamp on the message is earlier than all the data in the transform cache'[sync_slam_toolbox_node-1] [INFO] [1690227142.091980800] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650548404.870 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

On terminal 2:[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
I have attached the Create3 log file.
Thanks.

 
 

  
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 syslog.info syslogd started: BusyBox v1.30.1
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.crit kernel: [    0.000000] timer_probe: no matching timers found
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6647862422 ns
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.000012] sched_clock: 32 bits at 287MHz, resolution 3ns, wraps every 7469508094ns
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.000057] Calibrating delay loop... 380.92 BogoMIPS (lpj=1904640)
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.059947] pid_max: default: 32768 minimum: 301
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.060232] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.060248] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.062625] devtmpfs: initialized
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.067447] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.067469] futex hash table entries: 256 (order: -1, 3072 bytes)
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.067576] pinctrl core: initialized pinctrl subsystem
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.068157] NET: Registered protocol family 16
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.068878] cpuidle: using governor ladder
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.104753] mt7621_gpio 10000600.gpio: registering 32 gpios
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.105213] mt7621_gpio 10000600.gpio: registering 32 gpios
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.105856] mt7621_gpio 10000600.gpio: registering 32 gpios
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.107170] usbcore: registered new interface driver usbfs
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.107258] usbcore: registered new interface driver hub
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.107359] usbcore: registered new device driver usb
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.110319] Advanced Linux Sound Architecture Driver Initialized.
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.111001] Bluetooth: Core ver 2.22
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.111093] NET: Registered protocol family 31
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.111099] Bluetooth: HCI device and connection manager initialized
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.111112] Bluetooth: HCI socket layer initialized
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.111120] Bluetooth: L2CAP socket layer initialized
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.111153] Bluetooth: SCO socket layer initialized
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.113455] clocksource: Switched to clocksource MIPS
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.118434] NET: Registered protocol family 2
Jul 16 02:19:16 iRobot-02D154EDE40449C5845F2263ADB1EFA0 kern.info kernel: [    0.119341] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[... cut ...]
Jul 16 02:21:04 iRobot-02D154EDE40449C5845F2263ADB1EFA0 user.notice create-platform: VSTAT: No config for vstat_nav_rosKinemEngineTimer_dur_us
Jul 16 02:21:04 iRobot-02D154EDE40449C5845F2263ADB1EFA0 user.notice create-platform: VSTAT: No config for vstat_nav_rosKinemEngineTimer_int_ms
Jul 16 02:21:04 iRobot-02D154EDE40449C5845F2263ADB1EFA0 user.notice create-platform: VSTAT: No config for vstat_nav_kinem_engine_deadreck_int_ms
Jul 16 02:21:04 iRobot-02D154EDE40449C5845F2263ADB1EFA0 user.notice create-platform: [INFO] [1689474064.680747899

hilary-luo, 2023-07-24:
Have you switched to Fast DDS yet? if not please try that but continue using Simple Discovery. You can do this by running the ROS_config mentioned [here](https://iroboteducation.github.io/create3_docs/releases/g_4_4/). If you have imaged your pi to the latest (0.1.3) then the script should already be present in the /usr/local/bin folder. If not you can find it [here](https://github.com/turtlebot/turtlebot4_setup/blob/galactic/scripts/ros_config.sh).

Then reboot and try to run slam again and assuming it still causes issues, attach the create3 logs as well as sharing the terminal output.

languyen14507, 2023-07-24:
Do the attached screenshots look right? I am not sure how to switch to Fast DDS and continue using Simple Discovery.

languyen14507, 2023-07-24:
Hello @hilary-luo,
1. I saved the file  /usr/local/bin/ros_config.sh on the Pi to a new name ros_config.sh.orig2. I copied the file at "https://github.com/turtlebot/turtlebot4_setup/blob/galactic/scripts/ros_config.sh" into file /usr/local/bin/ros_config.sh on the Pi (essentially overwrote the existing ros_config.sh in that directory /usr/local/bin)3. I rebooted the Turtlebot 4 4. I ran SLAM on two terminals:Terminal 1 with the command "ros2 launch turtlebot4_navigation slam_sync.launch.py"Terminal 2 with the command "ros2 launch turtlebot4_viz view_robot.launch.py"
Still got an empty rviz map and the following errors:On terminal 1:[sync_slam_toolbox_node-1] [INFO] [1690238399.407103160] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650546062.551 for reason 'the timestamp on the message is earlier than all the data in the transform cache'[sync_slam_toolbox_node-1] [INFO] [1690238399.514193217] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650546062.686 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
On terminal 2:[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
I have attached the Create3 log file.
***@***.***

 

 
 

turtlebot4_setup/scripts/ros_config.sh at galactic · turtlebot/turtlebot...

TurtleBot 4 setup tools and configuration files. Contribute to turtlebot/turtlebot4_setup development by creatin...
 |

 |

 |

hilary-luo, 2023-07-25:
I cannot see any screenshots in your response. That script is a configuration script that you need to run on the robot. It will prompt for information and when you run it, and based on the information, will reconfigure the dds settings. It does not automatically run on its own.

languyen14507, 2023-07-25:
I ran the downloaded script ros_config.sh and got the following responses
***@***.***:/usr/local/bin$ source ros_config.shROS_DOMAIN_ID (default 0): RMW Implementation [rmw_cyclonedds_cpp,rmw_fastrtps_cpp] (default rmw_cyclonedds_cpp): rmw_fastrtps_cppDiscovery Server (default off): Workspace (default /opt/ros/galactic/setup.bash): ROS_DOMAIN_ID: 0RMW_IMPLEMENTATION: rmw_fastrtps_cppDiscovery server: offWorkspace: /opt/ros/galactic/setup.bashPress enter to apply these settings.  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current                                 Dload  Upload   Total   Spent    Left  Speed100   237  100   137  100   100     69     51  0:00:01  0:00:01 --:--:--   121{"status":"Reboot in progress."}Failed to stop turtlebot4.service: Unit turtlebot4.service not loaded.b'/usr/lib/systemd/systemd'Job not found, nothing to remove.usage: install.py [-h] [--domain DOMAIN] [--rmw RMW] modelinstall.py: error: unrecognized arguments: --workspace /opt/ros/galactic/setup.bash --discovery offFailed to start turtlebot4.service: Unit turtlebot4.service not found.
Please let me know if this selection is correct. What do the errors mean?
Thanks.

languyen14507, 2023-07-25:
Also, for your information, I went back and ran the original ros_config.sh that came with the TB4 and the following selections

ubuntu@ubuntu:/usr/local/bin$ source ros_config.sh.orig
ROS_DOMAIN_ID (default 0): 
RMW Implementation [rmw_cyclonedds_cpp,rmw_fastrtps_cpp] (default rmw_cyclonedds_cpp): rmw_fastrtps_cpp
ROS_DOMAIN_ID: 0
RMW_IMPLEMENTATION: rmw_fastrtps_cpp
Press enter to apply these settings.

sudo systemctl daemon-reload && sudo systemctl start turtlebot4
source ~/.bashrc

Then tried the SLAM again. 
This time, terminal 1 (launch turtlebot4_navigation slam_sync.launch.py) did not output any error:

languyen:~$ ros2 launch turtlebot4_navigation slam_sync.launch.py
[INFO] [launch]: All log files can be found below /home/languyen/.ros/log/2023-07-25-14-43-11-242166-languyen-ROG-Strix-G731GT-GL731GT-37598
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [37600]
[sync_slam_toolbox_node-1] [INFO] [1690314191.425754029] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1690314191.461162320] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1690314191.461384526] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[sync_slam_toolbox_node-1] [INFO] [1690314332.474645117] [rclcpp]: signal_handler(signal_value=2)
[INFO] [sync_slam_toolbox_node-1]: process has finished cleanly [pid 37600]

Terminal 2 (ros2 launch turtlebot4_viz view_robot.launch.py) has these errors:

[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp

and the rviz display is still empty.

hilary-luo, 2023-07-26:
Please reboot the robot, then after it is fully booted (still on fastdds), run slam then please provide the create3 log.

languyen14507, 2023-07-26:
I did the followings:

1. Rebooted the TB4

2. Logged into the Pi and ran the original ros_config.sh (the one that came with TB4)

   ubuntu@ubuntu:/usr/local/bin$ source ros_config.sh.orig
   ROS_DOMAIN_ID (default 0): 
   RMW Implementation [rmw_cyclonedds_cpp,rmw_fastrtps_cpp] (default rmw_cyclonedds_cpp): rmw_fastrtps_cpp
   ROS_DOMAIN_ID: 0
   RMW_IMPLEMENTATION: rmw_fastrtps_cpp
   Press enter to apply these settings.
        % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
   100   203  100   137  100    66   2044    985 --:--:-- --:--:-- --:--:--  3029
   {"status":"Reboot in progress."}b'/usr/lib/systemd/systemd'
   Preparing to remove the following paths:
     /etc/ros/galactic/turtlebot4.d
     /etc/ros/galactic/turtlebot4.d/.installed_files
     /etc/ros/galactic/turtlebot4.d/standard.launch.py
     /etc/systemd/system/multi-user.target.wants/turtlebot4.service
     /lib/systemd/system/turtlebot4.service
     /usr/sbin/turtlebot4-start
     /usr/sbin/turtlebot4-stop
   Now calling: /usr/bin/sudo /opt/ros/galactic/share/robot_upstart/scripts/mutate_files
   WARNING: Unable to remove directory /etc/ros/galactic/turtlebot4.d. Additional user-added files may be present.
   Filesystem operation succeeded.
   Installing TurtleBot 4 standard. ROS_DOMAIN_ID=0, RMW_IMPLEMENTATION=rmw_fastrtps_cpp
   candidate : /opt/ros/galactic/share/turtlebot4_bringup/launch/standard.launch.py
   b'/usr/lib/systemd/systemd'
   Preparing to install files to the following paths:
     /etc/ros/galactic/turtlebot4.d/.installed_files
     /etc/ros/galactic/turtlebot4.d/standard.launch.py
     /etc/systemd/system/multi-user.target.wants/turtlebot4.service
     /lib/systemd/system/turtlebot4.service
     /usr/sbin/turtlebot4-start
     /usr/sbin/turtlebot4-stop
   Now calling: /usr/bin/sudo /opt/ros/galactic/share/robot_upstart/scripts/mutate_files
   Filesystem operation succeeded.
   ** To complete installation please run the following command:
   sudo systemctl daemon-reload && sudo systemctl start turtlebot4
   Source ~/.bashrc to apply these changes to this terminal.
   
   ubuntu@ubuntu:/usr/local/bin$ sudo systemctl daemon-reload && sudo systemctl start turtlebot4
   ubuntu@ubuntu:/usr/local/bin$ cd
   ubuntu@ubuntu:~$ source .bashrc
   
3. Opened terminal 1 on PC and ran the following command:

   languyen:turtlebot4_ws$ ros2 launch turtlebot4_navigation slam_sync.launch.py
   The system came back with these messages:
   
   [INFO] [launch]: All log files can be found below /home/languyen/.ros/log/2023-07-26-15-23-13-706667-languyen-ROG-Strix-G731GT-GL731GT-5223
   [INFO] [launch]: Default logging verbosity is set to INFO
   [INFO] [sync_slam_toolbox_node-1]: process started with pid [5225]
   [sync_slam_toolbox_node-1] [INFO] [1690402993.831859914] [slam_toolbox]: Node using stack size 40000000
   [sync_slam_toolbox_node-1] [INFO] [1690402993.847378843] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
   [sync_slam_toolbox_node-1] [INFO] [1690402993.847488341] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
   ^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
   [sync_slam_toolbox_node-1] [INFO] [1690403606.807962098] [rclcpp]: signal_handler(signal_value=2)
   [INFO] [sync_slam_toolbox_node-1]: process has finished cleanly [pid 5225]
   
4. Open terminal 2 on PC and ran the following command 

   languyen:turtlebot4_ws$ ros2 launch turtlebot4_viz view_robot.launch.py
   
   The system came back with these errors:
   ...
   [rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
   [rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
   [rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
   [rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
   ...
   
Please find attached the Logs file "messages(8).txt" for Create3 and the Create3 config screenshot

The rviz view is empty.

   
   

   

[image]

[messages (8).txt](https://github.com/turtlebot/turtlebot4/files/12177051/messages.8.txt)

roni-kreinin, 2023-07-27:
@languyen14507 Can you provide the output from `ros2 topic list` on both the RPi4 and your PC?

languyen14507, 2023-07-27:
@roni-kreinin  These are the lists with SLAM running
[ros2_topic_list_from_RPi_with_SLAM_running.txt](https://github.com/turtlebot/turtlebot4/files/12187419/ros2_topic_list_from_RPi_with_SLAM_running.txt)
[ros2_topic_list_from_PC_with_SLAM_running.txt](https://github.com/turtlebot/turtlebot4/files/12187421/ros2_topic_list_from_PC_with_SLAM_running.txt)

hilary-luo, 2023-07-31:
Your PC is showing the SLAM topics, however they are not showing on the pi. This points to an issue with your PC configuration. When you switched to FastDDS simple discovery, did you update your computer's DDS profile to match? Your RMW_IMPLEMENTATION shoudl be set to rmw_fastrtps_cpp (`export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`) but the FASTRTPS_DEFAULT_PROFILES_FILE should not be set to anything, to use the default profile.

Once you have checked that, and assuming it does not fix the issue, please check the following:

1. Are you able to drive the robot from your PC and visualize the robot and sensor data in RViz? 
2. If you ssh into the robot and run slam on the robot, does it work?

languyen14507, 2023-08-01:
@hilary-luo 
1. I have added the statement "export RMW_IMPLEMENTATION=rmw_fastrtps_cpp" to my PC's .bashrc file and SLAM as activated from my PC still did not work (same errors as before). I can drive the robot with the controller but there is no lidar visual data on the rviz display. 
2. I ssh'ed into the RPi and tried to run SLAM from there.
   Terminal 1 "ros2 launch turtlebot4_navigation slam_sync.launch.py" had these errors:

[sync_slam_toolbox_node-1] [INFO] [1650546314.878625565] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650546314.877 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1650546315.012388176] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650546315.011 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[sync_slam_toolbox_node-1] [INFO] [1650546315.102862287] [rclcpp]: signal_handler(signal_value=2)
[INFO] [sync_slam_toolbox_node-1]: process has finished cleanly [pid 1885]
ubuntu@ubuntu:~$ 

Terminal 2 "ros2 launch turtlebot4_viz view_robot.launch.py" has these errors:

ubuntu@ubuntu:~$ ros2 launch turtlebot4_viz view_robot.launch.py
Package 'turtlebot4_viz' not found: "package 'turtlebot4_viz' not found, searching: ['/opt/ros/galactic']"
ubuntu@ubuntu:~$ 

Thanks.

languyen14507, 2023-08-04:
@hilary-luo 
Update: By using the command "sudo systemctl start bluetooth", I managed to resolve the issue with the physical controller connection to the RPi, so now I can drive the robot again with the physical controller, but the problem with SLAM remains unchanged.

languyen14507, 2023-08-07:
> Your PC is showing the SLAM topics, however they are not showing on the pi. This points to an issue with your PC configuration. When you switched to FastDDS simple discovery, did you update your [...]

@hilary-luo 
I reviewed the list of topics files and noticed that the one from RPi was not copied correctly. I have corrected it and used an editor to compare the difference between the two files: the topics from PC and RPi are almost identical except the RPi has the /visualization_marker_array and the PC doesn't have it. Please see attached.

[image]

languyen14507, 2023-08-07:
> Your PC is showing the SLAM topics, however they are not showing on the pi. This points to an issue with your PC configuration. When you switched to FastDDS simple discovery, did you update your [...]

Also, the Quick Start Guide doesn't mention the FASTRTPS_DEFAULT_PROFILES_FILE at all, so I don't have it in my PC's .bashrc file. The Quick Start Guide also says that CycloneDDS needs to be configured on the user PC in order to see the robot topics properly so I have always run with CycloneDDS and it used to work.

languyen14507, 2023-08-08:
@roni-kreinin @hilary-luo 
Are there any other data that you would like to see? I really need to get the SLAM and Navigation working again ASAP on this turtlebot 4. Thanks.

languyen14507, 2023-08-08:
@roni-kreinin @hilary-luo 
How can I be sure that my robot is configured for Simple Discovery correctly? The whole problem with SLAM started when I tried to convert to Discovery Server unsuccessfully but I am not sure if the robot has all the correct settings for Simple Discovery.

hilary-luo, 2023-08-09:
To answer your most recent question, in order to verify your simple discovery settings:
- on the raspberry pi, can you share with us the ~/.bashrc file?
- on the create3 if you go into the webserver and the application tab, the check box for discovery server should be unchecked, can you share a photo like this one below? 
[image]
- on your pc, can you share any / every line from your ~/.bashrc file that pertains to turtlebot? (or the whole thing if you are comfortable)

closed 2023-09-11 by languyen14507