turtlebot/turtlebot4 #219: Multiple Turtlebots Control

MikeDegany, 2023-07-17:
- OS:  Ubuntu 22
 - ROS Distro: Humble
 - Built from source or installed:
 - Package version: 1.0.2-1jammy.20230516.173216

 **Expected behaviour**
Connected Two Turtlebot4 Standard using AccessPoint having two separate namespaces and even DOMAIN_IDs. 
I should be able to control both robots from User PC ( e.g. by publishing cmd_vel command to /{namespace}/cmd-vel topics) if I change the ROS_DOMAIN_ID environment variable for each robots.

 **Actual behaviour**
I can control one of the robots, but I'm incapable of controlling the other one ( I change the ROS_DOMAIN_ID env variable) while I can publish cmd_vel msgs to cmd_vel topic when I ssh to that robot.

---
**on the robot (ssh)
ubuntu@ubuntu:~$ ros2 topic info -v /TB5/cmd_vel
Type: geometry_msgs/msg/Twist

Publisher count: 1

Node name: teleop_twist_joy_node
Node namespace: /TB5
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.91.1c.87.04.93.65.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: motion_control
Node namespace: /TB5
Topic type: geometry_msgs/msg/Twist
Endpoint type: SUBSCRIPTION
GID: 01.0f.40.4d.fa.05.1b.eb.01.00.00.00.00.00.d2.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
---

---
** on the User PC
UserPC@UserPC:~$ ros2 topic info -v /TB5/cmd_vel
Type: geometry_msgs/msg/Twist

Publisher count: 1

Node name: teleop_twist_joy_node
Node namespace: /TB5
Topic type: geometry_msgs/msg/Twist
Endpoint type: PUBLISHER
GID: 01.0f.91.1c.87.04.93.65.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: _ros2cli_2109
Node namespace: /
Topic type: geometry_msgs/msg/Twist
Endpoint type: SUBSCRIPTION
GID: 01.0f.91.1c.3d.08.b5.56.01.00.00.00.00.00.05.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

---

##It sounds that the motion_control node does not subscribe to the /cmd_vel topic on the User PC
[image]

MikeDegany, 2023-07-17:
The issue is that the Create 3 of the robot occasionally fails to establish a connection with the Access Point. Consequently, the 'Connect' tab of its webserver does not display any IP address. How can I troubleshoot and resolve this issue, ensuring consistent connectivity between the Create 3 and the Access Point?

hilary-luo, 2023-07-20:
@MikeDegany having seen your other issue where it was discussed to use namespacing instead of changing domain ID's are you still looking for help to troubleshoot this? If so please provide the [create3 log](https://iroboteducation.github.io/create3_docs/webserver/logs/) from when this happens.

MikeDegany, 2023-07-31:
@hilary-luo  Create3 unable to get IP when connecting to DHCP-enabled hotspot or Access Point. Error logs screenshot attached. Thank you for assistance 
[image]

hilary-luo, 2023-08-03:
Is this happening consistently or is it intermittent? - has it worked with this network before?

closed 2023-09-11 by hilary-luo