turtlebot/turtlebot4 #221: RPi 4 is not connecting with the wife, despote all efforts and going thru tutorial. Once we put the ssh IP in, nothing happens after that.

Muhammadchughtai, 2023-07-17:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04)
 - ROS Distro: (e.g. Galactic)
 - Built from source or installed:
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**
 A clear and concise description of what you expected to happen.

 **Actual behaviour**
 A clear and concise description of what you encountered.

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
Add anything else you thing is important.

Muhammadchughtai, 2023-07-17:
Please provide the following information:

OS: (Ubuntu 22.04)
ROS Distro: (humble)
Built from source or installed: Source
Package version: (if from repository, give version from sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE, if from source, give commit hash)
Expected behaviour
RPi4 to connect as shown in tutorials

Actual behaviour
Not connecting with home (office) wifi at all. Despite using 5Ghz.

Add anything else you thing is important. Tried all tutorials and online instructions.

hilary-luo, 2023-07-17:
Hello, thanks for reaching out. Unfortunately I need more information to try and help. 

1. You said you built the packages from source, what is the hash for the version that you are using?
2. Please tell me the starting condition. Is this a new robot and the first time that you have connected it to wifi? 
3. Which robot version do you have? (lite or standard)
4. Please tell me the exact process you went through (link the specific tutorial) and add details about what worked successfully and where things began to fail.

Muhammadchughtai, 2023-07-18:
1. build the ROS2 package as per instructions from the following tutorial https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html & https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

2. It is a brand new robot and we are trying to connect it to the wifi for the first time.

3. Turtlebot 4 Standard.

4. We followed this tutorial https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html and the official getting started video and it works until we need to connect the raspberry pi to the network. SSH into the Pi works just fine.
Two things happen depending on the method used:
if we follow the video and do "sudo wifi.sh ...." - the command prompt freezes after some time and the Pi doesn't reboot or if it does reboot, the IP address shown on the robot's screen stays the same (10.42.0.1). Turtlebot4 wifi is still on.
if we follow the GitHub tutorial with "turtlebot4-setup" it changes the IP address to Unknown and the turtlebot4 wifi disappears.  
In both results, when we check the devices connected to the chosen network, the Pi doesn't appear. 
We also tried ssh into it with the Ethernet cable but it wouldn't help setting up the wifi. We also flashed a new SD card as per the instructions in the troubleshooting guide but we still get stuck connecting the Pi to the network.

hilary-luo, 2023-07-18:
Thank you for sharing those details. Have you tried doing a full reboot after setting those credentials?

These results sound consistent with incorrect wifi credentials, have you verified these credentials with another device to ensure they work? Additionally, these instructions for connecting to wifi are for wifi systems with basic password authentication. Does your wifi use more advanced authentication? (requiring a username and password or certificate?).

Muhammadchughtai, 2023-07-18:
Yes, we have tried connecting with these credentials using other devices.
We tried rebooting several times.
Our wifi does not require any advance or second authentication. 

We have also tried:
1. Flashing a "normal" Raspberry Pi OS into the RaspberryPi 4 to check if the Pi wasn't faulty and it worked fine. Our assumption is that something in the turtlebot image isn't allowing the Pi to connect to the wifi.
2. We checked via ssh into the Turtlebot4 image on the Raspberry Pi if it could read wifi through _nmci wifi list_ and it would only show the Turtlebot4 wifi none of the other wifis available in our labs. Hence our assumption that something in the Turtlebot image is faulty regarding wifi setup.

Muhammadchughtai, 2023-07-19:
Hello, Is there any ways you are looking forward to help?

hilary-luo, 2023-07-20:
Hi Muhammad, 

Please ssh into the robot and verify the contents of /etc/netplan/50-wifis.yaml to ensure that it matches the intended SSID and password. If everything there looks good to you then please share the contents of that file other than the password.

Also, please note that with the Humble Turtlebot4 packages, only the turtlebot4-setup tool should be used to update the wifi connection. The other script you mentioned is for the galactic packages.

hilary-luo, 2023-08-22:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-08-22 by hilary-luo

judyakl, 2023-10-20:
> Thank you for sharing those details. Have you tried doing a full reboot after setting those credentials?
>
> These results sound consistent with incorrect wifi credentials, have you verified these [...]

@hilary-luo what if the wifi has more advanced authentication? what should we do in this case

hilary-luo, 2023-10-30:
@judyakl If the wifi has more advanced authentication then you need to use discovery server to route all communication through the raspberry pi and then you will need to modify the netplan files manually to connect the raspberry pi for that given authentication method. You should be able to find documentation online about netplan and/or connect with your IT group for specific instructions on how to configure devices.