turtlebot/turtlebot4 #222: Simultaneous Multi-Robot Control with different ROS_DOMAIN_IDs

MikeDegany, 2023-07-17:
I have established connections between multiple Turtlebot4 Standards and an Access Point, linking them to a PC. Currently, I can only observe the topics of each individual robot by adjusting the ROS_DOMAIN_ID environment variable on the PC to match the corresponding parameter on the respective robot. How can I view all the topics simultaneously on the PC, enabling me to control all the robots concurrently?

MikeDegany, 2023-07-17:
When modifying environment variables in separate shell terminals within, they remain separate and do not conflict with each other. So, you can control two robots in two different shells. But I'm not sure how to do so when writing a bash or having a separate launch file to control them simultaneously.

ipa-rar, 2023-07-18:
@MikeDegany I have been working with tb4 for a multirobot use case for a while and I can share some of my experience to use multiple tb4s. 
1. Used a fixed ROS_DOMAIN_ID across all your robots that needs to be in the multi-robot setup including the computer that you use to control them. This would act like a virtual private network for your multirobot system. 
2. The robots should be separated by ROS_DOMAIN_IDs rather each robot should be seperated by using a namespace.
3. The namespace would seperate the topics and services with the namespace thereby not interfering with other robots. For example, Robots with namespace _agent_0_ and _agent_1_ will have its cmd_vel topics like _agent_0/cmd_vel_ and _agent_1/cmd_vel_. So you just have to send your command velocity to the namespaced topic to control a specific robot. 
4. Adding seperate ROS_DOMAIN_ID is necessary only if you want each robot to be isolated from one another, however doing this is not necessary for a multirobot use case. This could be useful when you have multiple robots but each robot has its own PC and none of them should talk to each other.
 
@roni-kreinin Please correct me if there are any mistakes here

hilary-luo, 2023-07-20:
@MikeDegany if this issue addresses your questions / concerns then please close this issue. If not feel free to follow up with more information. 

An additional note, there is documentation on the namespacing available [here](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/multiple_robots.html#namespacing).

closed 2023-08-22 by hilary-luo