turtlebot/turtlebot4 #223: Static IP for multiple Create 3 robots

MikeDegany, 2023-07-18:
I would like to establish a static IP configuration for multiple Create 3 robots. This would ensure consistent and reliable connectivity for each robot. Could you please provide guidance on how to set up a static IP configuration for Create 3 robots?

Jayden-F, 2023-07-18:
Hello @MikeDegany,

If you are using the Humble install, you can set a static IP address by:
- Use SSH to connect to the turtlebot 4 raspberry pi.
- run `turtlebot4-setup' .
- navigate to Wi-Fi Setup.
- Enter the your static IP Address and set DHCP to False.
- Finally save the settings and apply the settings. 

You can find more information in the turtlebot4 documentation.
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html

MikeDegany, 2023-07-18:
This only sets a static IP for RPi. I'd like to set a static IP for Create3.

Jayden-F, 2023-07-18:
Yeah, you're out of luck. The create3 is never a fun time.

m-usama-z, 2024-08-24:
Hi. I've been unable to move the TurtleBot 4 by sending twist messages to /cmd_vel after configuring a static IP address, even though /cmd_vel appears in the topic list. Can anyone kindly advise about this? Thank you very much.

m-usama-z, 2024-08-24:
I figured it out: the robot has to be restarted before use after changing Wi-Fi settings. Would be nice for this to be mentioned in the user manual, and would be even nicer if this, or an alternative solution, were implemented to be executed automatically upon Wi-Fi reconfiguration :)

pvg6, 2025-01-06:
If we set the static ip to raspberrypi ,. it will connect to the access point ssh ubuntu@192.168.185.3

I am unable to connect to ssh ubuntu@192.168.185.3. I want to connect to this to reset my ip how to do it....

@MikeDegany