turtlebot/turtlebot4 #226: Can no longer run SLAM after Create 3 firmware updates

languyen14507, 2023-07-21:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04) Ubuntu 20.04
 - ROS Distro: (e.g. Galactic) Galactic
 - Built from source or installed: installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**
After updating Create 3 firmware to G.4.5, then G.5.1, I tried to run SLAM by:
Opened one terminal and ran "ros2 launch turtlebot4_navigation slam_sync.launch.py"
Opened another terminal and ran "ros2 launch turtlebot4_viz view_robot.launch.py"

Expected to see a map of the turtlebot surroundings on rviz view.

 **Actual behaviour**
Terminal 1 showed these errors:
[sync_slam_toolbox_node-1] [INFO] [1689964833.805125697] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650547252.230 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1689964833.953461740] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1650547252.377 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

Terminal 2 showed these warnings:
[rviz2-1] Warning: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "wheel_drop_left" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp

The rviz view looks empty like this:
[image]

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
I used to run SLAM and Navigation successfully before the Create 3 firmware updates as suggested by @roni-kreinin and @hilary-luo

teufelweich, 2023-09-04:
This seems to be related to the CPU usage issue from the Create3 Base https://github.com/iRobotEducation/create3_docs/issues/381 and results in a similar behavior as we experience in #134 

@languyen14507 why did you close this issue? have you resolved the problem?

languyen14507, 2023-09-08:
@teufelweich 
I still have the problem and did not close the issue myself.

hilary-luo, 2023-09-10:
@languyen14507 I have not responded to this ticket because the status shows that you closed it as completed. I will reopen it and take a look.

hilary-luo, 2023-09-11:
@languyen14507 Can you clarify if this issue is different from https://github.com/turtlebot/turtlebot4/issues/218 ? If it is the same issue then I will re-close this one and you can continue responding on that one. Or, if the other issue is resolved then please close that ticket.

languyen14507, 2023-09-11:
I have decided to upgrade to Humble, therefore I am closing this issue.

closed 2023-09-11 by languyen14507