turtlebot/turtlebot4 #227: Issue with turtlebot 4 lite navigation

ValkyrX, 2023-07-24:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: Installed

 **Expected behaviour**
when following the manual as per https://turtlebot.github.io/turtlebot4-user-manual/tutorials/turtlebot4_navigator.html , i run the command ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py nav2:=true slam:=false localization:=true rviz:=true , when i do this (on my remote pc with all packages installed and can ros2 topic list see all topics from pu and create 3, the simulation loads but i get a screen full of errors stating "[rviz2-53] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist" , i cant then get past this point, i do see the robot in the waregouse on the simulation but i cant move it or undock etc, shoudl this also link to my actual robot or is it just a sim ? is there a way to get this working ?

Thanks
Kurt

 **Actual behaviour**
 A clear and concise description of what you encountered.

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
Add anything else you thing is important.

hilary-luo, 2023-07-24:
Hi Kurt,

This command that you mentioned: 
`ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py nav2:=true slam:=false localization:=true rviz:=true` is specifically for simulating the robot. It will not have anything to do with a physical robot, to make things simple, make sure your robot is off / disonnected when you are doing the simulation. 

To assist in getting the system working can you please add more details about the exact goal, the process you followed and what specific step won't work. You link to the turtlebot4 navigator tutorials page but your description sounds more like you are trying to do the [navigation tutorial](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html). In the Turtlebot4 Navigator examples, the next step is to run another node and you don't mention anything about doing that step. Also please expand on the fact that you cannot undock, how are you trying to initiate the undock step? 

What I can say is that the error that you mentioned is expected to occur until you have the system set up. So you will see it prior to pressing play on gazebo and either setting the initial robot pose on the map manually or running the turtlebot4 navigator tutorial nodes which do it as part of the code.

ValkyrX, 2023-07-24:
Hi, so i am silly i didnt realize that was a simulation only, i dont suppose anyone has a dashboard with commandslike undock and dock with cam view i can take a look at ? im new to ros and turtlebot i can get the navigation running by activating the nodes but would like to be able to use some kind of web dash ?

Thanks

hilary-luo, 2023-07-24:
Quite alright. If you are wanting to work with the physical robot I would encourage you to go through the generating a map and navigation tutorials. This should take you so far as being able to use rviz to tell the robot where to go. 

RViz is the best place to start for any sort of "dashboard" to interact with your system. You can use it to display the information that is being published on different topics including the camera feed. See the video attached on how to view the camera feed:

https://github.com/turtlebot/turtlebot4/assets/103377417/d6c1e069-c7bf-4932-89d8-74de45a729ad

The simplest way to dock or undock is with the physical buttons (on either side of the main button on the create3 base). However, you can also send dock / undock commands by running custom code (from the tutorials discussed below) or [from the command line](https://iroboteducation.github.io/create3_docs/api/docking/#undocking). 

The simulation tools are very useful for learning in a predefined environment so I would still encourage you to go through the turtlebot4 navigator tutorials to understand how the robot is told to dock / undock and navigate autonomously. These commands will still be the same for the physical robot but the locations that the robot is sent to in the examples are not necessarily going to be present or make sense in your actual environment. Once you have your own map of where you are working, you could modify these example scripts to send the robot to coordinates that exist in your map. Then you will be able to run those scripts with your physical robot.

closed 2023-07-25 by ValkyrX