turtlebot/turtlebot4 #228: Powersave issue

ValkyrX, 2023-07-24:
I have noticed that when power saving is activated that when the robot boots the lidar and camera are off, which is good, and they start when undocking, ok so far so good, however when i redock the robot the camera seems to power off but the lidar doesn't stop spinning ? is this expected or should it stop ?

Thanks
Kurt

hilary-luo, 2023-07-27:
Hi Kurt, 

If there is something still subscribed to the lidar /scan topic then it will continue running.

ValkyrX, 2023-07-27:
How do i tell if somethings subscribed, i mean it runs even from a reboot so unless something subscribes on startup then im not sure what it could be

hilary-luo, 2023-08-01:
You can tell if something is subsribed by calling `ros2 topic info /scan`. That being said, I am seeing a similar issue. Can you confirm
1) what robot type you are running (standard vs lite)
2) which turtlebot4-robot software package you are running (e.g. Humble 1.0.1-1jammy.20230728.162203)
3) what version of rplidar-ros that you are running (e.g. 2.1.3-3jammy.20230724.202913)

hilary-luo, 2023-09-11:
@ValkyrX Can you confirm if this issue is resolved in the latest update of the rplidar driver?

closed 2023-10-12 by hilary-luo