turtlebot/turtlebot4 #231: Saving map from Slam not working right if using namespaceing

NachtaktiverHalbaffe, 2023-07-27:
**Please provide the following information:**
 - OS:  Ubuntu 22.04 (Docker)
 - ROS Distro: Humble
 - Built from source or installed: Installed
 - Package version: 1.0.3-1jammy.20230722.003833

 **Expected behaviour**
Saving the map when calling the corresponding service although a namespace is used

 **Actual behaviour**
Map saver fails because it tries to spin the global `/map` topic instead of the namespaced `/<namespace>/map` topic

**To Reproduce**
Provide the steps to reproduce:
1. Run slam with namespace (`ros2 launch turtlebot4-navigation slam.launch.py namespace:=/turtle1`
2. Do the mapping itself until map is finished
3. Run `ros2 service call /turtle2/slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: 'map_name'"` in terminal

**My workaround**
I remapped the `map` topic back to a global `/map` topic in the correpsonding launch file and it worked again. This makes this package work with namespaces, but also has drawbacks e.g. the map itself isnt in the namespace and as a result not isolated in a multi-roboter-environment

hilary-luo, 2023-07-27:
There is a bug with the slam_toolbox, see https://github.com/SteveMacenski/slam_toolbox/issues/607 and https://github.com/SteveMacenski/slam_toolbox/pull/613.

In the meantime, you can run the map_saver_cli directly and apply the namespacing manually:
`ros2 run nav2_map_server map_saver_cli -f "map_name" --ros-args -p map_subscribe_transient_local:=true -r __ns:=/turtle1`

NachtaktiverHalbaffe, 2023-08-04:
Ah perfect, have also looked into slam_toolbox, but not found that particular issue request. Great seeing that this issue is already adressed. The direct use of map saver works in my case as a workaround.

Is it already known when the fixes gets merged? Otherwise maybe a info with this workaround in the [slam tutorials](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html) on the Turtlebot4 Documentation would be heloful for others who also stumble other this issue.

hilary-luo, 2023-08-08:
I made an inquiry on the PR, if they don't then I can look at adding a note. Thanks for the feedback.

closed 2023-08-08 by hilary-luo