turtlebot/turtlebot4 #232: Navigation Issues

ValkyrX, 2023-07-27:
Running ubuntu 22.04 with humble , i have a turtlebot 4 lite, so i have created a map and it starts and seems to know where it is after setting pose, however when i click a nav goal very often the robot will run into the doorway or hit something on the way, i have noticed 2 things , 

1 : the map and the actual scan image seem to go out of sync just a little bit ? so the scan doesnt always exactly match the map, 
2 : i am not sure if this is relevant but on my tb4 lite i have made a parts mounting plate in teh center of the robot, so instead of being directly on top of the robot the lidar and camera are about 8 - 10 inches up ? would that cause issues ? does the tb4 expect the lidar to be i a exact position on the robot ? if yes then how can i change where it thinks the lidar is ?

Any help would be graety appreciated, i also logged another issue that noone has repsonded to regarding the lidar not stopping when docked so if anyone has any ideas about that as well :-) 

Thanks in advance
Kurt

hilary-luo, 2023-07-27:
1. Take a look at the videos in the [turtlebot4 navigator tutorials](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/turtlebot4_navigator.html#watch-navigation-progress-in-rviz). Are the map and actual scan going out of sync more than in those videos?
2. The proper way to do this is to modify the urdf to match your configuration. That being said, nav2 does not use the camera feed and does not take into consideration the height of the lidar. If that is all you changed then it shouldn't be causing these issues. 
The lidar does need to be in the exact same position as viewed from the top though. If the lidar was repositioned towards the front, side or back of the robot then the relative footprint of the robot will not match and will allow for collisions. If this is the case then the urdf definitely needs to be changed.

As a general note, also remember that the robot will only be able to navigate around obstacles that the lidar can see, so they need to cross the plane of the lidar, not be too skinny and not be too reflective or transparent.

If your issue persists then please let me know the turtlebot package version is that you are running.

ValkyrX, 2023-07-27:
so its about the same going out of sync now and then, where is the urdf i need to change and what fields relate to footprint, the lidar is in the middle of the robot  roughly the same as the images online o fwhere it is on a normal turtlebot4 lite. My camera is at the front but of it doesnt use it then that shouldnt be the problem, When the robot is navigating it doesnt seem to follow exactly the path it draws ? could the odom be out ? is there a calibration script i can run to ensure everything is correct ? i notice that i get errors in the diagnostics regarding the imu being out , however speaking to irobot they advised its the correct value and that its the turtlebot4 code thats wrong as they changed the imu threshold ?

ValkyrX, 2023-07-27:
actually in relation to above its actually going out of sync more than the videos, it just doenst seem to see the wall or is ignoring it and trying to drive through it

hilary-luo, 2023-07-31:
>where is the urdf i need to change and what fields relate to footprint

The urdf is here: https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_description/urdf/lite. You will have to build from source and modify this file to match your robot. This will update the transforms between frames and the visualization of the robot in rviz and in sim.  To only account for the change in position of the sensors, you can update the offsets. If you are adding anything else to the robot then there are lots of good resources on how to build urdfs, it is easiest to start with adding simple shapes like a cylinder. 

>When the robot is navigating it doesnt seem to follow exactly the path it draws ? 

The robot will not follow the exact global path that it plans. There are two planners, the global planner and the local planner, both have planned paths that can be visualized in rviz. There are separate sets of parameters that govern each planner. I would encourage you to explore Nav2 documentation regarding this point. 

>i notice that i get errors in the diagnostics regarding the imu being out , however speaking to irobot they advised its the correct value and that its the turtlebot4 code thats wrong as they changed the imu threshold ?

When you reference any error please do give full details about what the error is. I am assuming that you are referring to the imu publishing frequency and if so, that is not of concern. 

> actually in relation to above its actually going out of sync more than the videos, it just doenst seem to see the wall or is ignoring it and trying to drive through it

Can you please provide a screen capture video of rviz to show how it goes "out of sync"? I can't comment without seeing it.

In regards to troubleshooting why it seems to be ignoring obstacles, please first ensure that you are running the latest turtlebot4 software. Next, verify that the local costmap is being populated properly. As an example of what it should look like in rviz:
[image]
[image]
See how there are these blue/pink/purple halos around all of the obstacles? The faint ones are the global costmap (They are faint because Alpha is set to 0.2 in the visualization as shown in the first pic). The dark ones near the robot are the local costmap (left at the defaul Alpha of 0.7). You will need to have the robot near an obstacle to see the local costmap show up. Can you confirm that both costmaps are populated?

closed 2023-08-31 by hilary-luo