turtlebot/turtlebot4 #233: LiDAR cannot be stopped through service

jonaskuResearch, 2023-07-28:
**Please provide the following information:**
 - OS: Ubuntu 22.04.02 LTS
 - ROS Distro: Humble
 - Built from source or installed: installed
 - Package version: ros-humble-turtlebot4-node 1.0.1-1jammy.20230228.204604

 **Expected behaviour**
 When calling the service `/stop_motor`, the motor of the LiDAR should stop, as it did in Galactic.

 **Actual behaviour**
 Instead, the LiDAR continues to turn.

**To Reproduce**
1. Connect to the TurtleBot4 using ssh.
2. Check that the service is available (`ros2 node info /turtlebot4_3/turtlebot4_node` and `ros2 service list` both show the available service `/turtlebot4_3/stop_motor: std_srvs/srv/Empty`).
3. Call the service: `ros2 service call /turtlebot4_3/stop_motor std_srvs/srv/Empty`.

**Other notes**
We have used the setup utility to set the namespace of the robots. The service for starting and stopping the camera works without any issues. Prior to upgrading the robot from Galactic to Humble, the same service (without the namespace) was able to stop and start the LiDAR motor.

jonaskuResearch, 2023-07-28:
We could reproduce the issue also on a brand-new delivered TurtleBot4 that we set up without namespacing. Also in this case, the service call (`ros2 service call /stop_motor std_srvs/srv/Empty`) seems to be executed successfully but the LiDAR keeps turning.

ValkyrX, 2023-07-28:
Its because the diagnostics is running, however when i stopped my diagnostics service the whole robot went mental and i needed to restart it again

hilary-luo, 2023-08-01:
@jonaskuResearch after you send the service call and get the response, can you check `systemctl status turtlebot4.service` on the robot? You should be able to see info output from the service indicating why the stop command did not work. I expect you will see the response (`Ingnoring stop_motor request because rplidar_node is in 'auto standby' mode`). This is because the lidar motor is set up to start and stop based on whether or not there are any subscribers on the `/scan` topic. That being said, there are some people experiencing some issues with the lidar not stopping and that is being looked into. 

If you wish to turn off the auto standby functionality then you can modify the launch file: https://github.com/turtlebot/turtlebot4_robot/blob/3d92a3f2b7ea889e198352322e231a7fab540dcf/turtlebot4_bringup/launch/rplidar.launch.py#L34

hilary-luo, 2023-08-02:
@ValkyrX can you please confirm what version of rplidar-ros that you are running (e.g. 2.1.3-3jammy.20230724.202913)? As I said on your other ticket, I am able to reproduce this issue, if you are pursuing the issue in this ticket then please close your other ticket. 

I have created a ticket on the lidar driver repo, I would encourage you to follow along on it: I would encourage you to follow the issue here: https://github.com/Slamtec/rplidar_ros/issues/129 .

ValkyrX, 2023-08-04:
I am not sure where to find the version m8, its a new install with ros humble thats all i can see

hilary-luo, 2023-08-04:
Please run `apt show ros-humble-rplidar-ros` and share the output.

ValkyrX, 2023-08-04:
ok here you go

Package: ros-humble-rplidar-ros
Version: 2.1.3-3jammy.20230724.202913
Priority: optional
Section: misc
Maintainer: Wang DeYou <deyou.wang@slamtec.com>
Installed-Size: 857 kB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.3.1), libstdc++6 (>= 11), ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-srvs, ros-humble-ros-workspace
Download-Size: 156 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2-testing/ubuntu jammy/main arm64 Packages
Description: The rplidar ros package, support rplidar A1/A2/A3/S1/S2/S3/T1

hilary-luo, 2023-08-09:
@ValkyrX I understand that. Please follow along and feel free to chime in on the ticket that I created on the lidar driver repo regarding this issue https://github.com/Slamtec/rplidar_ros/issues/129.

deyouslamtec, 2023-08-11:
The bug has been resolved, a new version will be released later.

ValkyrX, 2023-08-11:
has thsi been released yet ? its still not working for me

hilary-luo, 2023-08-12:
@deyouslamtec Thank you very much for the fix and the update!

@ValkyrX As DeYou mentioned, a new version will be released later. You can see what versions have been released here: https://github.com/Slamtec/rplidar_ros/tags and use `sudo apt update && sudo apt upgrade` to check for and apply updates when they are available. In the meantime, if you are in a rush before it is released, you are free to build from source.

hilary-luo, 2023-08-28:
@jonaskuResearch & @ValkyrX The fix has been released and as such I am closing this ticket.

closed 2023-08-28 by hilary-luo

dlitvak, 2023-09-11:
Hi,

I'm new.  Is this going to be propagated to ros-humble-rplidar-ros package?
Our version of the pkg:
Package: ros-humble-rplidar-ros
Version: 2.1.3-3jammy.20230807.180936
...

The lidar still doesn't stop.  
Thx

dlitvak, 2023-09-11:
Sorry, I'm not seeing the fix version 2.1.4-1.
We see: ros-humble-rplidar-ros/jammy,now 2.1.3-3jammy.20230807.180936 arm64 [installed,automatic]

hilary-luo, 2023-09-12:
@dlitvak I am not sure why it was not available for some time but I see it again today. Can you confirm that you are able to update now?