turtlebot/turtlebot4 #234: How to factory reset the TB4 when the page is not available anymore

y2d2, 2023-07-28:
Dear, 

**Information**
create3:  version H.2.2
using ros humble 

**In short** 
I somehow managed to change the TB4 so that I can no longer access the create3 platform from the webpage (rPI_IP:8080) moreover the comm and batt light on the create 3 platform stay out after reboot chime. I can however ssh into the rPi4. I have tried several reboots, but those did not change the situation. I believe I did something wrong when I was changing the wired subnet. How can I perform a hard factory reset for the create3 platform either with buttons on the platform or through a command from the rPI? 
I was also wondering what the best strategy is to have 2 turtlebots with a discovery server setup each to connect to 1 single external PC? 

**In long** 
I am trying to setup an experiment with 2 TB4s. I setup the first TB4 to work with a discovery server on the respective rPI of this TB4, However When I tried to do the same with the other TB4 I got an issue since the address `192.168.186.0/24` for the ip route add command was already in use for the IP of the first TB4 rPI. Since I could not directly find out how to solve this I thought to change the subnet of the second  TB4. I saw an option to change this on the page of the create3 platform, I noticed that `192.168.185.0/24` was already in use for the ethernet port so I changed the submask to `192.168.187.0/24`. Additionally in the rPi I changed the `/etc/netplan/40-ethernets.yaml` file to reflect this change. However I think I forgot to change it in the application configuration. At first everything seemed to work fine. 
However now the page to connect to the create3 platform does not load anymore, it keeps on loading continuously without result he comm and batt light on the create 3 platform stay out after reboot chime.
I can still connect to the rPI of this TB4 using ssh and I can see all the topics of the TB4 both on my PC aswell as on the rPI. But I cannot access the create3 platform page to change things, or just reboot the robot or application. (I already rebooted the rPI, and although I am not sure it  seems it does reboot the full platform.) 

Thanks

y2d2, 2023-07-30:
I was able to reset the create3 platform by following the instructions https://edu.irobot.com/create3-setup to setup the create3 as an access point and perform the reset from there and after resetting the `/etc/netplan/40-ethernets.yaml`  and the different options in the `turtlbot4-setup` configuration was able to get back to the start point. 

**Workarround**
I then retried to do the same changing the subnet mask on the create3 and changing the back the `/etc/netplan/40-ethernets.yaml`  and got the same result as not being able to get to the page of the create3 platform. 
when I am sshed into the rPI I can ping create3 platform on its new ip: `192.168.187.2` 
Moreover after installing `links` I was able to get the homepage of the create3 platform using `links 192.168.187.2:80` : 
[image]

So I think maybe there is some network file that has to be updated? ( It seems there is some kind of port forwarding active since from the create3 documentation the standard port is 80 and this has been changed to 8080 in the TB4. (However, I am not so proficient with network setups. So I am not sure.) 

At least for me through 'links'  I can access and change the create3 settings and have a workaround to my issue.

**Open Question**
However, I would still like to know what was the best way to set up a multi TB4 with discovery servers, since using simple discovery would not be very scalable?  (Although I am not sure about this.) 

Cheers, 
Yuri

y2d2, 2023-07-30:
Alrigth after consulting https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html it is clear that a discovery server would be more scalable overtime. 

However, now that I have everything setup it seems I am not able to see the topics of the TB4 set with internal network mask `192.168.187.2/24`, on my controlling PC as I was able to do with the normal setup. I think this is related to the webpage issue? ( Maybe because of the port forwarding issues the server does not see these topics? Although I can see the topics on the rPI.) 

However, I am not sure if this is the best way forward to connect multiple robots together since they would all be running their own server and still not be able to connect with each other.  I could maybe give each of them a different server number and then add them to each individual TB4 in the `$ROS_DICOVERY_SERVER` env variable. This would correspond to the setup: 
[image]
However, this would be similar to having a simple discovery. 

Therefore, I tried to setup the server on my controlling PC. 

- started a discovery server on the controlling PC: ` fastdds discovery -i 0` 
- Added `export ROS_DISCOVERY_SERVER="ip_controlling_PC:11811"` and `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
` in the `~/.bashrc` of the controlling PC
- Disabled the discovery server option in the rPI `turtlebot4-setup` and applied the new settings. 
- Added   `export ROS_DISCOVERY_SERVER="ip_controlling_PC:11811"` in the ~/.bashrc of the rPI and sourced again. 
- Changed the ip of the Discovery server of the create3 to ip_controlling_PC. 
- Rebooted the TB4 and waited for the chime. 
- opened a new terminal on my controlling PC and used `ros2 topic list`  but could only see the `/rosout` and `/parameter-event` topics (even after several repeats. On the rPI I can see all the expected topics of the TB4.  

Not sure what I did wrong, but finally reverted to the simple discovery for this setup. But I fear scalability issue when the number of robots will increase.

y2d2, 2023-07-31:
Using fastdds gives me similar issues for open discovery: 
https://github.com/turtlebot/turtlebot4/issues/129
More specific the /odom topic and the TF were not publishing and not visible on the `ros2 topic list`

So switched to cyclonedds as described in https://github.com/turtlebot/turtlebot4/issues/129. But still the odom topics stop after view minutes. will continue on https://github.com/turtlebot/turtlebot4/issues/129.

y2d2, 2023-07-31:
should have been https://github.com/turtlebot/turtlebot4/issues/132 instead of https://github.com/turtlebot/turtlebot4/issues/129

y2d2, 2023-07-31:
**To many nodes**
From https://github.com/turtlebot/turtlebot4/issues/132 I think the issue is the number of nodes is too high when I also run full localization and navigation stack for both TBs.: 

`Jul 31 13:49:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811372.963887466] [tb1.system_health]: CPU usage: max 99 [%] mean 93 [%] RAM usage: 39/59 [MB]
Jul 31 13:49:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811372.974874518] [tb1.system_health]: Discovered 78 nodes! Considering using DOMAIN_ID to reduce the burden on the application
Jul 31 13:49:49 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.002503 is too high, ignoring
Jul 31 13:50:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811432.968177493] [tb1.system_health]: CPU usage: max 99 [%] mean 98 [%] RAM usage: 39/59 [MB]
Jul 31 13:50:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811432.983602297] [tb1.system_health]: Discovered 75 nodes! Considering using DOMAIN_ID to reduce the burden on the application
Jul 31 13:51:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811492.968680213] [tb1.system_health]: CPU usage: max 94 [%] mean 83 [%] RAM usage: 39/59 [MB]
Jul 31 13:52:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811552.972187252] [tb1.system_health]: CPU usage: max 100 [%] mean 91 [%] RAM usage: 39/59 [MB]
Jul 31 13:52:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811552.976669089] [tb1.system_health]: Discovered 76 nodes! Considering using DOMAIN_ID to reduce the burden on the application
Jul 31 13:53:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811612.973344389] [tb1.system_health]: CPU usage: max 95 [%] mean 91 [%] RAM usage: 39/59 [MB]
Jul 31 13:53:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811612.978596384] [tb1.system_health]: Discovered 80 nodes! Considering using DOMAIN_ID to reduce the burden on the application
Jul 31 13:54:07 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice ntpd: Script called in periodic mode`

The simple discovery with cycloneDDS works as long as you do not do too much additional stuff. 

**Using the discovery server** 
The issue that happened is that when you disable the discovery server on the rPI it automatically removes this in the create3 settings, which I changed before to change it on the rPI. So Once I noticed this and reset the discovery server on the create3 with the IP of the controlling PC it worked as well. (Had to also change the /etc/turtlebot4/fastdds_discovery_super_client.xml, to reflect the correct IP of the controlling PC on the rPI.)
Unfortunately, the same issue was raised with the number of nodes.   

**Conclusion and questions**
My conclusion is that the TB4s are just not designed to work for multi-robot applications that require close cooperation. I also think a lot of nodes are made that do not really contribute to the setup. I am looking at all the internal nodes. What is their purpose? Can they be suppressed? 
Would it be possible to have some disabled or on-demand enabled? Can we publish some nodes in a different ROS_DOMAIN_ID.  For example, the nodes needed for close cooperation are published in an additional ROS_DOMAIN_ID, while also being published in the ROS_DOMAIN_ID of the TB4 (rPI and create3 platform) such that it keeps on working, but some information can be shared between agents without overloading them?

**Strange behavior with fastDDS server** 
It seems when I run the discovery server the ROS_DOMAIN_ID has no influence. I changed it on one TB to see if then at least I could have both of them work independently, but I was able to see all the nodes and all the agents. Not sure if this is expected behavior.

roni-kreinin, 2023-08-01:
Regarding accessing the webserver once you have changed the `usb0` subnet, the webserver gets forwarded through the Pi using `socat` with [this] systemd service. It is hardcoded to `192.168.186.2`, so you will need to modify this file at `/etc/systemd/system/webserver.service` and change the subnet. You can then call `sudo systemctl restart webserver.service` to apply changes.

Regarding multiple robots with discovery server, we do not currently support it but it is being worked on. The changes needed are specified in https://github.com/turtlebot/turtlebot4/issues/145.

There are a couple of options you have for setting up discovery server with two robots. It will depend on your application:

1. Have one discovery server that all robots communicate with. This would likely be your PC and is the same setup that you have tried. I think the main issue with the way you set up the robots is that you only `export ROS_DISCOVERY_SERVER="ip_controlling_PC:11811"` in the `~/.bashrc` file. You need to add that line to the `/etc/turtlebot4/setup.bash` file for each robot as well. Only then will the `turtlebot4.service` systemd job be affected. You will also have to access the Create 3 webserver and set the same discovery server there.
2. Have each robot run its own discovery server and configure your PC to communicate with both servers. The robots will not be able to see each others topics/nodes/etc., but your PC can. Then you can have a node on your PC that would communicate with each robot. For this option, you can set up each turtlebot's discovery server using the `turtlebot4-setup` tool. On your PC you will need to add ip routes for each robot, and export both discovery servers like this: `export ROS_DISCOVERY_SERVER="192.168.186.3:11811;192.168.187.3:11811"`.

>  I also think a lot of nodes are made that do not really contribute to the setup. I am looking at all the internal nodes. What is their purpose? Can they be suppressed?

The internal nodes run on the Create 3 and their functions are to run the Create 3 reflexes and sensors. They are required and cannot be disabled. The main node you can disable is the diagnostics node running on the RPi4. There is an option in the `turtlebot4-setup` tool to disable that.

y2d2, 2023-08-01:
Hi Roni! 

Thanks for your reply and help, much appreciated!

- **1) discovery server on the controlling PC**
Yes, I did update these files on both robots and as I mentioned in the previous comment it actually works. *The thing that actually made it fail was that every time I changed something in the `turtlebot4-setup` (like the domain or the namespace) it would actually also overwrite the discovery server enable setting on the create3, once I figured this out and changed the setting back on the create3 after each change in the `turtlebot4-setup` it worked.* But I ran into the too many nodes issue similar to the simple discovery.  

- **2) discovery servers on the the TBs**
Indeed that was my original way of thinking. Since I believe this is the most scalable approach. But I missed the webserver.service to make it work. Thanks for that I will try out as soon as I can and report back on this. However in this scenario the namespace have to work, which I ran into an issue with for the oakd (that I need for my experiments) see https://github.com/turtlebot/turtlebot4/issues/236.   

Thanks for the tip on the diagnostics! 

Cheers,

Yuri

closed 2023-08-01 by y2d2

y2d2, 2023-08-02:
Hi Roni, 

So trying the discovery servers out:  
- create3 platform factory reset and reinstalled the H2.2 update
- reflashed the RPi card and applied `sudo apt-get update && sudo apt-get full-upgrade`  
- I did add a wifi connection to the create3 platform and tested out the simple discovery using fastDDS. Everything worked well.
- Changed the subnet of the create 3 to 211. 
- Updated the `/etc/netplan/40-ethernets.yaml` and the `/etc/systemd/system/webserver.service` to correspond to these changes.
- rebooted the robot
- Could access the webpage after reboot and ping the create3 from the RPi with `ping 192.168.211.2`. So far so good. 
- went into 'turtlebot4-setup' and enabled the discovery server. 
- Applied the changes gave me this error: ` % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:02:10 --:--:--     0
curl: (28) Failed to connect to 192.168.186.2 port 80 after 130108 ms: Connection timed out`
- This seems to come from https://github.com/turtlebot/turtlebot4_setup/blob/humble/turtlebot4_setup/turtlebot4_setup where the connection to the create3 platform is hardcoded at line 167 and 173 in the apply_ros_settings method: 
```
def apply_ros_settings(self):
        reinstall_job = False
        update_create3 = False

        # If one of Domain ID, Namespace, or RMW was changed, apply changes to Create 3
        for option in self.get_settings_diff(BashOptions):
            if option in [BashOptions.DOMAIN_ID, BashOptions.NAMESPACE, BashOptions.RMW]:
                update_create3 = True
            reinstall_job = True

        if len(self.get_settings_diff(DiscoveryOptions)) > 0:
            update_create3 = True
            reinstall_job = True

        for option in self.get_settings_diff(SystemOptions):
            if option is SystemOptions.MODEL:
                reinstall_job = True

        if update_create3:
            ros_domain_id = 'ros_domain_id=' + os.environ[BashOptions.DOMAIN_ID]
            ros_namespace = '&ros_namespace=' + os.environ[BashOptions.NAMESPACE]
            rmw_implementation = '&rmw_implementation=' + os.environ[BashOptions.RMW]

            if self.conf.get(DiscoveryOptions.IP) == '127.0.0.1':
                discovery_server = '&fast_discovery_server_value=192.168.186.3:' + self.conf.get(DiscoveryOptions.PORT)
            else:
                discovery_server = '&fast_discovery_server_value=' + os.environ[BashOptions.DISCOVERY_SERVER]

            if self.conf.get(DiscoveryOptions.ENABLED):
                discovery_server_enabled = '&fast_discovery_server_enabled'
            else:
                discovery_server_enabled = ''

            command = shlex.split(
                'curl -d \"{0}{1}{2}{3}{4}\"'.format(ros_domain_id,
                                                     ros_namespace,
                                                     rmw_implementation,
                                                     discovery_server,
                                                     discovery_server_enabled)) + \
                shlex.split('-X POST http://192.168.186.2/ros-config-save-main -o /dev/null')

            #print(command)
            subprocess.run(command)

            subprocess.run(shlex.split(
                'curl -X POST http://192.168.186.2/api/reboot'))

        if reinstall_job:
            self.ros.robot_upstart_menu.install()
            self.ros.robot_upstart_menu.start()
``` 

- However, except for that it seems all the settings have been applied correctly on the RPi side and I manually changed the settings to correspond to this on the create3 platform. 
- applied the ip route on the Controlling PC and added the `ROS_DISCOVERY_SERVER=ip_RPi:"11811"`  as well as the  `FASTRTPS_DEFAULT_PROFILES_FILE=fastdds_discovery_super_client.xml` Adpated with the correct ip.
- run `ros2 launch turtlebot4-navigation localisation.launch.py map:=own_map.yaml` and this worked!

Thanks for the webserver.service lead! 

I will now add the other one but using the default 186 subnet mask and report back here. (Not sure if necessary but maybe it can help others in the future.)