turtlebot/turtlebot4 #235: Unable to teleop turtlebot4

abhinavsagar, 2023-07-31:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04) - Ubuntu 20.04
 - ROS Distro: (e.g. Galactic) - Galactic
 - Built from source or installed: - Installed

 **Expected behaviour**

Turtlebot teleop using keyboard

 **Actual behaviour**

Unable to teleop turtlebot4 usingf keyboard

**To Reproduce**

On running:

ros2 run teleop_twist_keyboard teleop_twist_keyboard

I'm getting:

This node takes keypresses from the keyboard and publishes them
as Twist messages. It works best with a US keyboard layout.
---------------------------
Moving around:
   u    i    o
   j    k    l
   m    ,    .

For Holonomic mode (strafing), hold down the shift key:
---------------------------
   U    I    O
   J    K    L
   M    <    >

t : up (+z)
b : down (-z)

anything else : stop

q/z : increase/decrease max speeds by 10%
w/x : increase/decrease only linear speed by 10%
e/c : increase/decrease only angular speed by 10%

CTRL-C to quit

But the turtlebot is not moving.

closed 2023-07-31 by abhinavsagar

hilary-luo, 2023-08-03:
Are you able to see the turtlebot topics from this PC? (`ros2 topic list`). If so please share the list here.

abhinavsagar, 2023-08-03:
The problem is solved now, thanks.