turtlebot/turtlebot4 #236: Using the localisation and navigation stack with 2 TB4s in simple discovery and cycloneDDS fails after few seconds.

y2d2, 2023-07-31:
**Please provide the following information:**
 - OS: ubuntu22.04
 - ROS Distro: Humble
 - Built from source or installed: installed 
 - Package version: I suppose that turtlebot4-navigation. (although there is nothing wrong with it specifically) Version: 1.0.3-1jammy.20230722.003833 (Also in the template there is written: `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, but it should be `sudo dpkg -s ros-$ROS_DISTRO-turtlebot4-PACKAGE_WITH_ISSUE` ? at least so it is on my system.)

 **Expected behaviour**
When I connect 2 TBs using https://github.com/turtlebot/turtlebot4/issues/132 and run the localization and navigation stack for both of them I expect the TBs to work. 

 **Actual behaviour**
As long as the localization and navigation of neither is running everything seems fine. The moment I start one of them most of the time it works, however, I noticed that the /odom topic and others of the other TB start to stop providing updates. The moment I start this localization and navigation of the second TB it is unreactive and is waiting for the /odom to /base_link transformation: `[controller_server-1] [INFO] [1690835508.615078609] [tb2.local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist`
 Sometimes the other robots' navigation will also stop with the same issue or pause for long times with this message before continuing.  
The create3 log gives the following info:
`[tb1.system_health]: Discovered 78 nodes! Considering using DOMAIN_ID to reduce the burden on the application Jul 31 13:49:49 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.002503 is too high, ignoring Jul 31 13:50:32 iRobot-623CE57328134F83A5B4314DF6EBB89D user.notice create-platform: [INFO] [1690811432.968177493] [tb1.system_health]: CPU usage: max 99 [%] mean 98 [%] RAM usage: 39/59 `

which makes me believe that the create3 platform cannot handle that many nodes.  See  https://github.com/turtlebot/turtlebot4/issues/234 for more details of what I tried to get it to work.  

**To Reproduce**
Provide the steps to reproduce:
1. follow the instruction of https://github.com/turtlebot/turtlebot4/issues/132  to run the robots in simple discovery using the cycloneDDS 
2. run the localization stack `ros2 launch turtlebot4_navigation localization.launch.py namespace:=tb1 map=:./map.yaml` with ./map.yaml a custom build map, but can be any map,  and the navigation stack `ros2 launch turtlebot4_navigation nav2.launch.py namespace:=tb1` 
3. do the same bot for the other TB with namespace:=tb2 

**Other notes**
I made a more detailed description (https://github.com/turtlebot/turtlebot4/issues/234) of the things I tried to make it work but did not make it a bug report and cannot figure out how to change it.

y2d2, 2023-08-01:
I gave a wrong error message in the previous log it gave the following error message: 
` Could not transform base_link to odom: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees..`  

anyways. Since I did 'hack away` a lot the recent days, I decided to start fresh with 1 robot today: 
- Factory reset of the create3 platform and reinstalled the H2.2 update 
-  Flashed the rPi with the image `turtlebot4_standard_humble_1.0.0.zip` I downloaded on the 10th of July and resized the sd card.
- updated the rPi `sudo apt-get update && sudo apt-get upgrade -y` 
- Somebackages were kept back so I decided to also perform a `sudo apt-get full-upgrade -y' (Not sure if this was needed, I will test tomorrow with the other robot what the difference is and which packages were kept back and report back here. I did see a lot of depth-ai stuff, which could explain the OAKD issue covered in this message.)
- Put it back in the robot and followed the instruction in the user manual to setup the robot in simple discovery using the fastDDS middleware. 
- Tried out the 'ros2 topic list' on my controlling PC and this looked good. 
- However when I tried to run the localisation node `ros2 launch turtlebot4_navigation localization.launch.py map=:./map.yaml` with ./map.yaml a custom build map I directly got the message ` Could not transform base_link to odom: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees..`  
- This seemed new. The odom base_link transform in the past week has always been a bit shaky, but normally it comes through after some time, might drop some frames in the single agent setup. 
- Stopped everything and rebooted the TB4 
- Waited a bit used 'ros2 run tf2_tools view_frames` and got indeed a TF tree without odom frame: 
[image]
- The /odom topic however publishing as normal `ros2 topic echo /odom` as normal stayed blank for some time but then comes through and published the odom. However the TF tree remained the same. 
- It tried to broadcast the TF between base_link and odom my self using the `/odom` topic. 

```
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node

from tf2_ros import TransformBroadcaster
from geometry_msgs.msg import TransformStamped
from nav_msgs.msg import Odometry

class FramePublisher(Node):
    def __init__(self):
        super().__init__('tf_odom_base_link')
        self.subscription = self.create_subscription(Odometry, "odom", self.update_o_bl_tf, 10)

    def update_o_bl_tf(self, msg):
        t = TransformStamped()
        t.header.stamp = self.get_clock().now().to_msg()
        t.header.frame_id = 'odom'
        t.child_frame_id = 'base_link'
        t.transform.translation.x = msg.pose.pose.position.x
        t.transform.translation.y = msg.pose.pose.position.y
        t.transform.translation.z = msg.pose.pose.position.z

        t.transform.rotation.x = msg.pose.pose.orientation.x
        t.transform.rotation.y = msg.pose.pose.orientation.y
        t.transform.rotation.z = msg.pose.pose.orientation.z
        t.transform.rotation.w = msg.pose.pose.orientation.w
       
       self.tf_broadcaster.sendTransform(t)

def main(args=None):
    rclpy.init(args=args)
    node = FramePublisher()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()
```
- This however gave the following message: `[WARN] [1690903785.216893707] [tf_map_odom]: New publisher discovered on topic 'odom', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY`
- It seemed to come from bad QoS parameters for the `/odom` topic (Maybe this is the reason why in the past the `odom base_link tf` was not very stable and now maybe through some update is not working at all?  (However, this seems like a create3 issue I suppose?)
- Anyway update the code with: 
```
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node

from tf2_ros import TransformBroadcaster
from geometry_msgs.msg import TransformStamped
from nav_msgs.msg import Odometry

from rclpy.qos import QoSProfile, QoSReliabilityPolicy, QoSHistoryPolicy

class FramePublisher(Node):
[... 264 lines cut ...]
/tb1/ip
/tb1/ir_intensity
/tb1/ir_opcode
/tb1/joint_states
/tb1/joy
/tb1/joy/set_feedback
/tb1/kidnap_status
/tb1/mobility_monitor/transition_event
/tb1/mouse
/tb1/odom
/tb1/robot_description
/tb1/robot_state/transition_event
/tb1/scan
/tb1/slip_status
/tb1/static_transform/transition_event
/tb1/stop_status
/tb1/tf
/tb1/tf_static
/tb1/wheel_status
/tb1/wheel_ticks
/tb1/wheel_vels
-------------------------------------------------------------------------
```

Tomorrow I will take note of the versions of the distribution and do the same without the `sudo apt-get full-upgrade -y' with the other TB4 and report on the results.

y2d2, 2023-08-02:
**On the OAKD issue** 
Actually, it was a none issue. The robot was docked when I was using the namespace and you guys are probably tired of explaining https://github.com/turtlebot/turtlebot4/issues/193 that when the TB is docked the camera switches off. I apologize for this. 

**Odom baseline tf with namespace** 
The workaround provided in the previous comment seems to work well when using a namespace as well. 

I am currently updating the other TB similar to the previous comment and  will report the different steps here (although I am not sure if they are still relevant.)

y2d2, 2023-08-02:
**Concerning the RPi upgrade**
It seems everything works now on both TBs without the need of the workaround. (Maybe I needed an additional reboot?) I'm wondering however which node is actually publishing the odom base_link transform? I thought it would be a node on the create3, but seems like it is the AMCL which does it? ' 

Anyways here are the different outputs regarding the upgrade and full-upgrade, which apparently did not make much difference.

The output of `sudo apt-get upgrade` on a freshly flashed RPi: 
```
sudo apt-get upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  linux-headers-raspi linux-image-raspi linux-raspi ros-humble-depthai-examples ros-humble-depthai-ros-driver ros-humble-navigation2 ros-humble-ros2-controllers ros-humble-turtlebot4-description ufw
The following packages will be upgraded:
  alsa-ucm-conf apparmor apport apt apt-utils bind9-dnsutils bind9-host bind9-libs binutils binutils-aarch64-linux-gnu binutils-common ca-certificates cloud-init cpp-11 curl distro-info-data
  dnsmasq-base dpkg dpkg-dev flash-kernel fwupd-signed g++-11 gcc-11 gcc-11-base gcc-12-base gfortran-11 ghostscript git git-man graphicsmagick-libmagick-dev-compat initramfs-tools initramfs-tools-bin
  initramfs-tools-core iptables krb5-multidev libapparmor1 libapt-pkg6.0 libasan6 libatomic1 libavahi-client3 libavahi-common-data libavahi-common3 libbinutils libcap2 libcap2-bin libcc1-0 libctf-nobfd0
  libctf0 libcups2 libcurl3-gnutls libcurl4 libcurl4-openssl-dev libdpkg-perl libegl-mesa0 libegl1-mesa-dev libfreetype-dev libfreetype6 libfreetype6-dev libfwupd2 libfwupdplugin5 libgbm-dev libgbm1
  libgcc-11-dev libgcc-s1 libgfortran-11-dev libgfortran5 libgl1-mesa-dev libgl1-mesa-dri libglapi-mesa libglib2.0-0 libglib2.0-bin libglib2.0-data libglib2.0-dev libglib2.0-dev-bin libglx-mesa0
  libgomp1 libgraphics-magick-perl libgraphicsmagick++-q16-12 libgraphicsmagick++1-dev libgraphicsmagick-q16-3 libgraphicsmagick1-dev libgs9 libgs9-common libgssapi-krb5-2 libgssrpc4 libhwasan0 libidn12
  libinput-bin libinput10 libip4tc2 libip6tc2 libitm1 libk5crypto3 libkadm5clnt-mit12 libkadm5srv-mit12 libkdb5-10 libkrb5-3 libkrb5-dev libkrb5support0 libldap-2.5-0 libldap-common libllvm15 liblsan0
  libmm-glib0 libmysqlclient21 libncurses-dev libncurses5-dev libncurses6 libncursesw6 libnetplan0 libnss-systemd libpam-cap libpam-systemd libpcsclite1 libperl5.34 libpq5 libprotobuf23 libpulse-dev
  libpulse-mainloop-glib0 libpulse0 libpython3.10 libpython3.10-dev libpython3.10-minimal libpython3.10-stdlib librsvg2-2 librsvg2-common libsasl2-2 libsasl2-modules libsasl2-modules-db libssh-4
  libssh-gcrypt-4 libssl-dev libssl3 libstdc++-11-dev libstdc++6 libsystemd0 libtiff-dev libtiff5 libtiffxx5 libtinfo6 libtsan0 libubsan1 libudev-dev libudev1 libwebp-dev libwebp7 libwebpdemux2
  libwebpmux3 libx11-6 libx11-data libx11-dev libx11-xcb1 libxml2 libxml2-dev libxml2-utils libxtables12 linux-firmware linux-libc-dev mdadm mesa-va-drivers mesa-vdpau-drivers mesa-vulkan-drivers
  modemmanager ncurses-base ncurses-bin ncurses-term netplan.io open-vm-tools openssh-client openssh-server openssh-sftp-server openssl perl perl-base perl-modules-5.34 python3-apport python3-debian
  python3-distutils python3-gdbm python3-lib2to3 python3-problem-report python3-requests python3-rosdep python3-rosdep-modules python3-rospkg-modules python3-software-properties python3-tz python3.10
  python3.10-dev python3.10-minimal ros-humble-ackermann-msgs ros-humble-action-msgs ros-humble-actionlib-msgs ros-humble-admittance-controller ros-humble-ament-cmake ros-humble-ament-cmake-auto
  ros-humble-ament-cmake-copyright ros-humble-ament-cmake-core ros-humble-ament-cmake-cppcheck ros-humble-ament-cmake-cpplint ros-humble-ament-cmake-export-definitions
  ros-humble-ament-cmake-export-dependencies ros-humble-ament-cmake-export-include-directories ros-humble-ament-cmake-export-interfaces ros-humble-ament-cmake-export-libraries
  ros-humble-ament-cmake-export-link-flags ros-humble-ament-cmake-export-targets ros-humble-ament-cmake-flake8 ros-humble-ament-cmake-gen-version-h ros-humble-ament-cmake-gmock
  ros-humble-ament-cmake-gtest ros-humble-ament-cmake-include-directories ros-humble-ament-cmake-libraries ros-humble-ament-cmake-lint-cmake ros-humble-ament-cmake-pep257 ros-humble-ament-cmake-pytest
  ros-humble-ament-cmake-python ros-humble-ament-cmake-ros ros-humble-ament-cmake-target-dependencies ros-humble-ament-cmake-test ros-humble-ament-cmake-uncrustify ros-humble-ament-cmake-version
  ros-humble-ament-cmake-xmllint ros-humble-ament-copyright ros-humble-ament-cppcheck ros-humble-ament-cpplint ros-humble-ament-flake8 ros-humble-ament-index-cpp ros-humble-ament-index-python
  ros-humble-ament-lint ros-humble-ament-lint-auto ros-humble-ament-lint-cmake ros-humble-ament-lint-common ros-humble-ament-pep257 ros-humble-ament-uncrustify ros-humble-ament-xmllint ros-humble-angles
  ros-humble-backward-ros ros-humble-behaviortree-cpp-v3 ros-humble-bond ros-humble-bondcpp ros-humble-builtin-interfaces ros-humble-camera-calibration ros-humble-camera-calibration-parsers
  ros-humble-camera-info-manager ros-humble-class-loader ros-humble-common-interfaces ros-humble-composition-interfaces ros-humble-compressed-depth-image-transport ros-humble-compressed-image-transport
[... cut ...]
After this operation, 201 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

- After the upgrade the version of different packages I thought were interesting are: 
 
[image]

- I rebooted the TB  and ran again `ros2 launch turtlebot4_navigation localization.launch.py map:=/sharedDrive/map/map_name.yaml` which worked as well. 

So although I am not sure what went wrong everything seems to work as expected using the fastDDS, I will not try out the cycloneDDS again so for me all is good.

y2d2, 2023-08-02:
sorry forgot to push the 'close with comment' button.

closed 2023-08-02 by y2d2

MikeDegany, 2023-08-08:
Dear @y2d2 
How did you upgrade your packages? And how can I check the version of all turtlebot4 packages I have? 
I did ```sudo apt-get upgrade``` and ```sudo apt-get upgrade``` but the version of my packages are still older than yours. Thanks
[image]