turtlebot/turtlebot4 #239: Battery issue

halimimaha, 2023-08-04:
### Please consider the following information :

OS: Ubuntu 22.04 LTS
ROS Distro: ROS2 humble

The robot is full battery, we can see 100% on the robot screen,

**Expected behaviour:**
Navigate or control the robot with a driving method in a very normal way.

**Actual behaviour :**
The robot turns off 15/20 seconds after moving it with the joystick.

**Other notes :** 
When the robot is not in the namespace and there is no communication between TB4 and Create3 (only three LEDs are ON: MTR, WIFI, POWER), the robot does not turn off and remains on permanently.

Thank you in advance for your help !!

MikeDegany, 2023-08-08:
In terms of battery life, it's advisable not to leave the Turtlebot charging for extended periods. Regarding the three LEDs being on, this might be unconventional, but it's a common occurrence with Turtlebots.

If the robot remains unresponsive due to a red or yellow lightring, you can hold the power button for 20-30 seconds.

hilary-luo, 2023-08-08:
@halimimaha with the robot powered on and docked, can you please check and sharethe battery level in the [webserver](https://iroboteducation.github.io/create3_docs/webserver/about/) and download and share the [create3 logs](https://iroboteducation.github.io/create3_docs/webserver/logs/).

halimimaha, 2023-08-09:
Thanks for your reply, so for the battery level in the webserver : 
file:///home/robotics/Pictures/Screenshots/Screenshot%20from%202023-08-09%2008-42-11.png

And for logs : 
file:///home/robotics/Downloads/messages.txt

hilary-luo, 2023-09-11:
@halimimaha I just realized that I didn't get back to you again, I apologize, I must have missed your response. Are you still experiencing this issue? I can continue troubleshooting it with you

halimimaha, 2023-09-14:
@hilary-luo unfortunately I have not yet fixed the battery problem which is a bit surprising. we have 4 robots and I have this problem on one of the 4 robots, that's why I gave up trying to find a solution on the broken robot. do you have any suggestions ?

hilary-luo, 2023-09-14:
Have you tried doing a factory reset on the create3 and reinstalling the firmware? There are a lot of messages in your log that I don't usually see. If you do go about that can you provide another create3 log after the fact?

When you say that the robot turns off, do all lights on the robot turn off or just some? and do you have to put it back on the dock to turn it back on? or does it reboot on its own?

closed 2023-10-12 by hilary-luo