turtlebot/turtlebot4 #241: wifi name of turtlebot4 change to clear6b and turtlebot wifi didn't apper at all

Hanbali11, 2023-08-05:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04
 - ROS : (e.g. humble)
 - Built from installed:
 - 
 **Expected behaviour**
so basically I was working to connect the robot to my computer and it connected successfully after that I try to send the wifi and the name of the main wifi to the robot to connect both robot and laptop to the same wifi after sending the information to the robot I didn't know if he receives it or not so I try to do that after switching it off and try again so the wireless name in the robot change from 'turtlebot4'
to 'clear6bbr'  
note that before sending the information I was able to see the data of the robot but after switching it off the robot change to another wifi name as I mentioned   
 

 **Actual behaviour**
 A clear and concise description of what you encountered.

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
Add anything else you thing is important.

MikeDegany, 2023-08-08:
I'm not entirely sure I've captured your problem accurately, but it appears that you might have initially connected to the Turtlebot on access point mode, modified the Wi-Fi configurations, and lost the connection. One potential solution is to use an Ethernet cable to SSH into the Raspberry Pi.

hilary-luo, 2023-08-08:
Hi @Hanbali11,

My understanding is that you are running humble and thus used the turtlebot4-setup tool to configure your wifi credentials. When doing this, are you sure that you changed the robot from access point mode to client mode? If you left the robot in access point mode but changed the SSID and password, then the turtlebot will continue to broadcast its own network and adopt the new ssid and password that you entered. This could cause problems if there is already a network with this SSID.

Please connect to the turtlebot using the [recovery instructions](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi). Launch the turtlebot4-setup tool again and ensure that the Wi-Fi Setup settings show the mode as client
[image]

If you change any settings then ensure to apply the settings.

Hanbali11, 2023-08-08:
I  did what the recovery instruction say ,but the result came like ''$ ssh ubuntu@192.168.185.3
ssh: connect to host 192.168.185.3 port 22: Connection refused''

hilary-luo, 2023-08-17:
@Hanbali11 Can you restart by [re-imaging the raspberry pi](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image). Once you have it up and running again, ensure to follow the wifi set up instructions that correspond with the ros-distro that you are running (humble or galactic). As you run through this keep very detailed information on exactly what you do / run and then if you run into any issues give me the details here (include the name of which image you use and specific terminal outputs at each step). 

It is possible that the origin of your issues was running the wrong wifi set up instructions for your ROS distro.

closed 2023-08-27 by Hanbali11