turtlebot/turtlebot4 #242: Add tutorial for Navigating While Mapping (SLAM)

abhinavsagar, 2023-08-07:
It would be helpful to have a tutorial for Navigating While Mapping (SLAM) like the one on the nav2 documentation using turtlebot3 [https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html](https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html).

MikeDegany, 2023-08-15:
I did something similar and it worked: 

Run Nav2: 
```ros2 launch turtlebot4_navigation nav2.launch.py namespace:=/<namespace1>```

Run SLAM: 
```ros2 launch turtlebot4_navigation slam.launch.py namespace:=/<namespace1>```

See the map on Rviz: 
```ros2 launch turtlebot4_viz view_robot.launch.py namespace:=/<namespace1>```

hilary-luo, 2023-08-16:
I agree that this is a nice to have. If someone wants to take this on feel free to create a PR on https://github.com/turtlebot/turtlebot4-user-manual, following the format of the existing tutorials. At this time it isn't a planned priority.

halimimaha, 2024-01-23:
@MikeDegany @hilary-luo  please i have a question, this steps is for one robot, if i have 2 or 3 robots how can i launch it ? 

thanks in advance

halimimaha, 2024-01-30:
@hilary-luo I already have namespace the robots but the problem is that the ros domain id seems saturated, I cannot launch more than one robot in the same ros domain id :/

ajjaya18, 2024-03-02:
Hi, can you Help me with turtlebot4  with generating a map?

ajjaya18, 2024-03-02:
"When I try to map in the world 'depot', my Turtlebot 4 doesn't recognize the objects in the world, but I don't understand why this is happening because in the tutorial, there's already a pre-made map of the 'depot' world, and I want to do the same, but it's not working. Any ideas why this is happening?"

ajjaya18, 2024-03-02:
@hilary-luo

hilary-luo, 2024-03-07:
@ajjaya18 Please open your own ticket with all of the details about your system / setup. The info you have provided here isn't enough to really help you off of. Include the system info that it asks for as well as the specific commands you are running, the terminal outputs you get and images of what you are seeing in rviz if it is relevant. I assume that you are already going through the [tutorial for generating a map](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html).

ajjaya18, 2024-03-07:
ros2 launch turtlebot4_viz view_robot.launch.py, I run this command in the terminal 
[image]
"Another thing, as you can see in the images, my robot doesn't recognize my depot world items, and I don't understand what I'm missing because there is already a depot map in the files that are in the tutorial. If you can help me, it would be of great help because also, look in rviz, it doesn't create anything on the map, and then I don't know what the error could be."

ajjaya18, 2024-03-07:
@hilary-luo help me, please

ajjaya18, 2024-03-07:
I use ros galactic

hilary-luo, 2024-03-07:
@ajjaya18 please open your own ticket and fill the information out there.