turtlebot/turtlebot4 #246: Navigation/Robot does not want to work

nidia-bucarelli, 2023-08-14:
The Robot/navigation of my TB4 does not want to work.
The robot shows different problems every time I try to run the navigation. So, i am not sure what/how to report. 
Sometimes it just turns off or goes with the red light (like in dead mode), sometimes it asks if I have set an initial pose (which i have set using the 2D pose estimate in Rviz) and once I type Yes it goes Off or sometimes it loses communication with the pi4 and says: `timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist`.

The robot shows up in the map. Both the base and pi4 are connected to the internet. Also, I can see the topics.
I am using Humble and TB4 standard version.

The log from iCreate is here: https://drive.google.com/file/d/1-seGlZHm5GcH5XuEHu4gD8aGChS7JIVN/view?usp=sharing
I would appreciate your help in debugging this issue.

hilary-luo, 2023-08-16:
Please provide the following information:
OS: 
ROS Distro: 
Built from source or installed: 
Package version: 
Model: (standard or lite)
Networking Configuration & DDS vendor: (Simple Discovery or discovery server and Cyclone or FastDDS)

Additionally, are there any other ROS devices on the same network?

nidia-bucarelli, 2023-08-16:
OS: Ubuntu 22.04.2 LTS
ROS Distro: Humble
Built from source or installed: Installed
Package version: For the TB4? I have the latest TB4 image. 
Model: standard
Networking Configuration & DDS vendor: I have default settings, so i think it should be simple discovery and Cyclone.

Additionally, are there any other ROS devices on the same network?
No.

hilary-luo, 2023-08-17:
Can you please indicate how you are launching nav etc and which pieces you are launching on your pc vs on the turtlebot? 

Looking at the create3 log it does look like the create3 lost wifi connection. Do you see this message each time in the create3 log despite the different symptoms?
```
Aug 13 23:04:31 iRobot-CEE9049B6ECA4398AE20F21D5E3D6860 user.notice create-platform: [DBG] [PP] Nav PP delayed 24749 ms - Error Comms Lost!
Aug 13 23:04:31 iRobot-CEE9049B6ECA4398AE20F21D5E3D6860 user.notice create-platform: [DBG] proto event: PP_NOTIFY_COMM_LOST
Aug 13 23:04:32 iRobot-CEE9049B6ECA4398AE20F21D5E3D6860 user.notice create-platform: proto event: PP_NOTIFY_COMM_LOST[INFO] [1691967872.073999376] [motion_control]: Setting safety override to "none"
Aug 13 23:04:33 iRobot-CEE9049B6ECA4398AE20F21D5E3D6860 user.notice create-platform: [INFO] [1691967873.274481006] [system_monitor]: Updating ready to drive: false
```

Do you ever see the create3 lose wifi connection when you are using the turtlebot but not using nav?

nidia-bucarelli, 2023-08-18:
I just tested the robot by leaving it alone (without sending commands or ssh into) for ~5-10 minutes after turning it on. Devices were in communication. Yet, it just went off. It has a red light. The robot has been connected to the docking station during this whole process.

Either way, this is how i am launching the navigation:
1. Run IR obstacles detector:
`ros2 launch tb4_ir obstacles.launch.py` . This file can be found here: https://drive.google.com/file/d/1O8Lsy_8POdYH1EI5oYXBiJVDubU1-BoL/view?usp=sharing 

2. Run localziation:
`ros2 launch turtlebot4_navigation localization.launch.py map:=office-july13.yaml`

3. Run navigation:
`ros2 launch turtlebot4_navigation nav2.launch.py params_file:=/home/nidiaib2/turtlebot_ws/configs/nav2.yaml`

4. Run rviz
`ros2 launch turtlebot4_viz view_robot.launch.py`

hilary-luo, 2023-08-25:
Does that happen prior to the boot up melody that indicates the end of the boot up process or significantly after? I would expect you to see some of this behaviour during boot up and/or potentially shortly after due to just internal processes that are running.

Nothing is jumping out to me as a problem in the log. I do see a warning near the end about no ping received but no definitive error. You could reinstall the create3 firmware and set it up again if you want to see if that helps before trying discovery server.

closed 2023-09-21 by hilary-luo