turtlebot/turtlebot4 #247: Voxellayer/Obstaclelayer does'nt subscribe to right topic

NachtaktiverHalbaffe, 2023-08-15:
**Please provide the following information:**
OS: Ubuntu 22.04 (Docker)
ROS Distro: Humble
Built from source or installed: Installed
Package version: 1.0.3-1jammy.20230722.003833

**Expected behaviour**
With provided configuration file the voxellayer/obstaclelayer are detecting and showing obstacles in costmaps

**Actual behaviour**
Voxellayer (and also obstacle layer) not subscribing to right scan topic and thus not showing obstacles in costmaps. In Rviz2 its suggesting to add a /< namespace >/localcostmap/scan topic to visualization which seems very suspicious. This topic doesnt appear with my workaround when everything works as expected. I think the layers trying to subscribe to this topic to where clearly no publisher exists

**To Reproduce**
Provide the steps to reproduce:
- Start navigation2 (occurs in both cases when running with and without namespace) with launch files and configuartion parameters provided by turtlebot_navigation package
- Wait until everything is running
- Inspect Costmap and which nodes are subscribed to corresponding scan topic => Costmaps arent subscribed to corresponding scan topic

**My workaround**
I changed the topics of the data sources to "absolut" topics instead of ""relative" ones. A fellow student had also this issue with another turtlebot with stock trutlebot humble software (no namespacing) and this fix also helped. I provide the updated part of the config below:
```yaml
local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: false
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.06
      robot_radius: 0.15
      plugins: [static_layer, voxel_layer, inflation_layer]
      inflation_layer:
        plugin: nav2_costmap_2d::InflationLayer
        cost_scaling_factor: 4.0
        inflation_radius: 0.2
      voxel_layer:
        plugin: nav2_costmap_2d::VoxelLayer
        enabled: true
        publish_voxel_map: true
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        mark_threshold: 0
        observation_sources: scan
        scan:
          # original: topic: scan, not working
          topic: /<namespace>/scan # working
          max_obstacle_height: 2.0
          clearing: true
          marking: true
          data_type: LaserScan
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      static_layer:
        plugin: nav2_costmap_2d::StaticLayer
        map_subscribe_transient_local: true
        map_topic: /map
      always_send_full_costmap: true
```

I think the voxel layer isn't appending the prefixes right to the topics if using "relative" topics. Could be a nav2 issue, but I want to start reporting in this place in case you spot anything different before reporting to the nav2 project

NachtaktiverHalbaffe, 2023-08-17:
Great, keep me up to date if the have to forward the issue into the Nav2 project if its a pure Nav2-issue

Also forgot to mention that my fellow student runs it on bare metal (but also humble versions, not iron) instead of Docker, so it isnt a platform issue

hilary-luo, 2023-08-25:
@NachtaktiverHalbaffe I have entered a ticket with Nav2 to clarify how namespacing should be handled / remapped. The content of the ticket refers to the other clearpath repos but the solution will be applied to turtlebot4 as well. Please follow along here https://github.com/ros-planning/navigation2/issues/3772

NachtaktiverHalbaffe, 2023-08-28:
I have jumped into the github issue and also provided the things I found out during futher debugging and problem solving. Meanwhile the workaround is to specify absolute topics or use rewrittenyaml in the launchfile there these topics are rewritten to the right absolut ones

closed 2023-11-01 by hilary-luo