turtlebot/turtlebot4 #248: New Turtlebot 4 won't connect to wifi network and remains in AP mode

languyen14507, 2023-08-16:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04) Ubuntu 20.04
 - ROS Distro: (e.g. Galactic) Galactic
 - Built from source or installed: installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**
I followed the User Manual Basic setup https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html
Logged into the RPi with '"ssh ubuntu@10.42.0.1"
Ran command "sudo wifi.sh -s 'Buffalo_D125' -p 'd125_router' -r US && sudo reboot"
Expect the RPi to connect to Buffalo_D125 router network
 **Actual behaviour**
The RPi rebooted but remained in AP mode, the robot IP address display temporarily showed "UNKNOWN", then back to "10.42.0.1"

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
I also used the printed instructions at https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html which no longer exists on the web but the robot still remained in AP mode. These instructions had worked for two previous Turtlebot 4s.

hilary-luo, 2023-08-17:
Hi @languyen14507,

The latest turtlebots may be shipping out with Humble. You should be able to check this by ssh'ing into the robot and running `printenv ROS_DISTRO`. If this is the case then please [install the latest raspberry pi image](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image) for your desired ROS distro and then follow the appropriate wifi set up instructions for your ros distro.

languyen14507, 2023-08-18:
@hilary-luo 
I ssh'ed into the new turtlebot from my current PC (PC-1) and was able to verify that the new turtlebot runs ROS Humble. I decided to use another PC (PC-2), installed Ubuntu 22.04 and ROS Humble on it. When I tried to connect to Turtlebot4 wifi, it would not take the password turtlebot4. When PC-2 was running Ubuntu 20.04 and ROS Galactic, it did connect to Turtlebot4 wifi with password turtlebot4. What would be the quickest way for me to get  the new turtlebot running? I prefer to stay with ROS Humble since it's the future. Thanks.

hilary-luo, 2023-08-22:
Please [install the latest humble raspberry pi image](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image) to start from scratch and follow the approriate wifi set up instructions for humble. Running the galactic instructions may have messed up the configuration and the easiest way to get started will be to restart. After you reset you should be able to connect and use the `turtlebot4-setup` tool to configure the wifi settings.

languyen14507, 2023-08-24:
@hilary-luo 
My PC finally connected to Turtlebot4 wifi network. I proceeded with the turtlebot4-setup tool. Entered the required SSID and password, save, apply settings, yes. However, I got an error message as shown below
[image]

hilary-luo, 2023-08-24:
@languyen14507 Did you re-image the raspberry pi just before running the `turtlebot4-setup` tool as requested?

hilary-luo, 2023-08-25:
@languyen14507 Have you waited to let it boot up completely and play the melody? It may take slightly longer to boot up then you are used to. Once it is completely booted up, what is the IP address that it shows on the screen?

One option is that you can just wire into the pi and run the turtlebot4_setup tool from there.

hilary-luo, 2023-08-25:
Also make sure that you unzip the image before running the script. If you didn't try it again after unzipping.

languyen14507, 2023-08-26:
Yes, I did unzip the image, I did hear the melody but no IP address nor the Turtlebot4 wifi signal ever showed up. So I proceeded with the Recovering the Raspberry Pi process as shown in the picture below  . It shows the connection is there but ssh'ing into the RPi did not work. This happened to both the original microSD card that came with the robot and the microSD card that I used to flash the downloaded image. Note that I have used the ethernet connection to successfully recovered another Turtlebot4 before.

[image]

hilary-luo, 2023-08-28:
When you called the sd_flash.sh command did you provide it with the full path to the image including the .img extension on the name? It doesn't seem like the flash worked properly.

languyen14507, 2023-08-28:
Yes, the image file name was turtlebot4_standard_humble_1.0.0.img, I was very careful with the details after all that happened.

languyen14507, 2023-08-28:
The attached photos show:
1. The result of the bash sd_flash.sh command which shows the flash was complete without errors
2. Turtlebot4 with original SD card: RPi displays shows UNKNOWN IP address after powered on
3. Turtlebot4 with newly flashed SD card: RPi display is completely dark after powered on
I don't know what the properly powered on RPi should look like.

[image]
[image]
[image]

hilary-luo, 2023-08-29:
@languyen14507 Please only reach out with your issues through one method, in other words if you are interfacing with official Clearpath Support about an issue then please don't create a github issue for it and vice versa unless so directed. It wastes our time to be trying to support through two different routes for the same issue.

I am posting this here, even though you should be hearing from the support team, in case anyone else comes across this issue or references it in the future: You need to flash the full sd card `mmcblk0`, not a partition. Although the partition was flashed successfully, that is not compatible with how the turtlebot4 operates.

I am closing this ticket, if you continue to have issues getting your robot running, feel free to continue to pursue it with the support team and your open support ticket there.

closed 2023-08-29 by hilary-luo

hilary-luo, 2023-08-31:
@languyen14507 Thank you for the clarification. You can reopen this ticket if the issue is still persistent after re-imaging. That being said, if both robots are experiencing similar issues then it is beneficial for everyone if you pursue them both through the same venue.

languyen14507, 2023-09-16:
@hilary.luo,Sorry for the misunderstanding. I contacted ClearPath Support for a different Turtlebot4 than the one I initiated the issue on github with. We bought a total of 3 Turtlebot 4s and only one is working. The one I contacted Support is a Galactic model and the one I initiated an issue on github is a Humble model.