turtlebot/turtlebot4 #253: Map cannot be saved

joergwunsch, 2023-08-29:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed
 - Package version:
```
ros-humble-turtlebot4-description/jammy,now 1.0.3-1jammy.20230807.173752 amd64 [installed]
ros-humble-turtlebot4-desktop/jammy,now 1.0.0-1jammy.20230814.172216 amd64 [installed]
ros-humble-turtlebot4-msgs/jammy,now 1.0.3-1jammy.20230721.202703 amd64 [installed]
ros-humble-turtlebot4-navigation/jammy,now 1.0.3-1jammy.20230814.174121 amd64 [installed]
ros-humble-turtlebot4-node/jammy,now 1.0.3-1jammy.20230807.164024 amd64 [installed]
ros-humble-turtlebot4-viz/jammy,now 1.0.0-1jammy.20230814.171817 amd64 [installed,automatic]
```

 **Expected behaviour**
 The command given in the tutorial to save a map ought to work:
`ros2 run nav2_map_server map_saver_cli -f "office_floor_hires" --ros-args -p map_subscribe_transient_local:=true -r __ns:=/namespace`

 **Actual behaviour**
```
[INFO] [1693300735.675584289] [namespace.map_saver]: 
	map_saver lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1693300735.675964234] [namespace.map_saver]: Creating
[INFO] [1693300735.676152475] [namespace.map_saver]: Configuring
[INFO] [1693300735.677650111] [namespace.map_saver]: Saving map from 'map' topic to 'office_floor_hires' file
[WARN] [1693300735.677793933] [namespace.map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[WARN] [1693300735.677830920] [namespace.map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
[ERROR] [1693300737.680139479] [namespace.map_saver]: Failed to spin map subscription
[INFO] [1693300737.681739001] [namespace.map_saver]: Destroying
[ros2run]: Process exited with failure 1
```

**To Reproduce**
Followed the instructions from the tutorial to start the SLAM algorithm, RViz2 shows something. Tried in a different terminal to save the map, consistently failed, even across robot reboots. Failure was independent of whether the mapping was run on the Raspberry Pi or on the host Ubuntu.
All this used to work before, but now it is consistently failing, without more explanation than shown above.

hilary-luo, 2023-08-29:
@joergwunsch My understanding is that you are using namespacing. Please see details here about a current bug in the slam toolbox https://github.com/turtlebot/turtlebot4/issues/231. If indeed this is the issue that you are running into you can either use the work around listed there, or if you have time I would encourage you to test out the fix https://github.com/SteveMacenski/slam_toolbox/pull/613 and then comment on the PR whether or not it worked. Once there is confirmation that the fix works then it can be merged in.

joergwunsch, 2023-08-30:
Well, all I have been doing was to follow the tutorial:
https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html#save-the-map
That commandline uses namespace – I just copy&pasted it.

joergwunsch, 2023-08-30:
I can confirm, removing `-r __ns:=/namespace` from the commandline can successfully save the map. Maybe it should just be omitted from the tutorial then?

hilary-luo, 2023-08-31:
@joergwunsch that makes sense, if you are not using namespacing you should remove that part of the command. I see what you mean that part of the tutorial does not account for namespacing while other parts do. I have made note of this for a future update to offer clarification.

closed 2023-08-31 by hilary-luo

sdobbelaere, 2023-09-08:
I also ran into the same trap with the copy/pasting of the command.  
Since the entire tutorial kinda avoids using namespaces, this command really shouldn't contain that param.  Or at the very least provide some more clarity on the matter.

hilary-luo, 2023-09-08:
@sdobbelaere There is a PR for this change already [here](https://github.com/turtlebot/turtlebot4-user-manual/pull/15)