turtlebot/turtlebot4 #255: Turtlebot4 Humble Navigation Problem: Local Costmap Not Populating

S0nFra, 2023-09-01:
**Please provide the following information:**

 - OS: Ubuntu 22.04.3

 - ROS Distro: Humble

 - Built from source or installed: installed

 - Package version: from `sudo dpkg -s ros-$ROS_DISTRO-turtlebot4-navigation`

   ```
   Package: ros-humble-turtlebot4-navigation
   Status: install ok installed
   Priority: optional
   Section: misc
   Installed-Size: 2470
   Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
   Architecture: amd64
   Version: 1.0.3-1jammy.20230814.174121
   Depends: ros-humble-nav2-bringup, ros-humble-nav2-simple-commander, ros-humble-slam-toolbox, ros-humble-ros-workspace
   Description: Turtlebot4 Navigation
   ```

**Expected behaviour**
I need to navigate the robot in an environment of which I already have the map.
Following the instructions from the  [documentation](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html). I run the following commands on my laptop:

1. `ros2 launch turtlebot4_navigation localization.launch.py map:=my_map.yaml`
2. `ros2 launch turtlebot4_navigation nav2.launch.py`
3. `ros2 launch turtlebot4_viz view_robot.launch.py`

 **Actual behaviour**
By setting a goal from RViz, it appears that the local costamap does not populate correctly, in fact placing an obstacle along the trajectory is not detected, leading to an inevitable crash.

**To Reproduce**

My setup: 

- TurtleBot 4 Standard
- A telephone as a 2.4 Ghz network hotspot (Xiaomi 11 lite 5g NE in my case) 
- Discovery server

Provide the steps to reproduce:

1. Execute the above commands
2. Set the initial pose on RViz
3. Place an obstacle
4. Set a goal

Result:
https://github.com/turtlebot/turtlebot4/assets/80450457/d74ccf9b-c2e6-4053-8284-cdb8951a3588

I attach screen prints for the commands executed.
[turtlebot4_navigation localization.launch.py.txt](https://github.com/turtlebot/turtlebot4/files/12500024/turtlebot4_navigation.localization.launch.py.txt)
[turtlebot4_navigation nav2.launch.py.txt](https://github.com/turtlebot/turtlebot4/files/12500026/turtlebot4_navigation.nav2.launch.py.txt)
[turtlebot4_viz view_robot.launch.py.txt](https://github.com/turtlebot/turtlebot4/files/12500027/turtlebot4_viz.view_robot.launch.py.txt)

hilary-luo, 2023-09-01:
This bug is being addressed and is being tracked in https://github.com/turtlebot/turtlebot4/issues/247 and the subsequently linked issue. 

In the meantime you can modify the scan topic in the [nav2.yaml and localization.yaml files](https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_navigation/config) from `topic: scan` to `topic: /scan`. There are multiple occurances such as: [here](https://github.com/turtlebot/turtlebot4/blob/d4412899e428196b0905e4edd2896b8c98ce6580/turtlebot4_navigation/config/nav2.yaml#L169). Then when you are launching nav2 and localization, pass the path to the modified config files using the param key. For example:
```
ros2 launch turtlebot4_navigation localization.launch.py map:=my_map.yaml params:=path_to_config/localization.yaml
ros2 launch turtlebot4_navigation nav2.launch.py params:=path_to_config/nav2.yaml
```

If you are able to get the workaround working for you then please close this ticket and feel free to follow the other ticket for updates.

S0nFra, 2023-09-04:
Thank you for your response. Unfortunately, this solution did not work in my case.

These are the changes I made in the `localization.yaml` configuration file

```yaml
local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.06
      robot_radius: 0.175
      plugins: ["static_layer", "voxel_layer", "inflation_layer"]
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 4.0
        inflation_radius: 0.45
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        publish_voxel_map: True
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        mark_threshold: 0
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      always_send_full_costmap: True

global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: True
      robot_radius: 0.175
      resolution: 0.06
      track_unknown_space: true
      plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 4.0
        inflation_radius: 0.45
      always_send_full_costmap: True
```

 and in `nav2.yaml`

```yaml
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: /scan
```

I attach the complete files here:
[localization.yaml](https://github.com/turtlebot/turtlebot4/files/12515652/localization.yaml.txt)
[nav2.yaml](https://github.com/turtlebot/turtlebot4/files/12515653/nav2.yaml.txt)

hilary-luo, 2023-09-04:
Can you please give me the output of `ros2 topic list` while running navigation with the new configuration files?

S0nFra, 2023-09-05:
Sure, that's the output of `ros2 topic list` when the sistem is on

```
/amcl/transition_event
/amcl_pose
/bond
/clicked_point
/diagnostics
/downsampled_costmap
/downsampled_costmap_updates
/global_costmap/costmap
/global_costmap/costmap_updates
/global_costmap/voxel_marked_cloud
/initialpose
/local_costmap/costmap
/local_costmap/costmap_updates
/local_costmap/published_footprint
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_server/transition_event
/map_updates
/mobile_base/sensors/bumper_pointcloud
/mobility_monitor/transition_event
/parameter_events
/particle_cloud
/plan
/robot_description
/rosout
/scan
/tf
/tf_static
/waypoints
```

S0nFra, 2023-09-06:
Thank you for your response. I implemented the ntp fix, replacing the indicated lines.

The result does not change unfortunately. I attach the log of create3.
[messages.txt](https://github.com/turtlebot/turtlebot4/files/12535582/messages.txt)

Small caveat, in the initial post I had mistakenly written that the network configuration was the Discovery server when in fact it was always about the Simple discovery. I don't know if that changes anything.

hilary-luo, 2023-09-10:
@S0nFra Thank you for clarifying the network configuration, that does change some things. 

In your create3 logs, it is showing that the create3 is struggling to communicate, I can tell this from the repetition of this line among others:
```
Sep  6 08:19:33 iRobot-6C2F37976D2741578477A380B5795BC2 user.notice create-platform: 1693988373.095552 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
```

I would suggest either testing with a proper router network (instead of hotspot) or switching to discovery server (if possible).

hilary-luo, 2023-10-12:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-10-12 by hilary-luo

saad-waqar-robotics, 2024-06-15:
Hello @hilary-luo  I am still facing this issue. I have checked with tb3 simulation, it works good in there. But when I use my own robot (I have tested multiple robots), (especially with gazebo), this issue occurs. Even the local_costmap shows some data with ros2 topic echo ....
But in RVIZ it does not receive anything. 
I also noted that, when I check subs and pubs of the topic, I see many subscriptions from nav2 when it is working fine, but if it is not working in Rviz, the only subscriber being shown is Rviz only.

hilary-luo, 2024-06-17:
@saad-waqar-robotics Please open your own ticket with full details about your system and full terminal commands and outputs from everything that you are running. We can then help you troubleshoot your system to understand what is causing your issues.