turtlebot/turtlebot4 #258: Limited topics available to the user PC compared to Pi.

ramakrishnasakethi, 2023-09-04:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 - Built from source or installed: Installed

 **Expected behaviour**
Both user PC and the Pi returning all available topics published by Create 3 and Pi.

 **Actual behaviour**
ros2 topic list returning only a limited set of topics compared to pi. I have attached a screenshot for your reference. I followed the instructions under Simple Discovery to setup the communication between the bot and the user PC.
[image]

**To Reproduce**
Setup [https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html](url)
Verify running `ros2 topic list` in both user PC and Pi to see that the topics accessible match.

ramakrishnasakethi, 2023-09-04:
A snapshot of the log for your reference.
[image]

hilary-luo, 2023-09-05:
@ramakrishnasakethi I see that you have entered a number of issues that seem inter-related. Are they all on the same robot? If you don't have the create3 connected to wifi then simple discovery will not work properly and you will not be able to see all of the ros 2 topics from your computer. As such you will not be able to run nav. 

If these are on the same robot then I can work through these issues with you but one at a time starting with getting the create3 connected to wifi. Taking them out of order may make some of the troubleshooting steps not make sense.

ramakrishnasakethi, 2023-09-06:
@hilary-luo I tried following the steps mentioned by you in #257
1. I could not find the robot listed as Create-[xxx] in the Network Manager after pressing and holding buttons 1 and 2 for a few seconds. The light ring has the spinning blue light though.
2. The robot starts to make a maneuver that has not been mentioned by you in the procedure after pressing and holding the buttons 1 and 2 for approximately 4-5 seconds before the light ring turns spinning blue.

hilary-luo, 2023-09-06:
@ramakrishnasakethi Even if you were not able to connect through the create3 acess point, this still should have turned on the wifi connection of the create3. Assuming that was the only issue, you should be able to connect the create3 base to the internet now through the original method. (connect to the create3 webserver using the raspberry pi ip address and navigate to the connection page etc). 

In regards to the fact that the robot started driving, that is fine, depending on how you pressed them or the mode that the robot is in it may do so. Individually those buttons are used for Turtlebot4 for docking and undocking as well.

ramakrishnasakethi, 2023-09-07:
> Once I took those steps to enable wifi, it showed me my available wifi networks. Not the actual robot.

By it, do you mean the PC's network manager?

ramakrishnasakethi, 2023-09-07:
> @ramakrishnasakethi Even if you were not able to connect through the create3 acess point, this still should have turned on the wifi connection of the create3. Assuming that was the only issue, you [...]

I tried connecting the Create-3 through the original method now. It is the same like before. I see the following page after entering the Wi-Fi credentials and clicking on connect,
[image]

After 15 minutes, I opened the Create3 webserver in a different page and entered the connect tab and it seems as though I have to enter the credentials all over again,
[image]´

Log file for your reference: 
[messages.txt](https://github.com/turtlebot/turtlebot4/files/12544642/messages.txt)

hilary-luo, 2023-09-10:
Ensure that you have closed all browser windows that are connected to the webserver and then please try to follow the instructions again (from [the other page](https://github.com/turtlebot/turtlebot4/issues/257#issuecomment-1705624839)) and hold the buttons down until you get the **spinning** cyan light pattern as shown here: https://iroboteducation.github.io/create3_docs/hw/face/

Either you may not have held the buttons down long enough or perhaps having the webserver open through the raspberry pi maybe could have interfered.

ramakrishnasakethi, 2023-09-11:
> 
What do you mean by computer's wifi manager?

ramakrishnasakethi, 2023-09-11:
> Ensure that you have closed all browser windows that are connected to the webserver and then please try to follow the instructions again (from [the other [...]

An IP Address now appears on top of the create3 webserver when I open the connect tab, which was previously not the case. But I still am not able to connect to the robot to the network. A snapshot of the log corresponding to the failed attempt to connect to the network,
[image]

ramakrishnasakethi, 2023-09-12:
List of topics available enough to do mapping after connecting Create 3 to WiFi properly using solution provided [here](https://github.com/turtlebot/turtlebot4/issues/257#issuecomment-1705624839)

closed 2023-09-12 by ramakrishnasakethi