turtlebot/turtlebot4 #259: Unable to map the environment using Synchronous SLAM

ramakrishnasakethi, 2023-09-04:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: Installed

 **Expected behaviour**
Be able to map the environment after running `ros2 launch turtlebot4_navigation slam.launch.py` 

 **Actual behaviour**
Errors as can bee seen in the screenshots:
1. After running `ros2 launch turtlebot4_navigation slam.launch.py` and `ros2 launch turtlebot4_viz view_robot.launch.py` in the command terminal:
[image]

2. Error in rviz:
[image]

**To Reproduce**
Run `ros2 launch turtlebot4_navigation slam.launch.py` and `ros2 launch turtlebot4_viz view_robot.launch.py` from the command terminal.

**Other notes**
Only a limited set of topics are available in the User PC compared to Pi.
[image]

hilary-luo, 2023-09-05:
Hi @ramakrishnasakethi are you using simple discovery or discovery server?

Can you show me the list of topics on each of your robot and pc when you are trying to run slam? Your rviz window doesn't seem to be receiving the information from the SLAM toolbox.

ramakrishnasakethi, 2023-09-12:
Able to run slam after connecting Create-3 to WiFi using solution mentioned [here](https://github.com/turtlebot/turtlebot4/issues/257#issuecomment-1705624839)

closed 2023-09-12 by ramakrishnasakethi