turtlebot/turtlebot4 #260: Turtlebot fails to avoid obstacles during navigation

Saurabh0325, 2023-09-04:
**Please provide the following information:**
 - OS: 22.04.03
 - ROS Distro: Humble
 - Built from source or installed:
 - Package version: installed--"sudo apt install ros-humble-turtlebot4-desktop", "sudo apt install ros-humble-turtlebot4-navigation"

 **Expected behaviour**
Should avoid obstacles while navigating using a custom map.

 **Actual behaviour**
 Does not avoid obstacles, the planned path does not take obstacles into consideration. 

**To Reproduce**
Provide the steps to reproduce:
1. ros2 launch turtlebot4_navigation localization.launch.py map:=hiwi_raum.yaml
2. ros2 launch turtlebot4_navigation nav2.launch.py
3. ros2 launch turtlebot4_viz view_robot.launch.py

https://github.com/turtlebot/turtlebot4/assets/127971127/18de13e7-ee75-48de-8971-a2b3806f57b4

hilary-luo, 2023-09-05:
This bug is being addressed and is being tracked in https://github.com/turtlebot/turtlebot4/issues/247 and the subsequently linked issue.

In the meantime you can modify the scan topic in the [nav2.yaml and localization.yaml files](https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_navigation/config) from topic: scan to topic: /scan. There are multiple occurances such as: [here](https://github.com/turtlebot/turtlebot4/blob/d4412899e428196b0905e4edd2896b8c98ce6580/turtlebot4_navigation/config/nav2.yaml#L169) . Then when you are launching nav2 and localization, pass the path to the modified config files using the param key. For example:
```
ros2 launch turtlebot4_navigation localization.launch.py map:=hiwi_raum.yaml params:=path_to_config/localization.yaml
ros2 launch turtlebot4_navigation nav2.launch.py params:=path_to_config/nav2.yaml
```
If you are able to get the workaround working for you then please close this ticket and feel free to follow the other ticket for updates.

Saurabh0325, 2023-09-11:
@hilary-luo I have tried the suggested solution, with and without namespace. Unfortunately it didn't work. The robot behavior is still the same. It cannot avoid the obstacle in front of it. I am not able to figure out what is wrong. Can you please help?

Saurabh0325, 2023-09-12:
@hilary-luo  after updating the nav2 and localization yaml file the robot was able to avoid the obstacles. But I have three robots with 3 different namespaces, so should I change the namespace every time to try with different robot? or is there any other way?

hilary-luo, 2023-09-14:
@Saurabh0325 For now I would suggest going through this same process on each robot with their respective namespaces so that each robot's configs are pointing at the correct scan topic. That would be the simplest. I should be releasing a fix for this shortly (as soon as I have time).

closed 2023-09-18 by Saurabh0325