turtlebot/turtlebot4 #266: Create3 100% CPU usage

hilary-luo, 2023-09-10:
This is going to be a place to discuss and troubleshoot the CPU usage at 100%. I am guiding this process and the expectation is that anyone involved needs to commit to providing detailed information about their setup and implementing changes that may or may not fix the issue. The goal is to identify a common solution that can be deployed. Indiviudals are welcome to try additional tests and report back but must indicate what changes have been made in case it impacts the results of requested tests.

As a start, can each of you please provide:

 - **OS:** (e.g. Ubuntu 20.04)
 - **ROS Distro:** (e.g. Galactic)
 - **Built from source or installed:**
 - **Package versions:** (if from repository, give version from `sudo dpkg -s ros-$ROS_DISTRO-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash) -> really just make sure you are using the latest of all packages on the robot and on your PC
 - **Create3 Firmware version:** 
 - **Robot Version:** (standard or lite)
 - **RMW Implementation:** (FastDDS or CycloneDDS)
 - **Networking Configuration:** (Simple Discovery or Discovery Server)
 - **Network Information:**
   - Is the robot connected to 2.4GHz or 5 GHz (each of create3 & raspberry pi) 
   - is it a dedicated router over which you have full control or an institutional network?
   - Is there any other ROS communication on this network? (additional turtlebots or other devices and if so are they on the same ROS Domain or namespaced?)
 - Have you applied any custom changes through the create3 beta features? (NTP or DDS profile etc)
 - Do you have turtlebot4 diagnostics enabled?

Finally, can you describe what situations cause the 100% CPU issue and provide the create3 logs from such a situation. Feel free to share other symptoms that you see so we can track them. I want to ensure that I understand what each person is working with to make this process effective.

hilary-luo, 2023-09-10:
@garyservin @NachtaktiverHalbaffe Can you please share the information described?

NachtaktiverHalbaffe, 2023-09-11:
# Your requested informations
- **OS**: Ubuntu 22.04 (Docker image from ros:humble with additional deps installed)
- **ROS Distro**: Humble
- **Built from source or installed**: Installed from apt repos (except teb_local_planner which isnt in repos)
- **Package Versions**: 1.0.3-1jammy.20230814.174121
- **Create3 Firmware versions**: H2.2
- **Robot version**: Standard
- **RMW implementation**: FastDDS (tried to change to cyclonedds when obstacle layer didnt work, but didnt work, then directly switched back because found out that the fastdds bug regarding obstacle layer was fixed with repo version of the packages)
- **Networking cofiguration**: Simple discovery (Kinda hard requirement by my Supervisor because of the "Discovery Server currently does not support communicating with multiple TurtleBot 4’s simultaneously from one computer"-info from the turtlebot docs and for one of our model processes where the turtlebot is involved this is a hard requirement that we can do this)
- **Network Information**:
   - 2,4 Ghz or 5 Ghz: The Pis are connected to 5, the create3 bases to 2,4 (2,4 is a little crowded here in the university for obvious reasons, but connection seems free from significcant interference)
   - Its a dedicated router. However, its connected over WAN to the network of our institute. This is only a temporary solution until our IT team setups a Wifi sub network which is installed and configured to the standards of our IT. Also because of this I had to change the NTP servers at the Raspberry Pis 
   - On the router itself only our 3 turtlebots run (+ the docker clients which runs the navigation2 stack, central map server etc). However, at the moment we only have turned on and running one turtlebot at the same time. They are namespaced, but in different ROS domains until we are ready for the multi robot scenario. Dont know if other run other ros2 stuff on the WAN level of the network, but I assume that this is not the case in general. Only my own nodes show up each time
   - Have you applied any custom changes through the create3 beta features?: Yes, the NTP change like suggested here: https://github.com/turtlebot/turtlebot4/issues/216#issue-1797043215
   - Do you have turtlebot4 diagnostics enabled? Havent enabled/disabled it by hands. So if its enabled by default, then yes. The node appears `ros2 node list`

# Some more information about my setup
- Its a requirement that it is possible to write nav2 planner and costmap layer plugins with Python
   - To achive this, I wrote C++ plugins which calls ROS2 services when the plan should be computed or costs should be updated. 
   - This adds 5 service clients and 3 Nodes with one service server each to the navigation2 stack (local and global costmap call same service, thus for these two clients and one service exists)
   - I wrote GUIs with corresponding nodes. These adds one service and node for each service and there is one gui run for each turtlebot (currently one) and one central for the central server
- We have the map server running as a central instance for all turtlebots on another Computer  (with same docker image) and thus are the only place where namespacing isnt used. Thus we dont use the localization launch file by nav2, instead we only run amcl with lifecycle manager on each nav2 stack for each turtlebot. 
- I run amcl with use_composition which works. I tried same for nav2, but didnt work because static layer couldnt subscribe to the map topic (which runs outside the composition container into which the nodes are loaded). 
    - Hoped this could reduce the load during startup because of intra-process communication instead other rmw for all internal communication
    - I assume its because my map server runs on another machine as a regular node over RMW and the dynamic composition was designed to run it inside the localization launchfile inside the same node container as a composable node
    - Maybe use_composition in all places of the nav2 stack could work if the constraints allow it (which it didnt in my case)
- I wrote a own launchfile for the navigation2 stack, where first amcl, then with 3 seconds delay rviz and then with another 10 seconds to reduce stress when the nodes come up. Havent optimized the delays, but since them the turtlebot hasn't hang up with a red light as many times as it has before. The 100% CPU usage still exists because of this. However, this is only a observation and havent done any real experiments to measure this
- I changed the controller server plugin to teb controller because it performed better in our usecase (we have a parcour/maze with rather smaller corridors at some places)
- I know these are some more adjustments to nav2 and the turtlebot4 packages, but its required by the requirements of my Supervisor for the model process Im setting up
- With all these adjustments and no nav2-stack running (nav2, rviz and amcl), the create3 idles at ` CPU usage: max 96 [%] mean 78 [%] RAM usage: 38/59 [MB]`, rising to `[turtle1.system_health]: CPU usage: max 100 [%] mean 90 [%] RAM usage: 38/59 [MB]` when a node e.g. the GUI comes up

# When does the CPU max out
- When I launch the Nav 2 stack
- Then it climbs to 100% and the turtlebot becomes unresponsive until everything started
- The startup also takes like rather minutes than seconds. I assume it is because of the maxed out create3 and that the tf messages take to long to be send for which the nav2 plugins are waiting before going to an active state
[... cut ...]
Sep 11 10:35:57 iRobot-9FB4F821E7114AAEA6653A08B8DFFCAE user.notice create-platform: [INFO] [1694428557.929650019] [turtle1.system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 39/59 [MB]
Sep 11 10:36:57 iRobot-9FB4F821E7114AAEA6653A08B8DFFCAE user.notice create-platform: [INFO] [1694428617.933529025] [turtle1.system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 41/59 [MB]
Sep 11 10:37:57 iRobot-9FB4F821E7114AAEA6653A08B8DFFCAE user.notice create-platform: [INFO] [1694428677.932075858] [turtle1.system_health]: CPU usage: max 100 [%] mean 93 [%] RAM usage: 41/59 [MB]
```

NachtaktiverHalbaffe, 2023-09-12:
Im not in the lab on the robot today, so I can only answer the things I have observed. If needed, I can run additional tests tomorrow. I havent reliable tracked and recorded any number, so these are my impressions, but not anything specifically measured. 

Based on the question I have some more information so you can understand the behaviour better:
- When the create3 hangs up with a "red light of death", the whole Turtlebot cant be rebooted
     - If you press the power button, often it doesnt react directly
     - When it is finally responding to one of the repeated power button presses, then it directly reboots the create3 base instead of turning off. Only after that reboot a complete powercycle is possible
     - During normal booting the robot, the Create3 also has this red light, but in this case it disappears before its ready
- I think I miscommunicated the CPU usage when the Nav2 stack is running
    - The 80-90% only occur on better days and when the nav2 stack is idle for some time and the robot is driven with teleop
    - The common case is 90-100% Usage
    - The lower end is when idle and nav2 stuck running and having settled and/or using just teleop for driving
    - The upper end is when the robot drives around (nav2 stack) the parcour for some time
- I applied all changes around the same time (ntp, delayed launch files, using composition etc.), so I cant say sure which of  these had the biggest impact if they had one significantly
- On the create3 logs there occurs mqtt and discovery related errors. I thought this is due to discovery server which is deactivated in my case. Can it be that this some kind of real internal error which should be tracked down?
- I yesterday tried to introduce the system into the multi roboter scenario by putting two turtlebots into the same namespace, but booting the second nav2 stack after the first was up running. I run it one time and directly both turtlebots had the "red light of death". I try tomorrow if this is consistent, but seems that to turtlebots with nav2 stacks at same ros domain is not possible with these problems
 
To your questions:
1. Cant say. We directly upgraded it during our setup process and before I found out about this problem and thus before debugging this behaviour
2. Cant say either because 
a) before that all the default NTP servers werent reachable because blocked by our firewall. It appeared about the same time when I changed the NTP server on the Pi to our one  when I also applied these ntp settings. Before that the NTPs werent reachable 
b) like mentioned before I applied this at same time with other changes
But I can say that NTP related errors occur less often, but sometimes some warnings about delay problems. 
3. **Before my changes** I didn't find patterns other than these I mentioned. I had days where It occured every two nav2 startups and other ones where it didnt occur one time. The observation I have is that it is more likely to hang up and have the "red light of death" when it has occured before and it dissappears more likely if you take the time and powercycle all systems after that instead of using it right after the create 3 has rebooted after the "red light of death". At this time I was also integrating different modules regarding my plugin bridges, fixes to nav2 stack etc. so these could also had been impacted this.  
**After my changes** it doens't hang up with the "red light of death" that often. Had running it 3 days and it only occured two times at the end of my day. Had rebooted the nav2 stack multiple times at this day and also changed implementation of nodes (nav2 stack side) which also had then errors without crashing the turtlebot. So theres a correlation with my changes (but correlation doesnt mean causality). However, the unresponsivness problem during startup and the CPU maxing out still exists, it just doesnt crash that often
4. You understood right (here also again the correlation and causality problem though). I think I had it like one or two times also when everything was running, but I cant remember the circumstances and this was rare behaviour. Regarding the question: Havent tracked the numbers. These are my impressions: 
- The message filter dropping messages because of full cues occurs every time during startup and some time during operation and in combination with LIDAR messages. The Raspberry Pi seems to run fine and data is reliable received from the Lidar, so I think it isnt related to the create3 problem
- The mismatched timestamp issues occurs after the short "hickups" and often the robot was unresponsive at these state. I assume this is more to faulty messages in message cues (faulty in terms of timestamps and other header infos) which are introduced by these hiccups. If these hickups are on my host or on a messaging hickup I cant say. However, after his the nav2 stack must be rebooted because it is stuck (because of blocked message cues I think), but then its a crash on the nav2 side
- The extrapolation errors are most likely due to amcl etc. not running fast enough or not properly applying the transform tolerances. Its also appearing unregulary and not interupting navigation or leading to a crash

Do you meaning with reserving IP adresses that the Turtlebots gets the same IP adress every time? Then they are already reserved because our DHCP server is configured to assign the Turtlebots and the create3 bases the same IP adress every time.

hilary-luo, 2023-09-13:
Thank you for the detailed response and yes I was referring to having the same IP address every time so that sounds perfect. To respond to some of your thoughts from above:

> On the create3 logs there occurs mqtt and discovery related errors. I thought this is due to discovery server which is deactivated in my case. Can it be that this some kind of real internal error which should be tracked down?

My understanding is that the mqtt errors are internal to the create3 and can be ignored in this case.

> Multiple Turtlebots and navigation2 stacks in same ROS Domain is not reliable working. Best approach seems to isolate each turtlebot and nav2 stack into own DOmain and then bridge the topics, services etc. which are needed between robots for the individual usecase

Once we get the single robot issue sorted out this is something that I would like to look at more closely. The systems should support two turtlebots running nav2 on the same network. 

As the first requested change, can you make sure that the turtlebot4 diagnostics are disabled on the Turtlebot. If they were on previously then see if this makes a notable difference in the performance that you have described. (Steps to turn diagnostics off: SSH into the robot and run turtlebot4-setup -> go under ROS Setup -> Bash Setup -> Select TURTLEBOT4_DIAGNOSTICS -> Disabled -> Save -> Back out to the main menu -> Apply Settings -> Yes )

NachtaktiverHalbaffe, 2023-09-15:
I have left it on one Turtlebot enabled and disabled on the other. Each Turtlebot was in its own ROS Domain and bridged to the central map server with domain_bridge from ROS2. I ran these measurements two times (complete reboot of systems in between) to measure the number on the same PC/Docker container for each Turtlebot. My rough results are:
| **Situation**                             | **With this change**                   | **Without change**                    | **Gain (Difference in avg CPU usage)** |
| ----------------------------------------- | -------------------------------------- | ------------------------------------- | -------------------------------------- |
| Idle (direct after booting, 9 datapoints) | 70% at start, 53-63% after settled     | 75-77% at start, 70-73% after settled | 7-20%                                  |
| Idle with Nav2-Stack etc. running               | 95-100% at start, 73-77% after settled | 100% at start, 88-91% after settled   | 10-20%                                  |
| Robot driving (driving all the time => Full messaging load) | 86-93%                                 | 97-98%                                | 5-10%                                 | 

This behaviour is consistent then I disabled diagnostics on the second Turtlebot too. However, depending on the PC I ran the nav2 stack and the turtlebot itself I have around 5% of deviation in CPU usage, so I would rather set the gains in 10-15% range for the Idles and 5% when running.

> Once we get the single robot issue sorted out this is something that I would like to look at more closely. The systems should support two turtlebots running nav2 on the same network.

Yeah in my case there was a log that it discovered ~70 Nodes when everything was in same ros domain, I have some more because of my Plugin bridges etc., so I assume I put additional stress to the create 3 on startup of the Nav2 stack compared to the stock one

garyservin, 2023-09-17:
- OS: Ubuntu 22.04
- ROS Distro: Humble
- Built from source or installed: installed from debians
- Package versions: 
```
ii  ros-humble-turtlebot4-base                        1.0.1-1jammy.20230822.184857            arm64        Turtlebot4 Base Node
ii  ros-humble-turtlebot4-bringup                     1.0.1-1jammy.20230909.013636            arm64        Turtlebot4 Robot Bringup
ii  ros-humble-turtlebot4-description                 1.0.3-1jammy.20230822.192419            arm64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-diagnostics                 1.0.1-1jammy.20230822.190153            arm64        Turtlebot4 Diagnostics
ii  ros-humble-turtlebot4-msgs                        1.0.3-1jammy.20230721.201332            arm64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                  1.0.3-1jammy.20230822.234843            arm64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                        1.0.3-1jammy.20230822.184301            arm64        Turtlebot4 Node
ii  ros-humble-turtlebot4-robot                       1.0.1-1jammy.20230909.045132            arm64        Turtlebot4 Robot Metapackage
ii  ros-humble-turtlebot4-tests                       1.0.1-1jammy.20230721.205750            arm64        Turtlebot4 System Tests
```
- Create3 Firmware version: Current Firmware Version: H.2.2 (Humble)
- Robot Version: lite
- RMW Implementation: CycloneDDS
- Networking Configuration: Simple Discovery
- Network Information:
 - Create base is connected to 2.4GHz, Pi is connected to 5GHz
 - This is a dedicated router which I have control over
 - No extra ROS communication over the network, just a single Turtlebot
 - I have configuered NTP as recommended here https://github.com/turtlebot/turtlebot4/issues/216#issue-1797043215
- Do you have turtlebot4 diagnostics enabled? There is a /diagnostics topic, not sure what's the default, but I haven't enabled/disabled manually

## Some more information
CPU usage when the turlebot service isn't running and after letting the base to stabilize (about 30 minutes after powering on)
```
Sep 16 21:53:35 create3 user.notice create-platform: [INFO] [1694901215.651064233] [system_health]: CPU usage: max 57 [%] mean 48 [%] RAM usage: 32/59 [MB]
Sep 16 21:54:35 create3 user.notice create-platform: [INFO] [1694901275.653174351] [system_health]: CPU usage: max 68 [%] mean 54 [%] RAM usage: 32/59 [MB]
```
CPU usge when the turtlebot4 service is running on the Pi, after 30 minutes. CPU was at 100% for almost 10 minutes after starting the service
```
Sep 16 22:30:35 create3 user.notice create-platform: [INFO] [1694903435.668905870] [system_health]: CPU usage: max 79 [%] mean 72 [%] RAM usage: 31/59 [MB]
Sep 16 22:31:35 create3 user.notice create-platform: [INFO] [1694903495.669638204] [system_health]: CPU usage: max 82 [%] mean 73 [%] RAM usage: 31/59 [MB]
Sep 16 22:32:35 create3 user.notice create-platform: [INFO] [1694903555.670138188] [system_health]: CPU usage: max 61 [%] mean 55 [%] RAM usage: 32/59 [MB]
```

## When does CPU max out

When I start mapping, I get disconnections, and the robot disappears  for a second from rviz. It's worse when I set the mapping resolution to 0.01 via a param override.
These are the logs I got during that time
```
Sep 16 23:04:48 create3 kern.warn kernel: [ 5855.130573] MtAsicUpdateProtect: unknown protection mode(9)
Sep 16 23:04:49 create3 kern.warn kernel: [ 5855.680544] MtAsicUpdateProtect: unknown protection mode(7)
Sep 16 23:04:50 create3 kern.warn kernel: [ 5856.780542] MtAsicUpdateProtect: unknown protection mode(9)
Sep 16 23:04:50 create3 kern.warn kernel: [ 5856.890538] MtAsicUpdateProtect: unknown protection mode(7)
Sep 16 23:04:50 create3 kern.warn kernel: [ 5857.330524] MtAsicUpdateProtect: unknown protection mode(9)
Sep 16 23:04:50 create3 kern.warn kernel: [ 5857.440514] MtAsicUpdateProtect: unknown protection mode(7)
Sep 16 23:04:51 create3 kern.warn kernel: [ 5857.885612] MtAsicUpdateProtect: unknown protection mode(9)
Sep 16 23:04:51 create3 kern.warn kernel: [ 5857.994330] MtAsicUpdateProtect: unknown protection mode(7)
Sep 16 23:04:51 create3 kern.warn kernel: [ 5858.430520] MtAsicUpdateProtect: unknown protection mode(9)
Sep 16 23:04:51 create3 kern.warn kernel: [ 5858.540508] MtAsicUpdateProtect: unknown protection mode(7)
Sep 16 23:04:52 create3 kern.warn kernel: [ 5858.980494] MtAsicUpdateProtect: unknown protection mode(9)
Sep 16 23:04:52 create3 kern.warn kernel: [ 5859.090496] MtAsicUpdateProtect: unknown protection mode(7)
Sep 16 23:04:52 create3 kern.warn kernel: [ 5859.533156] MtAsicUpdateProtect: unknown protection mode(9)
Sep 16 23:05:07 create3 user.notice create-platform: [DBG] [SE]: !velocity_stopped - cmd_vel:4,-4 meas_vel:[INFO] [1694905507.556423011] [motion_control]: Setting safety override to "none"
Sep 16 23:05:10 create3 user.notice ntpd: ntpd: reply from 162.159.200.1: delay 0.144823 is too high, ignoring
Sep 16 23:05:10 create3 user.notice create-platform: 0,0
Sep 16 23:05:10 create3 user.notice create-platform: [DBG] [SE]: !velocity_stopped - cmd_vel:35,-35 meas_vel:0,0
Sep 16 23:05:10 create3 user.notice create-platform: [DBG] [SE]: !wheels_stopped - odom:46807,45813
Sep 16 23:05:10 create3 user.notice create-platform: [DBG] [SE]: !velocity_stopped - cmd_vel:32,-32 meas_vel:0,-12
Sep 16 23:05:10 create3 user.notice create-platform: [WAR] Packet handler running too long exec_time: 282  id: 67
[... cut ...]
Sep 16 23:10:16 create3 user.notice create-platform: [WAR] Packet handler running too long exec_time: 476  id: 19
Sep 16 23:10:16 create3 user.notice create-platform: [WAR] Packet handler running too long exec_time: 18  id: 18
Sep 16 23:10:16 create3 user.notice create-platform: [WAR] Packet handler running too long exec_time: 9  id: 81
Sep 16 23:10:16 create3 user.notice create-platform: [WAR] Packet handler running too long exec_time: 48  id: 18
Sep 16 23:10:16 create3 user.notice create-platform: [WAR] Packet handler running too long exec_time: 925  id: 25
```

hilary-luo, 2023-09-21:
@NachtaktiverHalbaffe Thank you for the detailed breakdown. I agree that the additional nodes are likely adding to the loading on each cpu. Before we look at any safety overrides, this is what I would like you to do. 

On the create3 webserver, go to beta features and [Override RMW Profile](https://iroboteducation.github.io/create3_docs/webserver/rmw-profile-override/). In the field there, paste the following profile with one modification:

```
<?xml version="1.0" encoding="UTF-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <profiles>
        <!-- UDPv4 Transport profile -->
        <transport_descriptors>
            <transport_descriptor>
                <transport_id>udp_transport</transport_id>
                <type>UDPv4</type>
                <!-- Reduce socket buffer size -->
                <sendBufferSize>4096</sendBufferSize>
                <receiveBufferSize>4096</receiveBufferSize>
                <!-- Reduce max message size, otherwise the participant creation fails -->
                <maxMessageSize>4096</maxMessageSize>
                <interfaceWhiteList>
                    <address>XXX.XXX.XXX.XXX</address>
                    <address>127.0.0.1</address>
                </interfaceWhiteList>
            </transport_descriptor>
        </transport_descriptors>

        <!-- Domain Participant Profile -->
        <participant profile_name="domainparticipant_profile_name" is_default_profile="true">
            <rtps>
                <!-- Use user defined UDPv4 transport -->
                <userTransports>
                    <transport_id>udp_transport</transport_id>
                </userTransports>
                <!-- Disable builtin transports -->
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>

        <!-- Default publisher profile -->
        <data_writer profile_name="default_publisher_profile" is_default_profile="true">
            <topic>
                <!-- Tune initial allocations -->
                <resourceLimitsQos>
                    <max_samples_per_instance>20</max_samples_per_instance>
                    <allocated_samples>20</allocated_samples>
                </resourceLimitsQos>
            </topic>
        </data_writer>

        <!-- Default subscriber profile -->
        <data_reader profile_name="default_subscriber_profile" is_default_profile="true">
            <topic>
                <!-- Tune initial allocations -->
                <resourceLimitsQos>
                    <max_samples_per_instance>20</max_samples_per_instance>
                    <allocated_samples>20</allocated_samples>
                </resourceLimitsQos>
            </topic>
        </data_reader>
    </profiles>
</dds>
```

The modification required is to change the `<address>XXX.XXX.XXX.XXX</address>` line to contain the IP address of the create3 over wifi. This can either be found using your router or the about page on the webserver. 

Please stop here and do an analysis again to see if there has been an improvement. 

Next you will need to make a matching change on the turtlebot. We are going to directly modify the file for this test although know that if you update your software it may get overwritten. Modify the contents of `/etc/turtlebot4/fastdds_rpi.xml` to the following, again with a modification:
```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <!-- UDPv4 Transport profile -->
        <transport_descriptors>
            <transport_descriptor>
                <transport_id>udp_transport</transport_id>
                <type>UDPv4</type>
                <interfaceWhiteList>
                    <address>XXX.XXX.XXX.XXX</address>
                    <address>127.0.0.1</address>
                </interfaceWhiteList>
            </transport_descriptor>
        </transport_descriptors>
        
        <!-- Domain Participant Profile -->
        <participant profile_name="turtlebot4_default_profile" is_default_profile="true">
            <rtps>
                <!-- Use user defined UDPv4 transport -->
                <userTransports>
                    <transport_id>udp_transport</transport_id>
                </userTransports>
                <!-- Disable builtin transports -->
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>
    </profiles>
</dds>
```

Once again, the modification required is to change the `<address>XXX.XXX.XXX.XXX</address>` line to contain the IP address but this time with the IP address of the raspberry pi over wifi. This can either be found on the turtlebot4 standard screen display or on the `/ip` topic. 

Once again see if this offers an improvement to the cpu usage.

@garyservin as a first test, can you disable the diagnostics on the Turtlebot and then see if this makes a notable difference in the performance that you have described. (Steps to turn diagnostics off: SSH into the robot and run turtlebot4-setup -> go under ROS Setup -> Bash Setup -> Select TURTLEBOT4_DIAGNOSTICS -> Disabled -> Save -> Back out to the main menu -> Apply Settings -> Yes )

NachtaktiverHalbaffe, 2023-09-21:
# RMW Profile Create3
| **Situation**                             | **With this change**                                   | **Without change**                                   | **Gain (Difference in avg CPU usage)**                               |
| ----------------------------------------- | ------------------------------------------------------ | ---------------------------------------------------- | -------------------------------------------------------------------- |
| Idle (direct after booting)               | 95-100% at start , 54-63% after settled | 70% at start (short time), 55-62% after  settled | 1-3%, but a longer peaking time until it settles to those low values |
| Idle with Nav2 etc. running               | 97-100% at start, 72-77% after settled                 | 100% at start, 76-78% after settled                  | 1-3%, but red light of death with crash reappeared                   |
| Robot driving (until differences show up) | 77-86%                                                 | 80-87%                                               | 10%  (when it didnt crash before)                                                                |
- Crashed multiple time with red light of death during startup
	- More unreliable
	- Crashed more times than successful startups
	- The reintroduction of the crashes were common across all turtlebots and setups
- Takes a little bit longer to idle/settle the CPU usage after startups
	- Sometimes the CPU Usages climb up about 20% for short time (logs only say that ntp reply  is ignored at this time because its delay is too high, so no real explanation for these climbs)
- RAM usage approx. 3 MB higher
- Seems to primary reduce CPU usage caused by messages
- Also got a lot of these warnings:
```bash
Sep 21 15:28:00 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.107399 is too high, ignoring
Sep 21 15:28:36 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [INFO] [1695310116.318682388] [turtle2.system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 44/59 [MB]
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: 1530  id: 25
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 256  id: 68
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 120  id: 67
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 420  id: 19
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [DBG] proto event : PP_NOTIFY_REMOTE_RESYNC
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: proto event : PP_NOTIFY_REMOTE_RESYNC
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 41  id: 14
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 66  id: 30
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 1405  id: 65
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [DBG] proto event : PP_NOTIFY_REMOTE_RESYNC
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: proto event : PP_NOTIFY_REMOTE_RESYNC
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 329  id: 67
Sep 21 15:29:02 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 427  id: 30
Sep 21 15:29:04 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 106  id: 67
Sep 21 15:29:04 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 200  id: 30
Sep 21 15:29:04 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 227  id: 30
Sep 21 15:29:04 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 116  id: 30
Sep 21 15:29:04 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 65  id: 25
Sep 21 15:29:12 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [WAR] Packet handler running too long exec_time: 124  id: [INFO] [1695310152.009546400] [turtle2.system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 44/59 [MB]
Sep 21 15:30:05 iRobot-030B6EE05A104B3AAD28DB9A4069B835 user.notice create-platform: [INFO] [1695310198.191014519] [turtle2.system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 43/59 [MB]
```

# RMW Profile Raspberry Pi
| **Situation**                             | **With this change**                   | **Gain (Difference in avg CPU usage)** |
| ----------------------------------------- | -------------------------------------- | -------------------------------------- |
| Idle (direct after booting)               | 48-54% after settled                   | 1-3%                                   |
[... cut ...]
The Turtlebot Pi RMW profile in the other hand helped at the startup, more or less reliable avoiding 100% average usage at startup (a little below, so could eventually peak at 100% depending on my deviations in the hardware and to which Container the turtlebot is connected). It seems to tackle the problem with the load at startup which seems to be the main problem to tackle at this point (80% when nav2 stack runs seems fine at this point)

Also the common pattern at startup is that the amcl takes the most time until it has localized, after that the rest of navigation2 is quickly up. Seems AMCL is the bottleneck in my case w.r.t. to the time it takes to start nav2

hilary-luo, 2023-09-21:
@NachtaktiverHalbaffe 

Can you try this RMW profile on the create3 now? and see if it deals with the stability issue:
```
<?xml version="1.0" encoding="UTF-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <profiles>
        <!-- UDPv4 Transport profile -->
        <transport_descriptors>
            <transport_descriptor>
                <transport_id>udp_transport</transport_id>
                <type>UDPv4</type>
                <!-- Reduce socket buffer size -->
                <sendBufferSize>4096</sendBufferSize>
                <receiveBufferSize>4096</receiveBufferSize>
                <!-- Reduce max message size, otherwise the participant creation fails -->
                <maxMessageSize>4096</maxMessageSize>
                <interfaceWhiteList>
                    <address>XXX.XXX.XXX.XXX</address>
                    <address>127.0.0.1</address>
                </interfaceWhiteList>
            </transport_descriptor>
        </transport_descriptors>

        <!-- Domain Participant Profile -->
        <participant profile_name="domainparticipant_profile_name" is_default_profile="true">
            <rtps>
                <!-- Use user defined UDPv4 transport -->
                <userTransports>
                    <transport_id>udp_transport</transport_id>
                </userTransports>
                <!-- Disable builtin transports -->
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>

        <!-- Default publisher profile -->
        <data_writer profile_name="default_publisher_profile" is_default_profile="true">
            <topic>
                <!-- Tune initial allocations -->
                <resourceLimitsQos>
                    <max_samples>0</max_samples>
                    <allocated_samples>0</allocated_samples>
                </resourceLimitsQos>
            </topic>
        </data_writer>

        <!-- Default subscriber profile -->
        <data_reader profile_name="default_subscriber_profile" is_default_profile="true">
            <topic>
                <!-- Tune initial allocations -->
                <resourceLimitsQos>
                    <max_samples>0</max_samples>
                    <allocated_samples>0</allocated_samples>
                </resourceLimitsQos>
            </topic>
        </data_reader>
    </profiles>
</dds>
```

Again swapping out the IP address for your create3's IP address over wifi.

NachtaktiverHalbaffe, 2023-09-27:
| **Situation**                             | **With this change**                   | **Gain (Difference in avg CPU usage)**     |
| ----------------------------------------- | -------------------------------------- | ------------------------------------------ |
| Idle (direct after booting)               | 73% at start, 46-52% after settled     | A little bit higher idle and sometimes more volatile (didnt find a pattern which could cause the spikes at idle) |
| Idle with Nav2 etc. running               | 95-100%  at start, 67-72% after settled | No crash/red light of death anymore        |
| Robot driving (until differences show up) | 75-83%                                 |                                            |

Hadnt much time today, so the numbers itself arent that precise. But I restartet the nav2 stack multiple times and it didnt crash today, so the stability is much better in my case.

Also observed that due to the deviations in hardware etc. it still can max out at 100%, but sometimes you can be lucky to be a little bit under which helps a lot. So I would generally say that the startup still maxes out, but "in a stable way"  (and my setup is more heavy than the reference nav2 stack)

sdobbelaere, 2023-09-27:
I'd like to add some feedback here as well.  Overall the robot is responding very delayed - often 30s+ 

OS: (e.g. Ubuntu 20.04). 22.04
ROS Distro: (e.g. Galactic): Humble
Built from source or installed: Installed
Package versions: (if from repository, give version from sudo dpkg -s ros-$ROS_DISTRO-turtlebot4-PACKAGE_WITH_ISSUE, if from source, give commit hash) -> really just make sure you are using the latest of all packages on the robot and on your PC
Create3 Firmware version: H.2.3 (Humble)
Robot Version: (standard or lite): Standard
RMW Implementation: (FastDDS or CycloneDDS): FastDDS
Networking Configuration: (Simple Discovery or Discovery Server): Simple Discovery
Network Information:
Is the robot connected to 2.4GHz or 5 GHz (each of create3 & raspberry pi):  create3: 2.4 / raspberry pi: 5
is it a dedicated router over which you have full control or an institutional network? Dedicated access point - router limited configurability
Is there any other ROS communication on this network? (additional turtlebots or other devices and if so are they on the same ROS Domain or namespaced?) No
Have you applied any custom changes through the create3 beta features? (NTP or DDS profile etc) No
Do you have turtlebot4 diagnostics enabled? No
Finally, can you describe what situations cause the 100% CPU issue and provide the create3 logs from such a situation. Feel free to share other symptoms that you see so we can track them. I want to ensure that I understand what each person is working with to make this process effective.

When taking any action, the cpu-load jumps to 100 and takes around 10s to respond even to simple commands like:` ros2 run teleop_twist_keyboard teleop_twist_keyboard`

When however I want to try to map out the space and run navigation inside of that map things become more challenging.  The response-time jumps to 30s, and more often than not I don't actually see any of the spacial information inside of the map.  
[image]

Scenarios, with and without [navigation](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html#nav2) running

Without navigation running: `ros2 action send_goal /dock irobot_create_msgs/action/Dock "{}"`
-> 7s

With navigation running:` ros2 action send_goal /dock irobot_create_msgs/action/Dock "{}"`
-> 30s

```logs
## Start bot (diagnostics on) idle
Sep 27 17:39:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836393.498958337] [system_health]: CPU usage: max 100 [%] mean 70 [%] RAM usage: 36/59 [MB]
Sep 27 17:40:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836453.485060285] [system_health]: CPU usage: max 100 [%] mean 83 [%] RAM usage: 37/59 [MB]
Sep 27 17:41:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836513.491705057] [system_health]: CPU usage: max 92 [%] mean 75 [%] RAM usage: 37/59 [MB]
Sep 27 17:42:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836573.493129801] [system_health]: CPU usage: max 91 [%] mean 74 [%] RAM usage: 37/59 [MB]
Sep 27 17:43:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836633.494686400] [system_health]: CPU usage: max 88 [%] mean 77 [%] RAM usage: 37/59 [MB]
Sep 27 17:44:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836693.494768125] [system_health]: CPU usage: max 88 [%] mean 79 [%] RAM usage: 36/59 [MB]
Sep 27 17:45:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836753.487647863] [system_health]: CPU usage: max 89 [%] mean 75 [%] RAM usage: 37/59 [MB]
Sep 27 17:46:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836813.487447025] [system_health]: CPU usage: max 93 [%] mean 76 [%] RAM usage: 37/59 [MB]
Sep 27 17:47:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836873.489677711] [system_health]: CPU usage: max 95 [%] mean 78 [%] RAM usage: 36/59 [MB]
Sep 27 17:48:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836933.488616985] [system_health]: CPU usage: max 93 [%] mean 76 [%] RAM usage: 37/59 [MB]
Sep 27 17:49:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695836993.493854332] [system_health]: CPU usage: max 92 [%] mean 75 [%] RAM usage: 36/59 [MB]
Sep 27 17:50:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695837053.494742417] [system_health]: CPU usage: max 91 [%] mean 75 [%] RAM usage: 36/59 [MB]
Sep 27 17:51:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695837113.495857533] [system_health]: CPU usage: max 88 [%] mean 74 [%] RAM usage: 36/59 [MB]
Sep 27 17:52:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695837173.488342678] [system_health]: CPU usage: max 93 [%] mean 75 [%] RAM usage: 37/59 [MB]
Sep 27 17:53:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695837233.488911308] [system_health]: CPU usage: max 93 [%] mean 79 [%] RAM usage: 37/59 [MB]
Sep 27 17:54:53 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695837293.491616844] [system_health]: CPU usage: max 91 [%] mean 74 [%] RAM usage: 37/59 [MB]
[... cut ...]
Sep 27 19:38:38 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695843518.118077836] [system_health]: CPU usage: max 97 [%] mean 81 [%] RAM usage: 39/59 [MB]
Sep 27 19:39:38 iRobot-6F2890FA31C0486E80A20C80E321CE4D user.notice create-platform: [INFO] [1695843578.118348200] [system_health]: CPU usage: max 83 [%] mean 73 [%] RAM usage: 39/59 [MB]
```

Overall the load seems very high, and response-times slow - to the point where I can't really tell if I'm doing something wrong or if this is expected behaviour.

Full post cold restart logs attached as well:
[messages-5.txt](https://github.com/turtlebot/turtlebot4/files/12742697/messages-5.txt)

NachtaktiverHalbaffe, 2023-10-09:
@hilary-luo Any updates?

hilary-luo, 2023-10-18:
@sdobbelaere @NachtaktiverHalbaffe I have not forgotten about this, and I will get back to you although I cannot give you a specific timeline.

marredir, 2023-11-06:
Hi, I have been interacting on this issue through [this ticket](https://github.com/turtlebot/turtlebot4/issues/266). The plan was to contact @hilary-luo via email to start similar tests to the ones mentioned above. But I now have a slightly different setup that I think it is very reliable and hopefully gives us some hints on the problem we are facing. I am running with the discovery server (so create3 and RPI communicates only over USB) on RPI. I have nav2 running on my laptop (so basically RPI has "just" the discovery server) : from what I understand,  we  probably agree that the discovery process is the faulty one (either increasing by far the CPU usage or by "loosing" some info along the way, TBD). What really makes the trick for me is to **_be sure that the ntp server is in sync before starting to operate_** (and by this term I mean SLAM or localization plus quite a lot more of nodes, e.g. vision or others with quite high-speed traffic exploiting tf  and IMU and yes, all running on a high-spec laptop). Look [here](https://iroboteducation.github.io/create3_docs/setup/compute-ntp/) and follow verbatim the instructions: after issuing (from RPI) the command "_sudo chronyc clients_", if I get the message "blabla  **_is too high_**, blabla" then I have problems, either I can't see some topics (e.g. scan or tf) or even if I can see them with "ros2 topic echo" from a shell then it is nav2 that fails to activate some components (be it map server or localization or others..). It might be useful if others could replicate this setup. @hilary-luo: if these discovery problems impact so heavily the create3 CPU can we mitigate them (at least temporarily) via fw having a finer control over the fw? By that the I mean less topics and tune the frequency of them ( I guess most of people use SLAM and localization without caring too much about other topics that are published at high rates). In the old days (I am talking ROS1), the amount of topics to run SLAM and localization was very minimal and even lower rates worked fine.. it is not the best but I think it would alleviate a lot of headaches till we find a more permanent solution. As usual I am available to test and experiment things if needed. Thanks

agorkor, 2023-11-06:
Will create3 have a fix for 100% CPU utilization in near future ?

marredir, 2023-11-06:
@agorkor looking at the  thread there is still quite a lot of uncertainty, and the tickets have been opened for quite a while... but back to the technical problem, also today I had a very reliable session (I added to nav2 ultra sound sensors and it worked reliably for more than 1h): can you guys please check if you see messages like this one you have failures:

"_**user.notice ntpd: Script called in periodic mode**_"

Try to issue the following command from RPI and see afterwards the create3 log from the webserver:
**_sudo chronyc clients_**

Thanks

hilary-luo, 2023-11-08:
@agorkor I can't promise a timeline on a universal fix although depending on specifics of your use case there are things that you can do (discovery server for single robot for example, or disabling diagnostics to decrease system load).

Things keep coming up that have taken me away from being able to focus on this issue although it is high priority for me. 

@marredir thanks for joining the conversation on this ticket and giving an idea to others on what to test. We are limited on what we can do with the create3 firmware given that it is not an open source product. You could mess around with changing the DDS profiles to use that to throttle the topics but I don't think that will be a universal solution that will work for everyone given different people's experience levels.

Simba185, 2023-11-08:
@hilary-luo Thanks for the update.

@marredir Yes my TB4 acts the same, if there is any time sync issue it is erratic, and then a sync issue can occur part way through a test and data will stop working.

For the Turtlebot4 system to be reliable this needs to be resolved.

There needs to be a method or system to control the frequency and importance of messages.  What happens when the WIFI signal is reduced and the bandwidth is reduced or more sensors are added.

marredir, 2023-11-09:
@Simba185 I understand, but do you have the nav2 on a laptop or on the raspberry? I don't have any glitch once I make sure that the system are in sync (before starting all the ROS2 nodes)

@hilary-luo yes I  understand, should we involve the icreate team? I think everyone would benefit from that. Probably you have some direct contact to add to this thread.

Thanks

Simba185, 2023-11-09:
@marredir nav2 is running on the Laptop. (2022 dell with i5 1245u) but system monitor shows only 1 core stuck at 100% (seems like rviz cant use the cores available.)

Then over 3 or 4 hours, with the system running the time sync will get lost and the Create will say "Delay Too high, Ignoring" and then rviz drops all the frames.

@roni-kreinin has stated to switch to from Simple Discovery to server discovery, I will test this and see if it behaves better.

marredir, 2023-11-09:
@Simba185 sorry may be I was not clear: the problem seems to be the Create3 spinning at 100%.  My configuration uses the discovery server and it seems to work reliably, nevertheless we need have a way to "reset" the ntp server. It seems that a "restart application" doesn't work. If you look [here](https://iroboteducation.github.io/create3_docs/setup/compute-ntp/), at point 10 it says that if the ntp has a too large gap "If this happens, simply restart the robot (not just the application) via the webserver over the network connection." but to shut everything down is not possible from the webserver when in discovery mode (that button seems to be available only when the base is in AP mode) and it would be anyway very unconvenient. @hilary-luo  can we contact iRobot and make something about it? May be if we "reliably" are able to restart the ntp the system might recover (still not a full solution but we would get the system more reliable). Anyone at iRobot that might get involved?

NachtaktiverHalbaffe, 2023-11-18:
Sorry for replying lately. Our usecase doesnt allow using discovery server. We had NTP issues, sometimes we get the "delay is too high" logs, but these are just increasing the usage for a short time and not causing a complete hang up or 100% utilization, this only occurs due to the node discovery problem. We have also have a special NTP config because the firewall of the lab is blocking all ntp servers except the one from university. Maybe this NTP problems werent that erratic in my case because I never launched the nav2 stack directly after the robot has booted so it had plenty time to sync time etc.

The both fixes from @hilary-luo which I tested (disabling diagnostics and custom dds configs) and the DDS fix from https://github.com/turtlebot/turtlebot4/issues/216#issue-1797043215 improved the situation so that the Turtlebot4 wont hang up completely anymore (only one Turtlebot per ROS Domain, each Turtlebot is isolated and the important topics are bridged). Since these fixes the Nav2 stack has started reliably again in our usecase

I finished my master thesis and thus I dont have unrestricted access to the Turtlebot4 anymore. I might do voluntary work on this topic, but I added my supervisor @MatWise to this issue as well who will be using my model process with the Turtlebot4 in the future

xiaoshan-lin, 2023-12-15:
I recently had a similar issue: 100% CPU usage and turtlebot took around 30 seconds or even longer to respond to /cmd_vel commands. At the beginning I thought the slow response was caused by 100% CPU usage, until I ssh into turtlebot and ran ros2 topic info /cmd_vel. I found the turtlebot is not even listening to the /cmd_vel. On my PC, I ran ros2 topic info /cmd_vel and found turtlebot is listening to /cmd_vel, even though it's actually not. 

My solution is:
1. ssh into the turtlebot and run ros2 topic info /cmd_vel to check if turtlebot is listening to the /cmd_vel topic. If not, run turtlebot4-daemon-restart
2. on PC, run ros2 topic info /cmd_vel. If there's no subscription, run ros2 daemon stop; ros2 daemon start

My turtlebot now works fine and doesn't have noticeable latency. Note: CPU usage still reach 100% after running the robot for some time, but it seems to have no issues.

hilary-luo, 2024-02-15:
@NachtaktiverHalbaffe congrats on finishing your thesis!

It is definitely an interesting thing to note that some people have had success using separate ROS Domain IDs and bridging certain topics. I want to bring some emphasis to that point for people who may be coming across this. This should also be able to be done using the same ROS domain ID and modifications to the DDS profiles but bridging the domain would definitely be easier to set up. 

@marredir You are able to restart the NTP daemon under beta features (if you don't have this option, update your firmware). We are actively engaged with iRobot even if they are not directly present in this thread and there are things coming down the line although I can't make any promises about dates or features. You mention that you build everything on top of a standard ubuntu image. Are you building everything from source? or have you tried running from installs? In regards to Zenoh, it is definitely on our radar for the future. If you end up testing this out feel free to provide feedback back and/or PRs.

Overall I am hoping to have some new tests for people to try out shortly but please do keep sharing what you find works / doesn't work for you.

Simba185, 2024-02-15:
@hilary-luo 

Thanks for the update, 

Really looking forward to seeing what can be done to mitigate these issues and how to move forward. ( we really need a solution as TB4 is not reliable in this current state).

Thanks

hilary-luo, 2024-02-15:
@Simba185 Did you get discovery server set up on your system? I don't think I heard feedback on how that went for you. (In the meantime, I do want to ensure that you have tried the work around tactics that have helped other people.)

garyservin, 2024-02-15:
Just to add an extra data point, I was able to successfully run SLAM and navigation on my TB4 after configuring the discovery server on the TB. 🎉

Simba185, 2024-02-15:
@hilary-luo 

Unfortunately no further time has been allotted to TB4 once we saw that there was lots of unresolved issues with time sync and these CPU load issues. 

I'm glad to hear Discovery server works for many as a solution for smooth SLAM (with simple discovery the robot will stop part way and continue). 

Until I am confident that there are 100% solutions for all these issues I'm not sure when I will have more time to work on any tests.  (as the consumer , I should not have to test solutions for the Manufacture.)

I look forward to any updates, Thanks.

hilary-luo, 2024-02-29:
Hey all, iRobot has released a new [create3 firmware (H.2.5)](https://iroboteducation.github.io/create3_docs/releases/h_2_5/) which should feature some improvements for this issue. Please try it out and let me know how it impacts your system, the more quantitative the feedback the better. I expect you should see improvements for both those using discovery server as well as simple discovery.

ooijenvangilbert, 2024-03-03:
I see that I am a bit late to the party, I've found out that my setup also runs into the 100% and red ring situation.
But my setup differs, I am on ubuntu 20.04 and running galactic with the create 3 running g5.4 version.

I have 2 rPi setups, one with the image from the Turtlebot 4 and one own build, my own build starts lidar en oak-d and I am able to create maps and navigate without problems. This setup does not use the usb communication between rPi and create and basically works fine.

It's the Turtlebot setup that uses the usb-c communication that runs the create 3 into trouble. I am curious as to what is the added value of running communication over the usb-c line (both create 3 and rPi have wifi and work fine on their own)?

Simba185, 2024-03-07:
@ooijenvangilbert Can you provide some further details about how you built your own image for TB4? How is the commands sent to the base without usb communication?

Has anyone created a ROS1 setup?

ooijenvangilbert, 2024-03-08:
> @ooijenvangilbert Can you provide some further details about how you built your own image for TB4? How is the commands sent to the base without usb communication?
> 
> Has anyone created a ROS1 setup?

basically the create 3 also has a wifi connection and therefor can handle all commands for movement.

I also used this repository 
[https://github.com/iRobotEducation/create3_examples ](url)

and added the ros implementation for the Luxonis cam to get it all working.

JJJ2777, 2024-04-05:
Hello, I have the same problem with the CPU at 100% and a red light. I am trying to figure out what is causing the problem. Do you use one of these two commands:

`ros2 run teleop_twist_keyboard teleop_twist_keyboard`

`ros2 launch turtlebot4_bringup joy_teleop.launch.py`

Do your CPUs increase after these commands or using Navigation2 in RViz? And if that's the case for you, which one courses the highest increase?

hilary-luo, 2024-05-06:
@ooijenvangilbert 
>I am curious as to what is the added value of running communication over the usb-c line (both create 3 and rPi have wifi and work fine on their own)?

If you have a setup where the wifi connection on the create3 works for you then there is no need to have the usb-c connection. Earlier in this ticket, one of the configurations that I was having people try was indeed blocking the usb-c communication. This will be easier to control in Jazzy, unfortunately it is harder to configure in Humble. However, there are many situations where it is desirable to disconnect the create3 from the wifi and only use the USB-C connection (what is done in the discovery server setup). Some networks do not handle 2.4GHz to 5GHz connections gracefully which can cause issues. Additionally, scaling up with many robots works best with the create3 only communicating over usb, not over wifi.

hilary-luo, 2024-05-06:
Is anyone still having the CPU at 100% issue using Humble firmware 2.6? I am going to be looking for some people to give feedback on the next round of updates and I want to understand the use cases that are still having issues and how the new update will work (or not) for your use case.

mauriz, 2024-05-07:
Hi, I can not run these experiments as I have switched to Zenoh (which seems to be extremely stable, I can shutdown and resume the system many times without a glitch). In this regard, do you have a roadmap of what you are going to support? Do you plan to stick with Humble for a while or do you plan to move to successive versions (e.g. Iron or similar)? Thanks

hilary-luo, 2024-05-07:
@mauriz I cannot speak to any long term road mapping. I will say though that as long as Zenoh is not officially supported by ROS, it is very unlikely to be supported here.

That being said, I would still encourage you to share your experience with Zenoh, how you are using it and how it has helped (maybe as a discussion topic).

ooijenvangilbert, 2024-06-11:
> @ooijenvangilbert
>
>
> If you have a setup where the wifi connection on the create3 works for you then there is no need to have the usb-c connection. Earlier in this ticket, one of the configurations [...]

That would be my next option to test to have the create3 only use USB-C so the RPI can send cmd_vel commands without the use of wifi. Did shutting down the wifi on de create3 releave any problems as described?

ooijenvangilbert, 2024-06-11:
> That would be my next option to test to have the create3 only use USB-C so the RPI can send cmd_vel commands without the use of wifi. Did shutting down the wifi on de create3 releave any problems as [...]

so you mention that to have communication over USB-C we have to do the discovery server setup. But I have just flashed the standard setup and when I run a Ros2 topic list it gives errors when the usb-c is not connected to the create3.
This would imply that even without discovery server there is communication ...

hilary-luo, 2024-06-13:
@ooijenvangilbert you don't need discovery server to have communication over the usb-c connection, in discovery server you ONLY have connection over the usb-c connection so it is required. That answer was addressing the question of why communication is allowed over usb-c.

In simple discovery it should be able to communicate over wifi or usb-c and if you are not able to connect without the usb-c connection then you need to take a look at your wifi connections / bridging between networks.

royito55, 2024-09-10:
Hi,

We have made a video explaining why this issue happens, and how it can be solved by simply pulling and running a docker container in your TurtleBot 4.

I hope it helps! 

https://www.youtube.com/watch?v=xmK2I0D5sas

closed 2024-09-16 by hilary-luo