turtlebot/turtlebot4 #267: RPLIDAR installation issue

Akalpita30, 2023-09-11:
**Please provide the following information:**
 - OS: Ubuntu version 22.04
 - ROS Distro: Humble
 - Built from source 
 - Package version: H.2.2

 **Actual behaviour**
 turtlebot4_bringup not found: 
 tried installing with "sudo apt update && sudo apt upgrade" still shows the same error. 

**To Reproduce**
Provide the steps to reproduce:
1. sudo apt install ros-humble-rplidar-ros
2. ros2 launch turtlebot4_bringup rplidar.launch.py
3. see the error

**Other notes**
Moreover "ros2 service call /stop_motor std_srvs/srv/Empty {}" doesn't stop the motor.

Akalpita30, 2023-09-11:
[image]

hilary-luo, 2023-09-12:
Are you trying to launch the lidar on the robot? and is the robot software in the original state from factory and/or flashed from the provided turtlebot 4 raspberry pi images?

As background information, why are you directly launching the lidar? If you are trying to do normal operation of the robot then you do not need to launch it directly, it will launch on its own as part of the turtlebot service. I assume this has been operating correctly based on you having seen and tried to use the stop motor service. 

In regards to stopping the motor, make sure that you are running version 2.1.4 (or higher) of the rplidar driver. There was a recent fix that has to do with stopping the lidar. Additionally, if you want to manually control the lidar starting / stopping you also need to diable the feature which automatically starts / stops the motor based on the number of subscribers. Information about that can be found in my response here: https://github.com/turtlebot/turtlebot4/issues/233#issuecomment-1660515345

Akalpita30, 2023-09-12:
it is in the original state of the factory. 
I tried to get the localization scan from the robot; it kept giving inf, and the robot could not avoid the obstacles.

hilary-luo, 2023-09-13:
Am I to understand correctly that the core issue that you would like to address is the fact that when you echo the scan topic, the values you receive back are all infinite? and that it was not avoiding the obstacles?

If so, then can you share the following details:
1. What were you doing when you saw that the robot was not avoiding the obstacles?
  a. Were you running Nav2 with SLAM or with localization? - what exact commands did you use to launch these?
  b. Were the obstacles that the robot ran into static or dynamic? and were they in the map or not?
2. How did you observe these inf values? - are you sure that all of the values are infinite vs some of them being infinite?
3. Can you see any lidar hits in rviz?

Akalpita30, 2023-09-14:
yes, you got that right.

1a. I am using Nav2 with localization. cmd- ros2 launch turtlebot4_navigation loacalization.launch.py map: ; ros2 launch turtlebot4_navigation nav2.launch.py params_file
1b. i tried with both static and dynamic. it shows on map but the robot instead to avoiding goes and bumps into them.

2. i used ros2 topic echo /scan- it gives inf values

3. i can see them 
[image]

Akalpita30, 2023-09-15:
@hilary-luo do you have any suggestions about the problem

Akalpita30, 2023-09-18:
[bag.zip](https://github.com/turtlebot/turtlebot4/files/12652396/bag.zip)

hilary-luo, 2023-09-19:
@Akalpita30 Did those bag files upload correclty? three of them only have one topic and the other isn't showing any topics. 

Can you also provide the [create3 log](https://iroboteducation.github.io/create3_docs/webserver/logs/) from when you replicate this situation?

Akalpita30, 2023-09-19:
create 3 -
[messages(4).txt](https://github.com/turtlebot/turtlebot4/files/12666313/messages.4.txt)

bag-
[bag09_19.zip](https://github.com/turtlebot/turtlebot4/files/12666315/bag09_19.zip)

hilary-luo, 2023-10-03:
Thank you for your patience, the past couple of weeks have been very busy. It does seem like you are running into the issue of the Create3 CPU maxing out. 

Are you needing to run multiple robots together? If not then I would suggest trying to switch to [discovery server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html) and see if this helps. Let me know how this goes or if this is not an option for you.

Akalpita30, 2023-10-09:
no, I am not running multiple robots at once. 
i tried switching to the discovery server, the ros topic list did not publish the topics. rviz was blank nothing popped up to rviz. 
let me know if there is another way to solve the issue

Akalpita30, 2023-10-10:
@hilary-luo any updates? 
i tried running rviz and observed it going and colliding with the wheels of the chairs. I would appreciate it if you could help me with the parameters i should tune to stop that from happening.

Akalpita30, 2023-10-11:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
/
[messages(4).txt](https://github.com/turtlebot/turtlebot4/files/12875522/messages.4.txt)

@hilary-luo @roni-kreinin i would appreciate your help on this issue as soon as possible. Attached here are the logs, processes I did and the results I got. The turtlebot4 and the remote computer is connected to the WiFi- TP-Link. I also changed the server to discovery.

Akalpita30, 2023-10-12:
The video of the robot colliding with the chair.

hilary-luo, 2023-10-12:
Thank you for your patience. I don't see any unexpected operation in the video and path planning images. I will explain why and then feel free to let me know if you feel like there is still something that is unexplained. 

First off, the lidar is a 2d lidar (as are most lidars that are used for indoor navigation) however, this means that it can only view obstacles on a single plane. These obstacles are denoted by the black pixels in the map or I think in your visualization, the fuschia pink of your costmap. The surrounding blue in the costmap is an envelope area around the obstacles that the centre of the robot will avoid and as the colour changes moving away from the obstacles it will avoid it less and less. (Look up costmap based path planning if you are interested in understanding this further). If you look carefully at your map, only the centre column of the chair is visible to the lidar and is in the map. This means that the robot has no knowledge of the wheels being there. The chair wheels are below the lidar plane and the chair seat is above the lidar plane. 

It is standard for lidar based navigation, that environments have to be prepared such that the lidar is able to see all obstacles. One way to evaluate this is to measure the height of your lidar and then go around your environment and measure the height of all of the obstacles. When doing this it is also good to look out for excessively shiny surfaces, surfaces that do not reflect any light, very thin surfaces and windows which are all difficult to detect with lidars. Underhanging and overhanging surfaces can be extended to the lidar plan and or blocked with items that the robot can see. A good visualization of this issue is shown here: https://docs.ottomotors.com/en/30495-30508-overhang-and-underhang-detection.html. Knowing this, if you look at the plan that the robot made and attempted to follow, it made complete sense given its knowledge of the environment. 

All that being said, try making this change to address the time sync issue that is showing up in the terminal: https://github.com/turtlebot/turtlebot4/issues/216 . Let me know how that works.

nidia-bucarelli, 2023-10-12:
Hello Hillary,
Thank you for your answer. I understand the LiDAR reasoning. But, What about the IR sensors? The sensors at the bumper. They at the height of the wheels. Should not this work?
Based on the configuration below, we understand they are also working.

This is the Nav2.yaml file. 
```
bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: rplidar_link #base_link
    odom_topic: odom
    bt_loop_duration: 10
    default_server_timeout: 20
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.
    plugin_lib_names:
    - nav2_compute_path_to_pose_action_bt_node
    - nav2_compute_path_through_poses_action_bt_node
    - nav2_smooth_path_action_bt_node
    - nav2_follow_path_action_bt_node
    - nav2_spin_action_bt_node
    - nav2_wait_action_bt_node
    - nav2_assisted_teleop_action_bt_node
    - nav2_back_up_action_bt_node
    - nav2_drive_on_heading_bt_node
    - nav2_clear_costmap_service_bt_node
    - nav2_is_stuck_condition_bt_node
    - nav2_goal_reached_condition_bt_node
    - nav2_goal_updated_condition_bt_node
    - nav2_globally_updated_goal_condition_bt_node
    - nav2_is_path_valid_condition_bt_node
    - nav2_initial_pose_received_condition_bt_node
    - nav2_reinitialize_global_localization_service_bt_node
    - nav2_rate_controller_bt_node
    - nav2_distance_controller_bt_node
    - nav2_speed_controller_bt_node
    - nav2_truncate_path_action_bt_node
    - nav2_truncate_path_local_action_bt_node
    - nav2_goal_updater_node_bt_node
    - nav2_recovery_node_bt_node
    - nav2_pipeline_sequence_bt_node
    - nav2_round_robin_node_bt_node
    - nav2_transform_available_condition_bt_node
    - nav2_time_expired_condition_bt_node
    - nav2_path_expiring_timer_condition
    - nav2_distance_traveled_condition_bt_node
    - nav2_single_trigger_bt_node
    - nav2_goal_updated_controller_bt_node
    - nav2_is_battery_low_condition_bt_node
    - nav2_navigate_through_poses_action_bt_node
    - nav2_navigate_to_pose_action_bt_node
    - nav2_remove_passed_goals_action_bt_node
    - nav2_planner_selector_bt_node
    - nav2_controller_selector_bt_node
    - nav2_goal_checker_selector_bt_node
    - nav2_controller_cancel_bt_node
    - nav2_path_longer_on_approach_bt_node
    - nav2_wait_cancel_bt_node
    - nav2_spin_cancel_bt_node
    - nav2_back_up_cancel_bt_node
    - nav2_assisted_teleop_cancel_bt_node
    - nav2_drive_on_heading_cancel_bt_node

bt_navigator_navigate_through_poses_rclcpp_node:
  ros__parameters:
    use_sim_time: True

[... 218 lines cut ...]
  ros__parameters:
    use_sim_time: True
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

velocity_smoother:
  ros__parameters:
    use_sim_time: True
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.26, 0.0, 1.0]
    min_velocity: [-0.26, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0
```

hilary-luo, 2023-10-12:
The IR sensors do not integrate into Nav2 by default. Nav2 is only using the lidar data. In regards to the nav2 config you have provided, the obstacle layer / IR obstacle section that has been added is an interesting modification. The create3 IR sensor information is published under the `/ir_intensity` topic and with type `irobot_create_msgs/msg/IrIntensityVector`. Based on this config it would seem that you are trying to interpret the IR sensors as laserscans off of an `/ir_obstacle` topic. This would require custom code running to create the new topic, translate the data into the laserscan format and then publish the data. This in addition to thorough testing to make sure that the obstacle layer plugin has been configured properly. If these modifications aren't working how you or your team intended, then you should revisit how that data is being published (or even that it is being published) and how that does or does not translate into Nav2.

Akalpita30, 2023-10-13:
Thank you for bringing this issue to notice. We forgot to run the code for ir_intensity.  We are using a custom code for ir_intensity.  We are running the custom code using the command - ros2 launch tb4_ir obstacles.launch.py. 

After doing this we observed that the robot was updating the position of small obstacles (the pink-ish) is showing but it kept on trying to navigate through it. There were instances where the robot would back up and try to navigate through the obstacle again. I am attaching the video below showing the collision. We are not sure if there is a parameter under Nav2 that we need to configure so it does not go through it. 

Would you please elaborate on your comment about the "obstacle layer plugin has been configured properly." and "then you should revisit how that data is being published (or even that it is being published) and how that does or does not translate into Nav2."

here is the video:

This is how the rviz screen looks like:
[image]
[image]

This is the custom code we are using
[tb4_ir.zip](https://github.com/turtlebot/turtlebot4/files/12899776/tb4_ir.zip)

This is the nav2.yaml file. 
[nav2.zip](https://github.com/turtlebot/turtlebot4/files/12899779/nav2.zip)

hilary-luo, 2023-10-15:
Unfortunately I don't have the time to delve into this too deeply. Since it does sound like the turtlebot4 is working properly, I would encourage you to reach out on [robotics stack exchange](https://robotics.stackexchange.com/?newreg=b1356041b96546379a29dc15be5fa786) or on the nav2 repo to better understand how that obstacle layer plugin should work, and whether or not you have set it up correctly.

closed 2023-11-16 by hilary-luo