turtlebot/turtlebot4 #268: oakd launch error

ramakrishnasakethi, 2023-09-12:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 - Built from source or installed: Installed
 - Package version: H.2.2

 **Expected behaviour**
Camera getting operational.

 **Actual behaviour**
ubuntu@ubuntu:~/turtlebot4_ws/src$ ros2 launch turtlebot4_bringup [oakd.launch.py](http://oakd.launch.py/)
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-09-12-08-35-17-651955-ubuntu-2499
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch_ros.actions.node]: Error while expanding or validating node name or namespace for 'package=rclcpp_components, executable=component_container, name=oakd_container, namespace=<launch.substitutions.launch_configuration.LaunchConfiguration object at 0xffffb38f28c0>':
[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'namespace' does not exist.

**To Reproduce**
ros2 launch turtlebpt4_bringup oakd.launch.py

**Other notes**
I did not give the robot any name.

hilary-luo, 2023-09-12:
You shouldn't need to launch the oakd separately (it is launched automatically). If you are specifically wanting to shut the turtlebot service down and launch the oakd camera alone then you may be better off launching `oakd_example.launch.py`. That or creating your own launch file that creates and passes in the namespace launch argument as blank, to oakd.launch.py. The oakd.launch.py file was not intended for the user to be launching directly.

If you look into this oakd.launch.py you can see that it is expecting a namespace parameter to be passed in. For the standard turtlebot, this is passed down from the standard.launch.py file. You can pass it in using the command line if you namespace your robot, but it isn't possible (to my knowledge) to pass in the parameter as blank from the terminal.

ramakrishnasakethi, 2023-09-14:
I still do not see the camera msgs under the list of available topic in both pc and pi.
[image]

A screenshot of the bot status after ssh'd into the bot and typing in `sudo systemctl status turtlebot4`
[image]

hilary-luo, 2023-09-14:
Is the robot docked? There is a powersaving feature that turns off the camera while docked.

ramakrishnasakethi, 2023-09-15:
True. Once I undock, I am able to view oakd topics. Thank you.

closed 2023-09-15 by ramakrishnasakethi