turtlebot/turtlebot4 #269: Turtlebot 4 Lite can't detect it's battery

SonOfFather, 2023-09-12:
- OS: Ubuntu 20.0 Focal Fossa
 - ROS Distro: Galactic
 - Built from source or installed: pre-installed/shipped version

 **Expected behaviour**
 I just wanted to setup the Turtlebot 4 Lite and teleop it via my remote PC (focal fossa and galactic). I've already set  up one tb4 before with exact same PC and versions

 **Actual behaviour**
I am currently working on a Turtlebot4 Lite at my institute and having issues to get it going. I followed the instructions on your GitHub User Manual to set it up. This was also my second set up, as I already successfully started another Turtlebot4. I connected it to my Wi-Fi and tried to teleoperate the bot via my remote PC. Both of them are running on focal fossa and ROS2 Galactic. It seems like the ROS2 connection works, because I can ping each device from each other. Also I can see the full ros2 topic list on both devices, so I assumed the ROS communication is not the problem. After that, I started the ROS2 diagnostics (ros2 launch turtlebot4_viz view_diagnostics.launch.py) and the RQT-Monitor showed several errors. One of them was a critical battery error suggesting I should immediately charge the battery. There were many more such as “wheels disabled” but I guess most of them are caused by the critical battery error in order to protect the bot. The ICreate3 platform indicated that the battery was fully charged and it obviously was because the lidar sensor and lights were running over several hours without any power source connected but the build in battery.

So I shh into the Raspberry PI of the Turtlebot and tried to run upower -e, copied the path and plugged it in with upower -i /path/to/battery (according to upower -e). The output shows that it can’t detect no battery….

 

power supply:                          no

updated:                                   Thu 07 Sep 2023 8:26:54 Am UTC (86 Seconds ago)

has history:                               no

has statistics:                           no

unknown

     warning-level:                    none

     icon-name:                          ‘battery-missing-symbolic’

 

I already tried to let the robot run into battery death without charging in hope that the system would recognize the battery after it ran fully out of charge. That didn’t work, still no detection of the battery.

Next I opened the bottom plate of the robot and disconnected the battery for 15 minutes and put it back in with the same result as my other attempts to figure this issue out.

 

Do you have any clue what could cause this? The next thing I’d try would be to reset the robot to factory settings, but I could not find any reliable instructions for that.

**Other notes**
I attached the logs from the ICreate3, it seems that it has difficulties to connect to the wifi, but i ensured that i entered the right credentials and also i can access the webserver which should not be possible without internet connection, right?

Another confusing thing under the about tab in the system information of the i create3 the IP Adress is unkown which seems kinda odd to me.

System Info

>>>>>>> system IP address

unknown

 

>>>>>>> system version

G.5.2

 

>>>>>>> system sku

RCi3099

 

>>>>>>> system battery

Percentage:74

Capacity(mAh):2037

 

>>>>>>> system properties

   PRODUCT         : create3

   SERIAL          : e17628

   CLEAN MODE      : normal

   CONNS SVC MODE  : normal

   AP RADIO        : wlan1/g/6/[SHORT-GI-20][GF]/Roomba-1234567890123456/US

   AP IP ADDR      : 192.168.10.1

   CL IP           : dhcp/#/#/#

   CL DNS          : #/#/#

   CL DHCPCD T/O   : 3

   CL REG DOMAIN   : ETSI

   ETH0 IP         : static/192.168.186.2

   USB0 IP         : 192.168.186.2/192.168.187.2/192.168.186.3

   SYSTEM_ACCESS   : locked

   DEVELOPMENT MODE: disabled

   RADIO_LIST      : wlan0 wlan1

   HOSTNAME        : iRobot-65B060DDED594BE997292F80E1115703

   ROBOT ID        : 65B060DDED594BE997292F80E1115703

   BOOT COUNT      : 20

   LANGUAGE PACK   : unknown

   EXTRA LANGUAGE  : not_installed

 

 

>>>>>>> system date

Thu Nov 10 23:48:31 UTC 2022

 

>>>>>>> system uptime

23:48:32 up 49 min,  load average: 1.69, 2.05, 2.05

 

>>>>>>> system memory

              total        used        free      shared  buff/cache   available

Mem:          59080       25732        7124         316       26224       31372

Swap:             0           0           0

 

>>>>>>> system ifconfig

lo        Link encap:Local Loopback 

          inet addr:127.0.0.1  Mask:255.0.0.0

          UP LOOPBACK RUNNING  MTU:65536  Metric:1

          RX packets:736 errors:0 dropped:0 overruns:0 frame:0

          TX packets:736 errors:0 dropped:0 overruns:0 carrier:0

          collisions:0 txqueuelen:1000

          RX bytes:53983 (52.7 KiB)  TX bytes:53983 (52.7 KiB)

 

usb0      Link encap:Ethernet  HWaddr 56:D8:08:21:B7:1C 

          inet addr:192.168.186.2  Bcast:192.168.186.255  Mask:255.255.255.0

          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

          RX packets:48093 errors:0 dropped:8 overruns:0 frame:0

          TX packets:9016 errors:0 dropped:0 overruns:0 carrier:0

          collisions:0 txqueuelen:1000

          RX bytes:5747357 (5.4 MiB)  TX bytes:3035892 (2.8 MiB)

 

wlan0     Link encap:Ethernet  HWaddr 4C:B9:EA:2E:92:A4 

          UP BROADCAST MULTICAST  MTU:1500  Metric:1

          RX packets:91340 errors:0 dropped:0 overruns:0 frame:0

          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0

          collisions:0 txqueuelen:1000

          RX bytes:29530067 (28.1 MiB)  TX bytes:0 (0.0 B)

          Interrupt:6

 

 

>>>>>>> system hciconfig
  
[messages.txt](https://github.com/turtlebot/turtlebot4/files/12584295/messages.txt)

roni-kreinin, 2023-09-13:
@SonOfFather The battery of the robot is inside the Create 3 and not directly connected to the RPi4. You can check the battery status by echoing the `battery_state` topic in ROS 2. Regarding the webserver, how are you accessing it?

SonOfFather, 2023-09-13:
I did that too, and from my remote PC I could see that there is a battery and it's status, which looked accurate and plausible. But echoing battery_state from the Turtlebot I didn't see anything.

I accessed the webserver via entering the IP address of the raspberry pi plus the 8080 port. And on the webserver I entered my wifi credentials

roni-kreinin, 2023-09-14:
The webserver gets forwarded through the RPi on the 8080 port, so it is possible to access the webserver this way without the Create 3 being connected to Wi-Fi. The issue seems to just be that the Create 3 is not connecting to your Wi-Fi network. I would recommend placing the Create 3 into AP mode by pressing and holding buttons 1 and 2 on the Create 3 for a few seconds. The lightring will turn blue and you should see a `CreateXXXX` Wi-Fi network appear. Connect to it and access the webserver by entering `192.168.10.1` into your browser. Then attempt to connect the robot to your network again.

SonOfFather, 2023-09-15:
@roni-kreinin Yeah, that worked fine, thank you. Now I feel stupid for not trying that 😭😂

closed 2023-09-15 by roni-kreinin