turtlebot/turtlebot4 #270: PC and Turtlebot4 communication issue

Kazimbalti, 2023-09-14:
Hello all!

I have some issues with the communication of PC and Turtlebot4.

For example, when I do the ros2 topic list on PC, it doesn't show the topics of create3 while when I go to the turtlebot4 by ssh from PC it shows all the topics of turtlebot4 and create3. As shown in Fig below
[image]
Turtlebot4 and PC communicate at the rate of 5GHz while create3 works at the rate of 2.4GHz.

I want to implement SLAM and NAV2 but am not able to do that, so please help me to resolve this issue.
thank you

ipa-rar, 2023-09-14:
@Kazimbalti To debug this,
-  Ensure that create3 is connected to 2.4Ghz Wifi and Rpi to the 5.8 Ghz Wifi band. 
- Check if Domain ID is same across create3, RPi4 and your PC. 
- Check if your terminal is sourced properly, i.e ``source /opt/ros/distro/setup.bash`` -> I suspect this in your case

Kazimbalti, 2023-09-15:
thank you @ipa-rar for your response and guidance.
I already did everything you mentioned:

- create3 is connected with 2.4GHz and RPI is connected with 5GHz band
- Domain ID of creat3=RPI=PC = 0
- I properly sourced it, on both PC and on RPI4 

Still, I am not able to get the topics on my PC, please guide me, is there anything I am missing? thank you

hilary-luo, 2023-09-18:
@Kazimbalti can you please clarify the folowing information:

- OS / Containerization info:
- ROS Distro:
- Built from source or installed?:
- Package version:
- Create3 Firmware version:
- Networking configuration (simple discovery or discovery server):
- RMW Implementation (CycloneDDS or FastDDS):

Also, are you connecting your system to a dedicated router using the default functionality? are there any firewalls that could be present, blocking communication. 

Finally, is this issue consistent even over restarting the robot and your PC and the ros2 daemon on the pi (`ros2 daemon stop; ros2 daemon start`)?

Kazimbalti, 2023-09-19:
- OS / Containerization info: Ubuntu 22.04.3
- ROS Distro: Humble
- Built from source or installed: Source 
- Package version: H.2.2
- Create3 Firmware version: H.2.2
- Networking configuration: simple discovery
- RMW Implementation: FastDDS

[image]

Yes, I have connected the RPI with 5GHz while the create3 is connected with 2.4GHz.
[image]

[image]

In connect to Robot to Wi-Fi it doesn't show IP address but from the application when I send commands it works like restart etc,
[image]

There are no firewalls present/active,
[image]

I restarted the PC and turtlebot4 many times, 

This is my ~/.baschrc file
[image]

I have been working on it for two weeks, I even tried on Galactic, and same issue there as well.
When I will give any commands or command velocity from RPI/ubuntu, it works fine but when I give any commands from PC it doesn't respond because it doesn't get any topics from create3. Please help me resolve these issues, I have two turtlebot4 and both have the same issue. I want to implement SLAM, NAV2 etc, 
thank you

hilary-luo, 2023-09-19:
Your create3 is not connected to the wifi. You are seeing some topics show up with no messages because the diagnostics on the raspberry pi are subscribing to those topics. You are connecting to the create3 webserver through the pi as well. 

In order to fix the create3 wifi connection, please hold the • and •• buttons for a few seconds until you get the spinning blue / cyan light. and then either connect to the create3 webserver using the AP that shows up or through the raspberry pi and reconfigure the wifi connection. The connect tab should show you a drop down of available networks and the SSID should be filled in on that page after it connects successfully. If you are unsuccessful following this please provide the create3 log.

closed 2023-10-12 by hilary-luo