turtlebot/turtlebot4 #272: RPi Topics do not appear on remote PC

MikeDegany, 2023-09-15:
- OS: Ubuntu 22.04.3
 - ROS Distro: Humble
 - Built from source or installed: Installed
 - Package version: H.2.2

 **Expected behaviour**
Expected to see same amount of topics (Create3 and RPi topics) on desktop and turtlebot4. 

 **Actual behaviour**
```ros2 topic list``` on the Remote PC can only see create3 topics, however no messages are published on these topics. 

**To Reproduce**
Provide the steps to reproduce:
```ros2 topic list``` on RPi (ssh from PC) pictured left, ```ros2 topic list``` on PC pictured right 
[image]

**Other notes**
We have connected the RPi, Create3, and PC to the same network (RPi on 5GHz band and Create3 on 2.4GHz band). Both the RPi and create3 have an IP address and we can ping both addresses from the remote PC. 

We have tried restarting the daemon with ```ros2 daemon stop ; ros2 daemon start``` on the remote PC and ```turtlebot4-daemon-restart``` on the turtlebot, however this did not solve our issue.

Kazimbalti, 2023-09-18:
I have the same issue and I posted it here but no response yet.
https://github.com/turtlebot/turtlebot4/issues/270

I am still waiting for the response, if you are able to get the ros2 topic list of create3 on your PC then please let me know as well. thank you

hilary-luo, 2023-09-18:
@Kazimbalti although your issues appear similar on a surface level they are different. You are seeing a subset of turtlebot and create3 topics while @MikeDegany is seeing only the create3 topics. 

@MikeDegany You may have issues with the bridging between the 2.4GHz and 5 GHz networks that isn't letting all of the data through. Can you please test the multicast functionality across from the raspberry pi to the pc? You can do so by first running `ros2 multicast receive` on the PC and then running `ros2 multicast send` on the raspberry pi. You should see `Received from xxx.xxx.xxx.xxx:xxxxx: 'Hello World!'` show up on the PC. If this is successful, please try the opposite direction for good measure. 

Something else to test would be to see if you wire directly into the router, are you able to see all of the topics then?

MikeDegany, 2023-09-18:
@hilary-luo It sounds that multi-casting works well. Do I need to turn off the WiFi for testing wired(LAN) connection. Generally, how computers manage network connections between WiFi and Ethernet?

Kazimbalti, 2023-09-19:
@hilary-luo thank you for the response,
Yes @MikeDegany can see the create3 topics and I am able to access the RPI topics, 
@hilary-luo and @MikeDegany Please can you guide me on how to get the create3 topics on my PC maybe I am missing something. 
I drop my turtlebot4 and PC details here
https://github.com/turtlebot/turtlebot4/issues/270

hilary-luo, 2023-09-19:
@Kazimbalti I am responding to you on your ticket, and if anyone else wants to provide you guidance that will also happen on your ticket. Tagging me on someone else's ticket does not increase my attention on your ticket. This ticket is for troubleshooting the issue that Mike is having so please respect that.

@MikeDegany When you say
> It sounds that multi-casting works well.
Does that mean that you ran that test and it worked in both directions between the pi and the pc? 

> Do I need to turn off the WiFi for testing wired(LAN) connection.
I assume you are referring to when you connect your PC to the router. In this case yes I would suggest turning your wifi off although it shouldn't matter in this case.

hilary-luo, 2023-09-22:
@MikeDegany Can you try using chrony and your pc to sync the time on both the raspberry pi and the create3? There shouldn't be any issues with running the turtlebot without internet access. I wonder if there was something else different about the system when you tested at home (eg. less traffic through the router, less topics etc.)

Can you replicate this situation, then downlaod and share with me the [create3 logs](https://iroboteducation.github.io/create3_docs/webserver/logs/)? this may give us insight about whether or not the create3 is struggling to broadcast data.

Finally, you can also try turning off the turtlebot4 diagnostics and see if that has any impact on the situation? This should hopefully decrease the amount of traffic on the network.

ipa-rar-sy, 2023-10-10:
@MikeDegany Are you able to use Navigation with custom map for more than 1 robot at the same time. If yes can you please explain your method to do so in short.

closed 2023-11-16 by hilary-luo