turtlebot/turtlebot4 #273: RVIS2 map, camera, and robot model error

Robo52, 2023-09-20:
**Please provide the following information:**
 - OS: (Ubuntu 22.04)
 - ROS Distro: (Humble)
 - Built from source or installed: Source 
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**
 I expected to launch slam, then RVIS2 and then open RVIS2 and have a life camera feed and lidar mapping

 **Actual behaviour**
 A clear and concise description of what you encountered.
Launched RVIS2 and I had errors under Robot Model, Map (No map received), and Camera (Empty Topic)

**To Reproduce**
Provide the steps to reproduce:
1. ssh into Turtlebot4 and run ros2 launch turtlebot4_bringup robot.launch.py
2. run ros2 launch turtlebot4_navigation slam.launch.py
3. run ros2 ros2 launch turtlebot4_viz view_robot.launch.py
4. See errors

**Other notes**
Add anything else you thing is important.

hilary-luo, 2023-09-21:
Unfortunately in general, this is not enough information for me to be able to go off of. Whenever you have errors and are posting about them to a community, please always provide the specifics of where you are seeing the errors and share the exact terminal outputs and/or logs. 

That being said, what I can say is that the only reason to ssh into the Turtlebot4 and run `robot.launch.py` is if you are an advanced user who has modified how the robot works or if you are doing advanced level troubleshooting. If that is not the case please restart your robot and do not do this, it will cause all sorts of errors. If you were directed to do so through some website etc. then please let me know so we can ensure that it is not misleading other folks as well.

hilary-luo, 2023-09-21:
The robot_upstart job launches all of the turtlebot nodes including running that robot.launch.py file. So you do not need to launch any of the robot nodes. You just need to launch SLAM and RVIZ. The [tutorials](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html) on the website do list all launch files that you need to run (and the commands to run) in order to complete each tutorial, so if you haven't already, I would suggest taking a look at those.

Robo52, 2023-09-21:
I tried running SLAM and RVIS2 and it is working much better now, I get a map but I still cannot get a camera feed

Robo52, 2023-09-21:
Also do you have any tips on making the map look better? On mine the edges and obstacles are very rough and jagged

hilary-luo, 2023-09-22:
Have you added the camera feed into your rviz view? the default view does not display the camera. If you don't know how to do this then see the video I posted [here](https://github.com/turtlebot/turtlebot4/issues/227#issuecomment-1648538315) on how to launch the camera feed. That example was with the simulation but the process is the same.

You can increase your map's quality by decreasing the [resolution value in the slam.yaml](https://github.com/turtlebot/turtlebot4/blob/d4412899e428196b0905e4edd2896b8c98ce6580/turtlebot4_navigation/config/slam.yaml#L23). This would mean having a custom slam.yaml config file and passing it in when you launch slam as described [here](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html#launch-slam). That being said, be warned that this comes at a cost. The smaller the resolution measurement, colloquially referred to as the higher the resolution, the more demanding it will be on your PC. In most cases there are no functional benefits of making this change and it will cause increasing issues the longer you try to map.

Robo52, 2023-09-22:
That video cannot be displayed on my device, it says the file is corrupted. I also tried running SLAM and RVIS2 again and was having connectivity and time sync issues so I researched that and found a fix using new .ntpd configs so I changed it. However, trying to use the beta feature "restart .ntpd" on create3's website does not work. It just continuously says "restarting .ntpd" and never finishes

hilary-luo, 2023-09-24:
@Robo52 Regarding the video can you please clarify the exact steps that you took and which step gives the error and the exact error message that you receive? I am not sure what file you are referring to that is corrupted. The process of displaying the video does not involve opening any file.

In regards to the "connectivity and time sync issues please share the exact error messages that you are referring to and the [create3 logs](https://iroboteducation.github.io/create3_docs/webserver/logs/)? The restart ntpd function is a beta function and as such is not necessarily very intuitive. After the restart you should wait to hear the boot up sound and/or see `Ready to go` in the create3 log.

Robo52, 2023-09-25:
I clicked on the link you sent me. It opens that other issue but it tells me the the video cannot be played.

I got an error message when trying to get the Turtlebot4 to move autonomously. It was expecting an initial pose from a certain time stamp but was getting one from a different time stamp. I followed the steps here to update the NTP config at this link: https://github.com/turtlebot/turtlebot4/issues/216

I do not ever hear the boot up sound or see Ready to go in the logs. 

The one message I see in the logs regarding NTPD is this: Sep 25 22:06:50 iRobot-6A7B127B1C0B47BC909CBD8FE930D20F user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.002567 is too high, ignoring

Robo52, 2023-09-29:
Just making sure this issue is still open, I am still getting the same error=

hilary-luo, 2023-09-29:
It is on my todo list to respond but this has been a very busy week. If you can't see that video then maybe try a different browser because the video is there, I am able to view it still.

If your create3 is never getting to the boot up sound then perhaps you should try reinstalling the [create3 firmware](https://iroboteducation.github.io/create3_docs/webserver/update/). Beyond that I will get back to you as soon as I can.

hilary-luo, 2023-10-12:
@Robo52 I will attach the video here, and maybe that will help:

https://github.com/turtlebot/turtlebot4/assets/103377417/174c4012-9893-4728-b882-3aa406f54346

Essentially the steps shown in the video are to open rviz and then add a new display, select by topic, navigate through oakd/rgb/preview/image_raw/ and select Image and then hit okay. The camera feed should show up in the bottom left of the screen.

If you are still having those time sync issues can you please share the full create3 logs?

Robo52, 2023-10-12:
I fixed the camera and the time sync issues! Thank you very much. Unfortunately now I am having another problem. Once the turtlebot gets a certain distance from the computer it disconnects. I was wondering if there is a way to boost the range?

hilary-luo, 2023-10-12:
Can you clarify approximately what distance range you are referring to? (on the scale of 5m, or 50m or 500m). Either way this is going to be based on the range and power of your network / router(s). The robot should be able to travel within the range of your network and generally the fix is to get a more powerful router.

Robo52, 2023-10-13:
It's roughly 10m that it disconnects at, however this is through two concrete walls. Would it be possible to put a range extender somewhere so the turtlebot could connect to that once the other connection becomes slower? Also I was wondering how to manually start the oakd camera from terminal

Robo52, 2023-10-13:
I also frequently get the error from slam: "Queue size has grown to: 321. Recommend stopping until message is gone if online mapping." and the queue size only increases, even if I am not moving the robot. Eventually this causes the robot to disconnect as well.

Robo52, 2023-10-13:
And in addition I now cannot get the robot to undock, dock, or use teleop commands from terminal. The LED screen on the robot is also not coming on, however the robot is connected to wifi. I can SSH into the robot and on our router management page I can see the RPI.

hilary-luo, 2023-10-15:
The robot is connecting to the wifi just like any other linux device so you can go ahead and implement any standard approach towards extending your wifi range. This could mean setting up repeaters or a mesh system etc. 

Why are you trying to launch the camera from terminal? This (alongisde the other nodes) is started automatically as part of the robot upstart job. 

>I also frequently get the error from slam: "Queue size has grown to: 321. Recommend stopping until message is gone if online mapping." and the queue size only increases, even if I am not moving the robot. Eventually this causes the robot to disconnect as well.

This is because your system is not able to process the data as quickly as the data is coming in. If you are wishing to allow the system to catch up when you stop moving the robot, you can use a custom slam.yaml config file and set the [minimum travel distance](https://github.com/turtlebot/turtlebot4/blob/d4412899e428196b0905e4eddurl2896b8c98ce6580/turtlebot4_navigation/config/slam.yaml#L34) to a non-zero value. This will indicate to the system to only expand the map when the robot has travelled that far instead of doing it periodically. 

If you are trying to map a very large environment then you may also need to decrease the resolution of the map or use a system with more processing power.

>And in addition I now cannot get the robot to undock, dock, or use teleop commands from terminal. The LED screen on the robot is also not coming on, however the robot is connected to wifi. I can SSH into the robot and on our router management page I can see the RPI.

Have these issues persisted over a full system reboot? what was the last thing that was done before this issue began?

Robo52, 2023-10-16:
The robot can now undock, dock, and accepts teleop keyboard commands however the LED screen is still not coming on. I am trying to upstart the camera from terminal because when I try to navigate to it in RVIS2, the OAKD topic does not come up, and it also does not come up when I run ros2 topic list. I will try the fix on the queue size, thank you!

Robo52, 2023-10-25:
Fixed the camera and the queue size issue, however still not able to get the LED screen on and the wifi range is still lackluster but I'm working on that fix.

Robo52, 2023-10-25:
When I run that command it outputs "-- No entries --"

hilary-luo, 2023-10-25:
@Robo52 to clarify, you need to ssh onto the robot and run it there, is that what you did?

Robo52, 2023-10-25:
Sorry I was doing it from PC. Here's the log: 
[image]

Robo52, 2023-10-25:
I tried to add a WiFi extender to my network, however when I turned it on it changed the RPI's IP address. I redid all of the necessary setup steps and I can now SSH into the Turtlebot4 RPI again, and I can see all of the topics, create3 and rpi, when calling ros2 topic list, however I cannot send any commands to TB4.

hilary-luo, 2023-10-26:
I would suggest reserving the IP address on your router and disabling DHCP on your extender and set it up so that the router controls the IP address allocation. This may help with the routing as well. 

Can you ssh onto your robot and tell me what file(s) are in `/etc/ros/humble/turtlebot4.d`?

Robo52, 2023-10-26:
it returns 'lite.launch.py'

Robo52, 2023-10-26:
Do I need to re flash the sdcard?

hilary-luo, 2023-10-30:
@Robo52 Given that your robot is a Turtlebot4 Standard, yes I would advise you to reflash the SD Card and make sure that you are using the standard image and not the lite image. That would explain why the screen and camera are not being launched properly.

Robo52, 2023-10-30:
I reflashed the SD using the standard image and did all the setup again however I still cannot send commands. The screen is on, I can see all topics when running ros2 topic list, and that folder on the RPI now contains standar.launch.py. Using discovery server if that makes a difference.

Robo52, 2023-10-31:
I reflashed the SD using the standard image and did all the setup again however I still cannot send commands. The screen is on, I can see all topics when running ros2 topic list, and that folder on the RPI now contains standard.launch.py. Using discovery server if that matters

Robo52, 2023-10-31:
I switched to simple discovery and can now undock and dock the robot, however it still won't fully launch SLAM. I've attached the output from trying to launch it as well as the create3 logs during the time SLAM, NAV2, and RVIS were active.

[image]
[image]

Robo52, 2023-10-31:
Forgot to include logs from RVIS2 and Nav2
[image]
[image]

Robo52, 2023-10-31:
ros2 topic list on PC and RPi don't line up
[image]
[image]

hilary-luo, 2023-11-01:
In general I would suggest using discovery server if that is an option to you. To address the topic lists not being complete / matching try running `ros2 daemon stop; ros2 daemon start`. I have seen this issue pop up when switching between network configurations. 

As for slam, sometimes I have seen (especially with simple discovery) that slam takes a while to load up fully. All of the outputs that you have shared are consistent with that situation. That being said, please do try to copy and paste terminal outputs instead of providing images. Once you have all of the topics matching up and ensure that you are able to echo the `/scan` topic on your pc, launch SLAM and then wait for a few minutes and see if everything comes up and mapping starts.

Let me know if you are interested in further troubleshooting your discovery server setup.

Robo52, 2023-11-01:
I've tried the daemon stop/start multiple times, no change. I am not in the lab now but I will try letting slam boot for a while the next time I'm there and I'll get back to you. Sorry about the images, I'll be sure to copy/paste the outputs next time.

Is it possible to control multiple robots on discovery server?

Also, thank you so much for your help through all of this, it's made working and troubleshooting much easier.

Robo52, 2023-11-01:
I am unable to get the topics to line up and I left SLAM open for ~30 minutes and nothing happened. The /scan topic is listed on the RPi but not the PC.

hilary-luo, 2023-11-02:
@Robo52 If you are still not able to get the scan topic on your PC then SLAM will not work. Currently you are restricted to simple discovery when using multi-robot (although I am hoping to change that soon). 

Looking back at your topic lists again, it looks like your PC is only showing the create3 topics, and not the raspberry pi topics. How is the raspberry pi connected to your network? is it possible that you don't have your 2.4 and 5 GHz networks bridged fully? As a test, you could wire your PC into your router and see if that changes anything.

Robo52, 2023-11-02:
RPi is connected via wifi, not ethernet, however it was working before so I don't know what would have changed. On PC does not have wifi so it already is wired directly to the router

Robo52, 2023-11-02:
Confirmed that RPi is connected to a 5ghz and Create3 is connected to the 2.4ghz. I don't know what you mean by bridge the networks, they are both under the same SSID and password though if that's what you mean.

Robo52, 2023-11-03:
Fixed it! Had forgot to change some leftover settings from Discovery Server. Robot is working great. One question I had, I'm using the collision_monitor node, and it works as intended, however when the robot stops, it takes a long time before rotating/backing up and trying to calculate a new path. I was wondering if there was a way to change this time so the robot tries to configure a new path sooner.

hilary-luo, 2023-11-08:
I'm glad to hear that you were able to fix it!

I can't tell you for certain what is causing this. One possibility is that the Create3 has certain recovery and safety features including a limit on how far it will back up (to avoid backing up over a "cliff"). If you are running into the backup limit then it is possible that Nav2 is telling it to continue backing up and the Create3 will refuse for safety reasons. If it is this situation then you can disable that feature as described here: https://iroboteducation.github.io/create3_docs/api/safety/

closed 2023-11-20 by hilary-luo